embedded-graphics/embedded-graphics #9: ssd1306 broken by latest changes?

therealprof, 2018-04-01:
It seems something broke in the SSD1306 driver in the last few hours:

```
   Compiling ssd1306 v0.1.0 (file:///Users/egger/OSS/ssd1306)
error[E0061]: this function takes 3 parameters but 4 parameters were supplied
  --> examples/image_i2c.rs:56:14
   |
56 |     let im = Image1BPP::new(include_bytes!("./rust.raw"), 64, 64, (32, 0));
   |              ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^ expected 3 parameters

error[E0061]: this function takes 1 parameter but 2 parameters were supplied
  --> examples/text_i2c.rs:51:15
   |
51 |     disp.draw(Font6x8::render_str("Hello world!", (0, 0)).into_iter());
   |               ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^ expected 1 parameter

error[E0061]: this function takes 3 parameters but 4 parameters were supplied
  --> examples/rotation_i2c.rs:67:14
   |
67 |       let im = Image1BPP::new(
   |  ______________^
68 | |         include_bytes!("./rust.raw"),
69 | |         64,
70 | |         64,
71 | |         (w as u32 / 2 - 64 / 2, h as u32 / 2 - 64 / 2),
72 | |     );
   | |_____^ expected 3 parameters

error[E0061]: this function takes 1 parameter but 2 parameters were supplied
  --> examples/text_i2c.rs:52:15
   |
52 |     disp.draw(Font6x8::render_str("Hello Rust!", (0, 16)).into_iter());
   |               ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^ expected 1 parameter
```

jamwaffles, 2018-04-01:
Correct, I've changed some function sigs. I'm allowed to do that because we're not 1.0 yet or something ;)

I'm going to fix `ssd1306` and pin it at a working version of `embedded_graphics` in a bit so we don't get this issue again. You can pin at [a52f8a86215c3606a73f6321359a014d07d7f3c4](https://github.com/jamwaffles/embedded-graphics/commit/a52f8a86215c3606a73f6321359a014d07d7f3c4) for now which should work though.

therealprof, 2018-04-01:
No worries, I'm still fighting with the traits.

closed 2018-04-01 by therealprof