embedded-graphics/embedded-graphics #14: Display simulator/plotter

jamwaffles, 2018-04-05:
As demonstrated by #4 and #11, it would be great to have a display simulator to render things out into. Extending it to do snapshotting as part of the CI process would be super useful.

byronwasti, 2018-05-03:
What are you thinking the implementation of this would look like? As in, would it be another module in the library, or would it be in the example code, or something else entirely? 

I would be happy to tackle this, since I have some SDL2 graphics code from another project of mine and I am in need of this functionality.

jamwaffles, 2018-05-03:
I started a simulator with KISS3D on the [`simulator` branch](https://github.com/jamwaffles/embedded-graphics/tree/simulator). I haven't got very far and performance is terrible but you might want to base something off that or replace it entirely. 

I went with Cargo workspaces to make organisation better. If you wanted to create a decent sim as a workspace sub-package that'd be great.

byronwasti, 2018-05-03:
I've made a pull-request for a basic simulator I threw together. It is good enough to serve my purposes, but I'm not convinced it is fully useable.
PR: #29

jamwaffles, 2018-10-10:
The simulator as it stands is pretty useful and has existed in its present state for a while, so I'm closing this issue. PRs welcome if there are bits missing.

closed 2018-10-10 by jamwaffles