embedded-graphics/embedded-graphics #21: Drawing integers and floats?

chocol4te, 2018-04-27:
Hello,

`std` is of course unavailable in embedded so I can't use the normal number to string conversion, what is the best practice for drawing integers and floats?

Thanks :)

jamwaffles, 2018-04-28:
It's not possible at the moment, but it would be a great feature to have. PRs welcome :)

chocol4te, 2018-04-29:
Progress update:

Found two libraries for converting numbers to strings, and they complement each other hilariously.

- numtoa is designed specifically for no_std and is working great except I cannot seem to pass a generic number to it from a parent function https://github.com/mmstick/numtoa/issues/12

- itoa was designed with generic implementations in mind and works great, except it doesn't play nice with no_std  https://github.com/dtolnay/itoa/issues/17

I have duplicated embedded-graphics twice, one for attempting to implement itoa/dtoa and another trying to implement numtoa. To get itoa working I need to create a writer, but Rust prevents traits being  implemented outside their crates. To get numtoa working I think I need to create a macro for creating `impl`s for all types :|

jamwaffles, 2018-05-01:
Thanks for having a crack at this.

> As a side note, I think at some point it might be a good idea to create a fmt::Display trait

I had a quick go at this the other day but didn't get far. I'm really keen to support it though as it would make drawing all sorts of text much easier. Manually nailing together bits of strings is obviously not very nice. Are you going to try to implement `fmt::Display`?

chocol4te, 2018-05-01:
I think I’ll just try and get numbers working for now, but of course a lot of code can be reused. I should have some free time after my exams :)

chocol4te, 2018-05-05:
Update 2:

I can get it printing unsigned and signed integers, but it only works when the type specified here `<T: NumToA<i64>>` is the same as the type called in the function input.

```
fn render_num<T: NumToA<i64>>(num: T, digits: usize, buffer: &'a mut [u8; 20]) -> Font6x8<'a> {
        let _n = num.numtoa(10, buffer);

        let text = from_utf8(&buffer[20-digits..20]).unwrap();

        Self { pos: (0, 0), text }
    }
```

`disp.draw(Font6x8::render_num(-211324, 7, &mut buffer).translate((0, 56)).into_iter());`

Do you know how I can get this working for all numbers? Would a new trait and a set of impls be the right way to do it? I will have to write a separate implementation for floats anyway (Display as two integers separated by dot?).

chocol4te, 2018-05-06:
Final update:

I think [`alloc-cortex-m`](https://github.com/japaric/alloc-cortex-m) is the best solution by far.

    disp.draw(Font6x8::render_str(&format!("test {:.2}", 123.4569f32)).translate((0, 56)).into_iter());

Thoughts? I might create a small example later. :)

jamwaffles, 2018-05-29:
I'm really sorry for not getting to this sooner. embedded_graphics _must_ work without an allocator. I'm open to adding more tertiary features that require an allocator, but I think printing numbers is a core requirement and should work in non-allocator environments. 

I noticed you closed mmstick/numtoa#12. Did you find a solution? Have you tried [`Num`](https://docs.rs/num-traits/0.2.4/num_traits/trait.Num.html) from [num-traits](https://crates.io/crates/num-traits)? It should work without an allocator by the looks of it.

chocol4te, 2018-05-30:
Sorry, I've been pretty busy with exams and haven't had much time to work on this. I closed that because apparently `numtoa` requires a Write impl for whatever data type you want to send your formatted text to, which puts us back to square one. I think the non-allocator method with `itoa` and `ftoa`  is our best option right now.

jamwaffles, 2018-05-30:
> I think the non-allocator method with itoa and ftoa is our best option right now.

Seems to be unfortunately. Do you mean [this `ftoa`](https://github.com/maciejhirsz/ftoa/) because it seems to require `std`.

jamwaffles, 2018-08-07:
I've just come across [this usage](https://github.com/MabezDev/ssd1351/blob/110ff84377dd875741fa055905486f51a7e95d24/examples/time.rs#L88-L100) of the [`heapless` crate](https://docs.rs/heapless/0.3.6/heapless/). I can't try it myself right now, but maybe that will do what you need?

chocol4te, 2018-08-07:
Definitely looks like it, trying now :)

chocol4te, 2018-08-09:
```
fn from_args(args: core::fmt::Arguments) -> Font6x8<'a> {
        let mut buffer: String<U16> = String::new();
        core::fmt::write(&mut buffer, args).unwrap();
        let output: &'a str = buffer.as_str();

        Self { pos: (0, 0), text: output }
}
```

I've written this as a trait for Font6x8 alongside `render_str`, to be called like this:

```
display.draw(
        Font6x8::from_args(format_args!("{} foo {:X?}", value, 2))
            .translate((5, 50))
            .into_iter(),
);
```
 but it is not working as `"buffer" does not live long enough`. I know I must be doing something stupid, but I can't see what. I'm creating a buffer, using a mutable reference to write data to it, then creating a slice with the required lifetime from it, and attempting to return the a struct containing the slice.

Any advice?

jamwaffles, 2018-08-10:
I like the idea of using `format_args!()`, but the `String<U16>` is limiting as it fixes the max length of a string to 16 chars AIUI. That's fine for user code, but not something this crate should have in it. A macro with the ability to specify which `String<U_>` to use might be the best way to go for creating formatted strings to give to `render_str`.

Not sure how to solve the lifetime issue as it's still an area of Rust I'm not too good with.

chocol4te, 2018-08-10:
Of course. Actually, since the only supported fonts are currently ASCII, we could even get away with String<U8>, but create different a method for a Unicode font.

jamwaffles, 2019-03-09:
I've recently needed to print a number in one of my projects, and I ended up using this pattern with the `heapless` crate:

```rust

let mut buf = String::<U8>::new();

write!(&mut buf, "Count: {}", rtc.seconds());

display.draw(
    Font8x16::render_str(&buf)
        .translate(Coord::new(0, 16))
        .with_stroke(Some(1u8.into()))
        .into_iter(),
);
```

I think this is acceptable for printing stuff in a no-alloc environment, and it leaves the implementor open to using an allocator and variable-length strings if desired. Embedded graphics will handle both, so that's all I think the library needs to do for now (font rendering improvements notwithstanding). Thanks for all the input!

closed 2019-03-09 by jamwaffles