embedded-graphics/embedded-graphics #22: [unsorted] Enchancements

puzrin, 2018-04-30:
Continue of https://github.com/rust-lang-nursery/embedded-wg/issues/22#issuecomment-385438053.

This is assorted thought, to discuss general library improvements, prior to create separate issues. At first, i'd like to say, that i don't invent something new. Everything described below were already implemented in other languages, and i only point to missed things. I strongly recommend to investigate [u8g2](https://github.com/olikraus/u8g2) sources and [wiki](https://github.com/olikraus/u8g2/wiki) as source of ideas. That doesn't mean it's ideal, but it has a lot of useful solutions like font formats.

Basic things (to be honest, what i need fisrt to migrate from C to Rust :)

- The most demanded, more font sizes (up to 24-30px), with tools to extract required char subsets. +more compact font store format.
- (probably) More compact framebuffer format for monochrome display (1 byte per 8 pixels, in your [ssd1306](https://github.com/jamwaffles/ssd1306))
- (probably) Async inteface support

new

- (probably) Consider what is needed to be a good base api layer for window manager (library to create interactive menus and other things). I did not investigated this area yet. Just suppose, that some menu system is required in many devices with LCD.

This does not include all supported u8g2 features. Need time to collect more details and analyze other libs.

---

Answers to your questions in https://github.com/rust-lang-nursery/embedded-wg/issues/22#issuecomment-385438053:

> The fonts are stored as 1 byte per 8 pixels, so I'm not sure how much more compact they could get without using some sort of runtime decoding which I'd like to steer away from.

See https://github.com/olikraus/u8g2/wiki/u8g2fontformat

- At first, you do not support partial lists & lists with holes. For example:
  - One may wish to have english and russian languages. But still work with "normal" strings when writing code. User should be able to extract only required chars, and font should be able to store those.
  - Mad case: one may wish to write "hello wold" only. Then all english letters are not required.
  - Real case: I need very big font (60px), but used to display digits only. Then i need only `0..9` and `.` chars to save space.
- u8g2 seems to support some data compression, but i did not investigated how effective is it (for example, it does not use bytes to encode space).

May be, using u8g2 fonts format can be convenient, because you will be able to use [existing tools](https://github.com/olikraus/u8g2/tree/master/tools/font) until native ready.

Probably, fonts & tools could be in separate crate to avoid licences mixture.

> I'm not sure what you mean for embedded-graphics as it doesn't use a framebuffer at all. Can you elaborate?

`embedded-graphics` itself does not use framebuffre, but ssd1306 has it https://github.com/jamwaffles/ssd1306/blob/master/src/mode/graphics.rs#L46.

Also, it may be useful to optimize display bulk writes and to stack async API calls effectively:

```js
eg
  .draw_line(...)
  .draw_text(...)
  .draw_anything(...)
  .flush() // <- future
```

That's only a semantic example, convenient async api is a separate topic to discuss. Also result can affect desision, if `embedded-graphics` should have direct access to framebuffer or not.

I don't know your plans on this library. To be honest, i have some pressure from existing projects support, and would like to avoid direct participation in code development. But i could help to analyze existing solutions & suggest possible end-user api/tools designs, because have some experience with such things.

jamwaffles, 2018-05-01:
That's a lot of stuff, thanks for taking the time to type it all out. I'll address some points below:

* Fonts:

  > The most demanded, more font sizes (up to 24-30px)

  Certainly. embedded-graphics only has one font at the moment only because I haven't had time to add more :)

  > May be, using u8g2 fonts format can be convenient, because you will be able to use existing tools until native ready.

  That might be an option, yes. I'm a little worried about how brittle a custom format would be though. I like the idea of using BDF files as source data. The current font format is pretty inefficient. In the short term, I don't think supporting RLE (Run Length Encoding) would be too difficult. Fonts definitely need some more thought in the longer term.

  > u8g2 seems to support some data compression, but i did not investigated how effective is it (for example, it does not use bytes to encode space).

  I think it just uses RLE?

* > embedded-graphics itself does not use framebuffre, but ssd1306 has it https://github.com/jamwaffles/ssd1306/blob/master/src/mode/graphics.rs#L46.

  That's correct, but it's an 8 pixel per byte buffer so it can't get any more compact :)

  > Also result can affect decision, if embedded-graphics should have direct access to framebuffer or not.

  Embedded-graphics should not have a framebuffer. It instead takes an `Iterator`-based approach. The consumer of embedded-graphics can then choose whether or not to use a display buffer, or (in the future) cache the output of multiple iterators into a reusable chunk of data.

* > (probably) Async inteface support

  Shouldn't the display driver be async as opposed to embedded-graphics itself? Can you explain your use case? I might be missing something.

* Chaining draw calls: absolutely! This is something I'd really like to see in embedded-graphics; a feature that allows the user to define custom objects composed of multiple primitives in an easy way.

So, moving forward (in the short term at least), I have some proposed features/fixes:

* Add more fonts:
  * A 24-30px font
  * Some intermediate sizes
  * A couple of sizes of number-only font
* Add a tool (and documentation) that Run Length Encodes the current/new font [image/bitmap files](https://github.com/jamwaffles/embedded-graphics/tree/master/data) as a small short-term win for reducing flash usage.
* Create some way of collecting/chaining draw calls to make reuse/creating custom drawable objects easier: I've opened an RFC in #25 with part of what I want for this. Another part is being able to pass your own structs into a `draw()` call, like:

  ```rust
  struct Window {
	  // ...
  }
  
  let mut disp = MyDisplayDriver::new();
  
  let window = {
	  title: "Hello world",
	  // etc
  }
  
  disp.draw(window.into_iter());
  ```

  which I hope to add soon. FWIW I really want to get rid of the `.into_iter()` calls in `.draw()` but I think we need `impl trait` to work with structs for that to happen :(

I realise the above doesn't address everything you want in embedded-graphics right now, but hopefully it will make some steps towards making it more useful for you and others. Let me know what you think.

puzrin, 2018-05-01:
> I think it just uses RLE?

As far as i understand, they try to keep only bounding box content. I did not measured savings. IMHO 30% of savings is not significant. Char subsets support is much more important.

> Shouldn't the display driver be async as opposed to embedded-graphics itself? Can you explain your use case? I might be missing something.

As far as i understand, everything freezes when you send data via i2c/spi (how long for?). And existing limitations are caused by HAL. Question is - should we design API to be async-extandable in future or not (my example is ok for both sync and async cases).

PS. May be i'm not correct, because RTFM is interrupt-driven.

jamwaffles, 2018-05-01:
> should we design API to be async-extandable in future or not 

I'm not sure how you'd go about asyncifying an `Iterator` but if it's easy to do in the future then we should. Right now I think it's ok to have synchronous iterators. Embedded-graphics doesn't go near the HAL so blocking on hardware is an out of scope issue.

puzrin, 2018-05-01:
At first, we should investigate, what is needed to be async in global and how to keep things simple. Drawing process consists of 2 parts (in general):

- Maths, calculating points to draw (sync, if we do not fall into very complex things like graphics accelerators)
- Sending data to LCD (may be fast, me be slow... who knows..., and this is the only place worth to be async)

Since we need convenient user interface, the most simple approach is:

- Draw LCD image in sync way
- Send data in async way

There can be 2 (similar) methods:

- raw framebuffer
- raw framebuffer + Dirty maps (to optimize sending)

Problem is how to split libraries, to make extensions development ([SD1306](https://github.com/jamwaffles/ssd1306) and others) simple and convenient. If we say "everything about framebuffer should be in driver", then driver may tend to be too complex.

puzrin, 2018-05-01:
<tl;dr>

Let's stay aside programming details and consider data flow only (top level architecture design). Drawing process in general look like this:

[User API] -> [Internal data representation] -> [Display]

Internal data representation depends on:

- What we need to draw (objects complexity)
- Hardware features and desired optimizations (acceleration)

**What we need to draw**

- Text (fonts). Almost all user interfaces should display texts. This can't be avoided. Let's accept this as minimal required complexity level
- Other things like lines, circles, images etc. Those are technically more simple than fonts. So, if our architecture is ok with fonts, it will be extendable for anything else.

**Hardware features and desired optimizations**

I guess, final goal is to support "simple" LCDs (without graphics accelerators). Then the only available optimization is bulk writes to LCD bus. What about desired hardware:

- Monochrome OLED (and other) displays with spi/i2c/... buses. Those are very popular, must have.
- Color TFT displays with spi-like buses. Would be nice, but less critical. Let's remember estimate - 3 bytes per pixel.
- eInk displays. Nice to have. Similar to OLED's, but may have 2bits per pixel (black, white, red).

**Internal data representasion**

When user draw something like line, we must have some internal data between API call and LCD. Most obvious are:

- Pixels. Simple, but ineffective. It's too slow to draw text letters (minimal required feature) pixel-by-pixel
- Shapes. Video cards approach. Out of scope.
- Framebuffer (with optional optimizations). Looks preferable for our needs.

---

Summary: this may explain, why it may be profitably to make framebuffer & related things a part of `embedded-graphics`, instead of keeping it in drivers.

jamwaffles, 2018-05-02:
> Those may exists as exported & extendable primitives, to simplify drivers implementation.

I could roll with that :). In this case, I think then it's important to make everything that comes out of embedded-graphics compatible with _both_ an `Iterator`-based API as well as a buffered API. I don't want to lose the low memory usage of iterators but there are certainly performance gains to be had by using a framebuffer.

puzrin, 2018-05-02:
> I don't want to lose the low memory usage of iterators but there are certainly performance gains to be had by using a framebuffer.

> I feel like most drivers will probably implement some sort of framebuffer anyway, so providing a useful common base for that in embedded-graphics is probably a good idea.

Yes, problem is, that there are no implementations without framebuffers if we look at whole system (or, may be i'm not familiar with such implementations). So, initial idea to have no framebuffer does not add something useful in the end (in driver), but seriously restricts us in optimizations and general api design.

We can go from another end - popular u8g2 lib use framebuffer and there are no claims in issues about memory. Do you have examples, when framebuffer size is critical?

jamwaffles, 2018-05-02:
> Problem is how to split libraries

Embedded-graphics should have practically no knowledge of what it's being rendered on. My original vision for this library is to provide a set of _things_ that can be used by other code to be rendered on something else, whether that be an SSD1306 connected to an ARM chip or a simulator on a PC.

There are a lot of display modules that have onboard memory, displays like the SSD1306 as you mentioned, as well as the SH1101, SSD1331, etc, which makes it possible to not use a framebuffer if you're careful. I can also see users of small micros (ARM, even AVRs and PICs when supported) wanting to reduce memory consumption as much as possible. 

> So, initial idea to have no framebuffer does not add something useful in the end (in driver), but seriously restricts us in optimizations and general api design.

Not really IMO. One must iterate through the pixels to generate a primitive/font/image/etc anyway, so exposing that interface to people who want to use it is a nice feature to have. Exposing some kind of framebuffer API _built on top of those iterators_ seems to be the best of both worlds to me.

puzrin, 2018-05-03:
Seems i have not enouth experience to design such low level API-s "in advance", without GUI lib examples.

Take a looks at https://github.com/vurtun/nuklear. It's [backends](https://github.com/vurtun/nuklear/tree/master/demo) are very similar to what you do, but font rendering is left to upper level (and nuklear has simpllified TTF renderer inside).

closed 2019-09-03 by jamwaffles