embedded-graphics/embedded-graphics #26: RLE-encode font bitmaps

jamwaffles, 2018-05-01:
From #22, a first step towards more efficient/better fonts is to compress their binary representations in flash. RLE seems easy enough to do. Now that some more font processing needs to happen (PNG -> 8BPP -> RLE), this work should create a tool to help out. It will replace `convert_1bpp.sh` with a Rust program.

* [ ] Create tool to convert PNG font bitmaps into 8BPP, RLE raw files. This should be a separate project using the Cargo workspaces feature
* [ ] Add tool link/quick docs to readme
* [ ] Convert raw font files and font iterators to use new RLE format.

puzrin, 2018-05-01:
I'd suggest to not invent new font format. One in u8g2 is good enougth and allow to use existing tools until native done. Also, it worth consider split out font data & tools into separate package, to avoid issues with lincenses mix. If i'm a "stupid user", i may wish simple thing: take ttf source and generate data files with desired subsets & size.

So, there may be such steps:

- Create a separate repo
- Start collect useful ttf fonts with acceptable licenses (mostly monospace)
- Create some default pregenerated fonts.
- Add native tools when you have time

jamwaffles, 2018-05-01:
Hm, you make good points. Thinking about it a bit more this issue is a bit pointless. I like the idea of a separate repo with bundled tools that isn't tightly coupled to a graphics project. I don't really have time to start something like that at the moment so for now I'll go with just adding more fonts in #27. Of course you could always get started on `embedded-fonts` yourself ;).

closed 2018-05-01 by jamwaffles

puzrin, 2018-05-01:
http://www.cl.cam.ac.uk/~mgk25/ucs-fonts.html - good page with explanations and references to X11 bdf fonts and tools.

Need some time to structure data and user workflows for new repo. Do you agree to switch `embedded-graphics` at some point to new font format? I could try to prepare repo structure, tools and documentation. I'm not rust developper, but this work does not require rust knowlege. For example, tools can be done in node.js :smile: and replaced later (or never replaced). Since published crate will be just a set of binary font data (other directories ignored), internal tools language is not critical (at least, at start point).

What i mean with words "i'm not rust developper". I have some hobby projects ([1](https://easyeda.com/speed), [2](https://easyeda.com/reflow/)), and i really wish to study rust with writing "rust firmware" for those. But since final goal is to have working hardware (we should not go too far from real world), right now it's more reasonable to write quick&dirty in C, until Rust Edition 2018 released. To switch to rust without pain, some libraries should be improved too. So, i can try to participate in parallel in this process, if my help is useful.

jamwaffles, 2018-05-02:
> Do you agree to switch embedded-graphics at some point to new font format?

Yep :) sounds good. Ideally embedded-graphics is as decoupled as possible from whatever format the fonts use but that shouldn't be too hard.

> I could try to prepare repo structure, tools and documentation.

Please do. Documentation will be particularly important for a custom(ish) binary format. I like the ability at the moment to look at a source font PNG and see what it will look like, so a sample page for fonts similar to how [Uzebox does it](http://uzebox.org/wiki/Font_Bitmaps) should be pretty high up the priority list IMO.

> For example, tools can be done in node.js

Sounds like a good opportunity for you to learn Rust ;). If it's going to be published as a Rust crate I would definitely like to see the tooling written in Rust at some point.