embedded-graphics/embedded-graphics #27: Moar fonts

jamwaffles, 2018-05-01:
From #22 

Add more bitmap fonts:

* ~A 24-30px font~ Can't find a pixel bitmap. Something to add in the future with a better font story
* [ ] [6x12](http://uzebox.org/wiki/Font_Bitmaps#6x12)
* [ ] [8x16](http://uzebox.org/wiki/Font_Bitmaps#8x16)
* [ ] Number-only versions of the above fonts

chocol4te, 2018-05-07:
I'll be making a PR with 6x12, 8x8, 8x12 and 8x16 soon, but I'm having an issue with wrapping text on the taller fonts if a string is longer than the width of the display. I can't see what's wrong, it should work for arbitrary font heights. Any advice?

Thanks :)

MabezDev, 2018-12-21:
I am wondering if we can use some of these fonts from [u8g2](https://github.com/olikraus/u8g2/wiki/fntlistall). A brief look at the repo and it seems they are using a tool to convert ttf's to bdf format. Haven't looked to see what they do after that but I assume there is another conversion step to get raw bytes. If we could figure out what they are doing we have a very good possibility of supporting a lot more fonts.

jamwaffles, 2018-12-21:
I've had a look at u8g2 also. The BDF format is a bit odd, but easy enough to work with I think. I started working on the [embedded-graphics](https://github.com/jamwaffles/embedded-fonts) project a while back, which aims to take a BDF file and convert it into a bitmap for use with embedded-graphics, but I haven't worked on it recently.

I'm not sure I like the idea of parsing BDFs at runtime, particularly on small micros. Perhaps a proc macro would be a good way to go? Either way, the ability to use the fonts provided by u8g2 in some way would be a huge enhancement to this crate.

MabezDev, 2019-02-09:
> I'm not sure I like the idea of parsing BDFs at runtime, particularly on small micros

So it looks like they convert bdf to there own format, you can find the parser [here](https://github.com/olikraus/u8g2/blob/master/csrc/u8g2_font.c) it looks like they just stick some info about the fonts at the beginning of the font, then at run time load that information into a struct.

I think there are two paths that we could go down, either try and mimic u8g2 parsing exactly or think of a better way to do it in Rust. The benefit of replicating u8g2 is that we can extract the fonts [from here easily](https://github.com/olikraus/u8g2/blob/master/csrc/u8g2_fonts.c). The benefit of doing it from scratch in Rust is that there is almost certainly a zero cost solution, like you said with proc_macro's or some otherway.

What do you think?

jamwaffles, 2019-02-10:
> The benefit of replicating u8g2 is that we can extract the fonts [from here easily](https://github.com/olikraus/u8g2/blob/master/csrc/u8g2_fonts.c).

They also have a pile of BDF fonts [here](https://github.com/olikraus/u8g2/tree/master/tools/font/bdf) (and a bunch of tooling for them in the parent directory) which might be a better way to go as the font data is decoupled from a language, but still in a pretty widely supported format. It's then possible to leverage existing TTF-to-BDF (and other) tooling both in u8g2 and elsewhere, whilst having the freedom to post-process these bitmaps into something more optimised for embedded Rust. It would be great to have an all-Rust font pipeline for embedded-graphics in the future, but keeping compatibility with at least some of what u8g2 does/can do would be a good idea for the medium term I think.

I started a BDF parser in Rust [here](https://github.com/jamwaffles/embedded-fonts) a while ago but haven't touched it in a while. I had in mind it would become part a BDF-to-embedded-graphics pipeline, where anything upstream from the BDF is handled by something else, maybe replaced by more Rust tooling in the future :smirk:. It's more work than writing some wrappers around the pre-compiled C you link to, but I think it's a better solution long term. Could this work?

MabezDev, 2019-02-10:
> I had in mind it would become part a BDF-to-embedded-graphics pipeline, where anything upstream from the BDF is handled by something else, maybe replaced by more Rust tooling in the future 😏

I agree, a Rust tool would be much nicer. What do you envision the BDF being converted to that embedded-graphics can understand?  Correct me if I am wrong, but currently the fonts are just bitmaps and embedded-graphics calculates the offset for each character. Perhaps a bdf-to-bitmap-glyph-map tool would work, then embedded-graphics wouldn't have to change at all. Are there any limitations that you know of in the current font system?

jamwaffles, 2019-02-11:
That's how it works, yeah. Either 1- or 8-bpp (haven't looked at the code in a while). I doubt it would be difficult to write a converter. My only concern is with non-ASCII characters and how large those offsets might make the image, but it'd be a good way to start using all the fonts available in u8g2.

MabezDev, 2019-02-13:
> My only concern is with non-ASCII characters and how large those offsets might make the image, but it'd be a good way to start using all the fonts available in u8g2.

It might be better to offer 'pages' of the font so that fonts with more characters than ascii don't end up being huge. Are you thinking about supporting unicode glyphs aswell?

jamwaffles, 2019-02-13:
Pages could work. I started writing a no-std TGA parser with run length encoding support which should be pretty efficient for the monochrome fonts in e-g. I like the idea of using an image format that's viewable/editable directly in a lot of software, but we'll have to see what the performance and size cost of using RLE TGAs is.

> Are you thinking about supporting unicode glyphs aswell?

I'm not sure what you mean. Things like joining characters and emojis, et al, or simpler things like character sets from different languages? The latter is pretty important, so I'd like to support that if possible.

Right now, the problem is getting fonts _into_ emedded-graphics, not necessarily what they contain or how they're stored - those problems can be solved later with pages, TGA, whatever. If I have some time (help always welcome!), I'll look into what it'd take to get u8g2 fonts into embedded-graphics with as much reuse of existing tooling as possible. The end goal is to have a complete Rust pipeline, but something is needed to just get off the ground first. u8g2 has a great selection of fonts, so let's leverage that in as lazy way as possible ;)

I'm closing this issue in favour of #65 for a more targetted discussion. Thanks to all for the feedback!

closed 2019-02-13 by jamwaffles