embedded-graphics/embedded-graphics #28: [sandbox] New font format & tools

puzrin, 2018-05-02:
That's a sandbox to discuss requirements for new font format, tools and related things.

Fonts:

- Both monospace and proportional fonts should be supported
- Should be extendable with kerning info in future (at binary format level)
- What strings encoding should be used?
  - utf8 is preferable, but how difficult is it?
  - hieroglyphs?
  - ligatures? (probably - not in this life :)
- Should we support marcos for recoding to write strings in convenient way? Example - if someone wish to use koi-8 encoding. Probably - no, because utf8 is preferable.
- What source format is preferable for generated data? I'd prefer text `.rs` files (they should define complex data structures).
- What optimizations needed for fast processing, to avoid bulk scans (and so we need such opts)? (sorting by glyph codes, etc)

Tools / data sources:

- Small fonts should be pixel-perfect - take from BDF (X11 and other)
- Big fonts should be taken from TTF (via `ttf2bdf` utility)
- Allow strip char subsets to save space
- Allow merge fonts (text + icons)
- Remap charcodes (if font has koi-8 encoding, or if we merge icons)
- Generate fonts demo page

Feedback needed. Add missed things.

TeXitoi, 2018-10-09:
https://github.com/jamwaffles/embedded-graphics/pull/48 gives you a way to add your own font with the encoding you want (there is latin1 fonts now, but supporting greek should be easy). That's only for fixed size and only by codepoint.

puzrin, 2019-05-09:
https://github.com/littlevgl/lv_font_conv - this is result of work:

- New font spec, satisfying most common needs (including size care and optional kerning)
- `node.js` convertor with merge/subsetting support.

TDHolmes, 2019-07-22:
I'm thinking about diving into better font support as well as a personal project needs larger font sizes than already exists. Looks like there's pretty good rust tooling around parsing font file formats that we could leverage (see [this](https://github.com/bodoni/font) and [this](https://github.com/bodoni/opentype)) which we could use to raster down to bitmaps for embedded use. (cc @BroderickCarlin)

puzrin, 2019-07-22:
@TDHolmes i'd suggest to separate bitmap format and live parsers. Those have different hardware requirements (memory, first), and different targets.

As soon as you have enougth RAM, it worth switch to FreeType.

BroderickCarlin, 2019-07-22:
@puzrin, I would be curious to hear your thoughts on defining `features` that can be utilized to provide functionality such that the developer can decide if fonts are rastered at compile time to bitmaps or rastered at run time. This allows a developer to use any font that they may desire without having to worry if their micro is able to support the memory/processing required to raster a desired font. 

I believe that an approach such as this would result in an efficient code base that is still exposes an easy to use API. I believe that it also can provide a good means for meeting a lot of the requirements you proposed in your original post, namely: 
- Big fonts should be taken from TTF (via ttf2bdf utility)
- Allow strip char subsets to save space
- Allow merge fonts (text + icons)
- Remap charcodes (if font has koi-8 encoding, or if we merge icons)
- Generate fonts demo page

puzrin, 2019-07-22:
I don't believe in silver bullets :). Also, i'd suggest to be more careful about existing solutions. For example, ttf2bdf will cause kerning loss. And that was one of reasons why i had to create new font spec.

jamwaffles, 2019-08-22:
Is there a desire for complex fonts to be available from `embedded-graphics` itself? There's a lot of work around getting fonts into the right format, specifying and implementing custom storage formats, etc. I also don't have experience or strong use cases for really good embedded fonts and, as maintainer, I'd be responsible for coordinating this work inside e-g even if I didn't write the code. 

I think a better solution would be to release an `embedded-fonts` (or better named) crate or set of crates that make a much richer fonts story than the built-in bitmap fonts for e-g. Integration can be achieved by implementing various e-g traits (which might need changing or adding to) whilst leaving maintenance to those with more experience and need for good fonts than me 🙂. Thoughts?

puzrin, 2019-08-22:
> There's a lot of work around getting fonts into the right format, specifying and implementing custom storage formats, etc. I also don't have experience or strong use cases for really good embedded [...]

See https://github.com/jamwaffles/embedded-graphics/issues/28#issuecomment-490886600. That's already done, and successfully used in lvgl. You can use all achievements, and need to implement decoder only. That's not difficult.

All that was done with intent to be language agnostic and useable for rust.

> I think a better solution would be to release an `embedded-fonts` (or better named) crate or set of crates that make a much richer fonts story than the built-in bitmap fonts for e-g. Integration can be achieved by implementing various e-g traits (which might need changing or adding to) whilst leaving maintenance to those with more experience and need for good fonts than me slightly_smiling_face. Thoughts?

To be honest, font renderer should be certainly separated from e-g. Full unicode rendering, with combined glyphs support and other things is not trivial thing.

Also, i'd suggest to revisit idea of e-g, because i can't understand architecture balance. It looks like a combination of assorted things.

For example, if one wish implement GUI lib, it needs multi-pass rendering, when no memory to store full frame buffer. But current primitives do not support region clipping.

Second strange thing is keeping primitives and drivers in the same package. That's too different abstraction layers.

That's only my personal opinion, not pretending to anything.

dutchmartin, 2019-08-25:
To me, there is a need for larger fonts.
I tried to make a image with all the characters I needed to use in the current implementation but I gave up on that since using a proven tool is just way better. Having complex fonts as a side effect by implementing a decoder for lv_font seems like the best option.

> See [#28 (comment)](https://github.com/jamwaffles/embedded-graphics/issues/28#issuecomment-490886600). That's already done, and successfully used in lvgl. You can use all achievements, and need to implement decoder only. That's not difficult.
The thing is that for someone who has no experience in this field, it is not super simple. But if you are open to mentor me in writing a decoder, I thing we can get it done.

Currently trying to implement a decoder, with the first effort to read out a [generated bin file here](https://github.com/dutchmartin/lv_font_reader/blob/master/src/main.rs)
However, I have no file to validate if what I read is correct. @puzrin, do you have a file you know the values of? That would help validate my file reader.

For implementing it in this library I [already made a bare-bone font builder](https://github.com/dutchmartin/embedded-graphics/blob/lv-font-impl/embedded-graphics/src/fonts/lv_font.rs), which is inspired upon the current fontbuilder. I just need to parse the file and generate the correct Pixel Iterator and it should work.

puzrin, 2019-08-25:
> The thing is that for someone who has no experience in this field, it is not super simple. But if you are open to mentor me in writing a decoder, I thing we can get it done.

I will help if any questions. Just don't change anything :)

> However, I have no file to validate if what I read is correct. @puzrin, do you have a file you know the values of? That would help validate my file reader.

No, guys from LVGL requested text output. So, you may be sure spec is valid, and  internal structures/algorythms are valid, but nobody tested bin output in real projects (only text output without compression was used). Though, all critical parts are covered with tests.

I'd suggest you  2 things:

- use `dump` format to quickly inspect glyphs
- see tests https://github.com/littlevgl/lv_font_conv/tree/master/test/font

puzrin, 2019-08-26:
> @puzrin, I am curious to hear what you think would need to be changed to the build-in traits to allow you to utilize them for the work you have already done in regards to your compact bitmap font crate

IMO in general, font drawing can be narrowed to a sequence of "bitmap draw" operations. That will simplify e-g API and allow use any external font engine. If you don't plan to avoid internal buffer for single decompressed glyph, that will be enougth. In other case, stream-like interface required.

I can not comment traits, because still don't use rust. But i can comment data flows in architecture. As far as i understand `e-g` could be 2d shapes engine (draw lines, circles, triangles, rects, bitmaps, ...). But it tries to do everything (drivers, fonts) and that make things more complex & tight coupled than needed.

jamwaffles, 2019-09-03:
I'm closing this issue as I think a solution has been reached - a separate crate for more complex fonts.

closed 2019-09-03 by jamwaffles