embedded-graphics/embedded-graphics #32: Why is translate always making a new object?

Caemor, 2018-05-29:
Hi, 

```Rust
    fn translate(&self, by: Coord) -> Self {
        Self {
            offset: (self.offset.0 + by.0, self.offset.1 + by.1),
            ..*self
        }
    }
```

While looking into the project i got a bit confused by the part above.
Why does translate always make a modified "clone" of the object instead of modifying itself?
I would have expected sth like the following, where you need to explicitly clone the object if you still need it:

```Rust
    fn translate(&mut self, by: Coord) {
        self.offset = (self.offset.0 + by.0, self.offset.1 + by.1)
    }
```

jamwaffles, 2018-05-29:
Thanks for reporting this. The idea behind the current behaviour was to allow things like particle engines to create translated clones of a single graphical element. It also fits with Rust's immutable-by-default idioms, but you're right it's confusing to call `thing.translate()` and not see any effect on the object you're moving. 

I'll make translations mutating in the next release. The only caveat is the programmer must add `mut` to objects they're transforming, but at least it's obvious what's happening then.

jamwaffles, 2018-07-23:
Thinking about this further (many months later... sorry), I've decided to add a [`transform_mut()` method](https://github.com/jamwaffles/embedded-graphics/pull/35). This gives users the freedom to either mutate in place, or get copies with the current behaviour.

PR #35 (2018-07-23): Add transform_mut method to objects
This allows one to transform an object in place instead of `transform()` returning a copied object.

There's also some `rustfmt` noise.

Closes #32
commit 5ed9d536 (2018-07-23): Add transform_mut method to objects

This allows one to transform an object in place instead of transform()
returning a copied object.

Closes #32
closed 2018-07-23 by jamwaffles