embedded-graphics/embedded-graphics #33: Colour support

MabezDev, 2018-07-22:
Hi all,

I've been working on a [SSD1351 driver crate](https://github.com/MabezDev/ssd1351) based on the work done with the ssd1306. I am at the point where I want to start integrating embedded_graphics traits for more advanced graphic work. The SSD1351 being a 16bit colour display driver the current u8 for colour is not going to work, and in the future this crate should probably be able to support even higher colour depths.

Just wanted to kick off some discussion on the matter :)

jamwaffles, 2018-07-22:
Hey, thanks for commenting! I saw your driver added to the awesome embedded Rust list, congrats :). Colour support is definitely on the roadmap, I just haven't needed it personally yet. I was thinking of defining an associated type for the Drawing trait, although I don't know if that would be enough. Do you have an idea of how you'd want this problem approached? Ideally it would support anything from single bit colour all the way up to 24 bit (and further?) for any type of display.

jamwaffles, 2018-07-25:
> How would an colour assiociated type work with the current Drawing trait?

Not sure yet! Probably some type/struct exported from the crate that the interface can use. It would then be up to the display driver implementors to convert the `Color` type (or whatever) that's returned from a pixel iterator into something the display itself can understand. I'll do some experimenting when I've got a spare mo.

MabezDev, 2018-07-31:
Hey, I've been a bit busy the last week but I have some time this week so based on the what you've said so far I'd like to get a rough implementation working. I plan on changing the Drawing trait to looking something like this
```rust
pub trait Drawing {
    /// Generic Color type
    type Color;

    /// Draw an object from an iterator over its pixels
    fn draw<T>(&mut self, item_pixels: T)
    where
        T: Iterator<Item = (drawable::Pixel, Self::Color)>;
}
```
where Pixel is basically an coord, with no colour information.

And the implementation of the new trait would look something like this
```rust
impl<DI> Drawing for GraphicsMode<DI>
where
    DI: DisplayInterface,
{
    // the SSD1351 uses 16bit colours
    type Color = u16;

    fn draw<T>(&mut self, item_pixels: T)
    where
        T: Iterator<Item = (drawable::Pixel, Self::Color)>,
    {
        for (pos, color) in item_pixels {
            self.set_pixel(pos.0, pos.1 color);
        }
    }
}
```
I'll have a stab and see where I get.

jamwaffles, 2018-08-01:
That's exactly what I was thinking. The driver implementor (in your case the `ssd1351` crate) should  also export that `Color` type so that users of the crate can use it their own code as well I think. Doesn't make much sense for a `u16`, but for richer `Color` types it would. Looking forward to seeing what you come up with!

MabezDev, 2018-08-01:
After an hour or two of playing, I have a functioning implementation. I ended up changing the draw method in the drawing trait to this
```rust
pub trait Drawing {
    /// Generic Color type - implemented by driver crates
    type Color;

    /// Draw an object from an iterator over its pixels
    fn draw<T>(&mut self, item_pixels: T, color: Self::Color)
    where
        T: Iterator<Item = drawable::Pixel>;
}
```
Moving the colour declaration here makes the u8 color in Pixel redundant, so I have changed Pixel type definition to
```rust
pub type On = bool;

/// A single pixel
pub type Pixel = (UnsignedCoord, On);
```

The primitives have had their color field removed, and thier iterators now return the new pixel type eg `Some((coord.to_unsigned(), true))`.

Finally the implementation of the new drawing trait looks like this
```rust
impl<DI> Drawing for GraphicsMode<DI>
where
    DI: DisplayInterface,
{
    /// Exported Color type from ssd1351 crate
    type Color = Color;

    fn draw<T>(&mut self, item_pixels: T, color: Self::Color)
    where
        T: Iterator<Item = drawable::Pixel>
    {
        for (pos, set) in item_pixels {
            if set {
                self.set_pixel(pos.0, pos.1, color);
            }
        }
    }
}
```

What do you think?

jamwaffles, 2018-08-02:
That's a good start, but I think the pixel value _must_ be given by the created object. `draw()` should support drawing objects that have changing colours of their yielded pixels (think a computed gradient or greyscale/colour images, etc) which means the object is responsible for tracking that pixel colour.

Can you open a PR with your changes so we can track progress? It's easier than pasting code blocks into issue comments I think :)

jamwaffles, 2018-08-02:
I started an experiment and it turned into a full PR at #41. Sorry if I stepped on your toes. Let me know what you think in that thread! Hopefully it solves your problem in an elegant-ish way.

MabezDev, 2018-08-02:
Ah, I've just submitted mine too, I think I might have overcomplicated my solution slightly, lol.

closed 2018-08-03 by jamwaffles