embedded-graphics/embedded-graphics #55: Strange Behaviour with 0.4.2 from crates.io

Caemor, 2018-10-18:
After testing my crate for a few hours I realised that crates.io 0.4.2 != embedded-graphics master 0.4.2.
Somehow the patch hasn't arrived on crates.io even though there was the new release.

Easy way to test it:
Go to the simulator and use the embedded-graphics version from crates.io instead of the local one and then run `cargo run --example fonts`

jamwaffles, 2018-10-18:
That's very confusing, my apologies for wasting your time. I've yanked 0.4.2 which didn't include the changes and published 0.4.3 which I've checked _does_ contain the font changes. Can you give 0.4.3 a test and report back?

Caemor, 2018-10-19:
No apologies needed, I am very happy with your fast responses :heart: 

It looks good now :+1:

closed 2018-10-19 by Caemor

jamwaffles, 2018-10-19:
Awesome!