embedded-graphics/embedded-graphics #65: Use fonts from u8g2

jamwaffles, 2019-02-13:
[U8g2](https://github.com/olikraus/u8g2) has a pretty big collection of fonts. It would be great to have support for them in embedded-graphics. This issue is a tracking issue of sorts to define what should be done as a _minimum first step_ towards u8g2 font support.

Any code that aims to close this issue should:

* Reuse as much u8g2 tooling as possible
* Treat the u8g2 repository as the source of truth for fonts (no mirrors, etc)
* Create embedded-graphics compatible font maps like the [`.raw` files here](https://github.com/jamwaffles/embedded-graphics/tree/master/embedded-graphics/data).

The first implementation will be non-ideal, but will show where things can be fixed or improved with more optimisations and tooling in the future.

puzrin, 2019-05-09:
I think this can be closed as outdated. See https://github.com/jamwaffles/embedded-graphics/issues/28#issuecomment-490886600 for more fresh solution.

dutchmartin, 2019-08-22:
I like to contribute and make more fonts available, since I like to use a bigger font in my personal project. However, I still have no clue where to start implementing a new font.
Can one of you please give me a clear explanation on what steps to take so I can start working on implementing new fonts.
@jamwaffles @puzrin

jamwaffles, 2019-08-22:
Hi, thank you for contributing! The fonts story is pretty poor right now, but the current bitmap fonts use a black and white font image ([example](https://github.com/jamwaffles/embedded-graphics/blob/master/embedded-graphics/data/font12x16.png)) that's converted to a 1BPP (8 pixel values per byte) image using this Imagemagick command (untested, YMMV):

```bash
convert font.png -depth 1 gray:"font.raw"
```

You can then use the [`FontBuilder`](https://docs.rs/embedded-graphics/0.6.0-alpha.1/embedded_graphics/fonts/font_builder/struct.FontBuilder.html) to build a font that samples characters from the image. I'd suggest looking at the existing fonts like [`Font6x8`](https://github.com/jamwaffles/embedded-graphics/blob/master/embedded-graphics/src/fonts/font6x8.rs) for a starting point. Hope that helps!

puzrin, 2019-08-22:
@dutchmartin I'd suggest you to carefully inspect this project & spec https://github.com/littlevgl/lv_font_conv and use it's binary format. It implements storage level with care about all known issues in embedded.

Better thing can be only direct TTF rendering (but that increase RAM requirements significantly).

jamwaffles, 2019-08-22:
I'm gonna close this issue as it's pretty stale. @dutchmartin please open a PR or another issue if you need more help and I'll do what I can.

closed 2019-08-22 by jamwaffles

jamwaffles, 2019-08-22:
Please see https://github.com/jamwaffles/embedded-graphics/issues/28#issuecomment-524076670 for a possible future direction.