embedded-graphics/embedded-graphics #70: Fix Image16BPP byte ordering

jamwaffles, 2019-03-03:
The current implementation is incorrect; two bytes `[ 0xcc, 0x00 ]` that represent a single 16 bit pixel value are currently interpreted as `0x00cc_u16` which is incorrect; it should be `0xcc00`.

Failing test case:

```rust
let image: Image16BPP<PixelColorU16> = Image16BPP::new(
    &[
        0xff, 0x00, 0x00, 0x00, 0xbb, 0x00, //
        0x00, 0x00, 0xcc, 0x00, 0x00, 0x00, //
        0xee, 0x00, 0x00, 0x00, 0xaa, 0x00,
    ],
    3,
    3,
)
.translate(Coord::new(-1, -1));
let mut it = image.into_iter();

assert_eq!(
    it.next(),
    Some(Pixel(UnsignedCoord::new(0, 0), 0xcc00_u16.into()))
);
assert_eq!(
    it.next(),
    Some(Pixel(UnsignedCoord::new(1, 0), 0x0000_u16.into()))
);
assert_eq!(
    it.next(),
    Some(Pixel(UnsignedCoord::new(0, 1), 0x0000_u16.into()))
);
assert_eq!(
    it.next(),
    Some(Pixel(UnsignedCoord::new(1, 1), 0xaa00_u16.into()))
);
```

sajattack, 2019-06-08:
Would this issue explain why I'm getting a blue ferris?

jamwaffles, 2019-06-08:
Could be! What display are you using? I got Ferris in his normal orange colour on an SSD1331 like [here](https://crates.io/crates/ssd1331). Perhaps your display is getting the bytes backwards or your image is encoded in the wrong endianness? It could also very likely be this library just being plain wrong 😂.

I'm not quite sure whether `[ 0xff, 0x00 ]` should turn into `0xff00` or `0x00ff`, so if you or anyone reading this has any insight into the correct conversion behaviour, I'd appreciate some correction!

sajattack, 2019-06-08:
I'm using an ST7735. I copied your `ferris.raw` from your SSD1331 repo. I thought colours were inverted maybe but white and black are correct. I've tried byte-swapping on my end with no luck.

jamwaffles, 2019-06-08:
I have an ST7735S (not sure if the `S` is significant) lying around. I'll see if I can figure out what's going on.

sajattack, 2019-06-08:
I'm testing with the 0.96" 80x160 model.

jamwaffles, 2019-06-08:
After some pain with the stm32f103 crate I've managed to repro:

[image]

It looks like the ordering got swapped from RGB to BGR. The datasheet for the ST7735 seems to have a config option for swapping the order, so could you try that and see if it resolves your issue for now?

There's an emerging need for a better way of handling pixel conversions in embedded_graphics, so I'll have a think about how best to do that.

jamwaffles, 2019-06-08:
Interestingly, if I reverse the colour order to `BGR` with this magic incantation:

```bash
convert ferris.png -flip -type truecolor -define bmp:subtype=BGR565 -depth 16 -strip ferris.bmp && tail -c 11008 ferris.bmp > ferris-backwards.raw
```

I get this random garbage:

[image]

I'd be interested to see if changing the display colour ordering fixes the issue!

sajattack, 2019-06-08:
I set the RGB/BGR thing in MADCTL and ferris is still blue somehow.

sajattack, 2019-06-08:
Yay! I got a properly coloured Ferris! Now I have to see how to accomodate all the different ST7735's out there in my library.

jamwaffles, 2019-06-08:
Sweet! What was wrong in the end? How did you fix it? I have an ST7735 and an ST7789 for testing if you need it - maybe open an issue/PR in the driver repo and tag me in it?

I'll open a PR in the driver repo when I'm next at my Linux machine. It's hugely messy and conflicts with the atsamd crate a bit but it should be a good starting point.

sajattack, 2019-06-08:
It was BGR like you said. I had to set the RGB bit of MADCTL to 1.

closed 2019-06-16 by jamwaffles