embedded-graphics/embedded-graphics #77: Polygon support

jamwaffles, 2019-03-12:
It should be possible to create an open and closed line loop. Closed line loops should support having a fill applied.

rfuest, 2019-09-01:
Should this also replace the current `Triangle` primitive?

jamwaffles, 2019-09-02:
Yeah probably, although it might be worth keeping a thin `Triangle` wrapper so it's not removed from the crate's API (one less breaking change - yay!), and as a convenience method for those wanting a simple graphics API.

When filled polygons are a thing, I think they'll be drawn with filled triangles anyway, so there'll be a lot of overlap there.

What would the polygon API look like? Something simple like this:

```rust
impl Polygon {
    fn new(points: &[Point]) -> Self {
        // ...
    }
}
```

This would require `points` to live for as long as the `Polygon` does which is fine, just a departure from the other primitives. Images also take references so it's not so bad. We could use const generics here I think, but that requires nightly which is unfortunate. Tracking issue: rust-lang/rust#44580.

rfuest, 2019-09-02:
Const generics would be useful in some places but it doesn't seem they will be stable soon. But using a slice for the points is OK.

How would this API handle open vs. closed polygons?

jamwaffles, 2019-09-10:
> How would this API handle open vs. closed polygons?

Maybe with a method on `Polygon`?:

```rust
let points = [Point::new(1, 2), Point::new(3, 4), Point::new(5, 6), Point::new(7, 8)];

let poly = new Polygon(&points).closed(true);
let poly = new Polygon(&points).closed(false);
```

rfuest, 2019-09-10:
Did you write any C++ recently? :smile:
But regardless of the syntax this would be an option. Another option would be to have two constructors:
```rust
let poly = Polygon::new_closed(&points);
let poly = Polygon::new_open(&points);
```
Or to different structs, like in SVG:
```rust
let poly = Polygon::new(&points);
let poly = Polyline::new(&points);
```

jamwaffles, 2019-09-10:
> Did you write any C++ recently? 

Worse! 😬 Javascript! 🤢 😂 

> Or to different structs, like in SVG:

That makes a lot of sense to me, particularly as `Polyline` wouldn't support a fill, but `Polygon` would. Let's go with that.

PR #292 (2020-05-04): Add Polyline primitive
Hi! Thank you for helping out with Embedded Graphics development! Please:

- [x] Check that you've added passing tests and documentation
- [x] Add a simulator example(s) where applicable
- [x] Add a `CHANGELOG.md` entry in the **Unreleased** section under the appropriate heading (**Added**, **Fixed**, etc) and appropriate crate (`embedded-graphics`, `simulator`, `tinytga`, `tinybmp`) if your changes affect the **public API**
- [x] Run `rustfmt` on the project
- [x] Run `./build.sh` (Linux/macOS only) and make sure it passes. If you use Windows, check that CI passes once  [... cut]
closed 2020-05-04 by jamwaffles