embedded-graphics/embedded-graphics #80: Drawable Trait

rfuest, 2019-03-13:
The documentation of the Drawable trait states: 
```rust
/// Marks an object as "drawable". Must be implemented for all graphics objects
pub trait Drawable {}
```

Is this still true? The current code doesn't seem to use this trait and the only references (other than empty implementations) are comments referring to a non existent draw() method.

jamwaffles, 2019-03-13:
All fonts, images and primitives implement `Drawable`. It was originally my intent to use it as part of the `Drawing` trait (implemented by display drivers) as a fix to the unergonomic `.into_iter()` calls that have to be plastered everywhere. I'd like to be able to use `Drawable` to be able to pass objects into display drivers without having to call `.into_iter()` at some point. I just haven't picked that up yet. Does that make sense?

rfuest, 2019-03-13:
OK, thanks for the explanation.

Did you consider changing `Iterator` to `IntoIterator` in the `Drawing` trait? This gets rid of almost all `.into_iter()` calls in the samples and replaces them with `&`. But I'm not very familiar with the code base, so this might blow up at a different place.

jamwaffles, 2019-03-24:
I think I did, but that means that chained iterators can't be passed in. They're broken at the moment IIRC, but it should be possible to be able to compose drawing operations using `Iterator::chain`. This means that `draw()` needs to accept `Iterator` for this case, as far as I know.

This crate is still quite young, so there might be limitations with the `Iterator`-based chaining approach, or better/more ergonomic ways to do it. I think it's worth exploring using vanilla iterators for now though so any pain points become clear.

rfuest, 2019-03-24:
If `draw()` is changed to `IntoIterator` it will still accept `Iterator` because a noop `IntoIterator` is implemented for all `Iterator`s.

Chaining should continue to work, but would still require an `into_iter` call for the first element in the chain. `Iterator::chain` also uses `IntoIterator` for its parameter, so anything passed directly into `draw` should also work in a chain.

jamwaffles, 2019-06-16:
Closed by #88

PR #88 (2019-06-10): Require IntoIterator trait instead of Iterator
This PR is spurred by https://github.com/jamwaffles/embedded-graphics/issues/80#issuecomment-475977383 and should make e_g a little more ergonomic to use by removing the `.into_iter()` call requirement.

This is a breaking change for e_g implementors, so will require a semver minor version bump.

Closes #80 
Closes #74
closed 2019-06-16 by jamwaffles