embedded-graphics/embedded-graphics #85: Change iteration order

etrombly, 2019-04-23:
For epaper displays you can set the draw order to X ascending than Y ascending or Y than X for a region, but if you iterate over a rendered string from this library it gives X to Y for each character, then resets the Y range for the next character. If it went Y to X you could iterate directly to the device without a buffer. So basically going left to right rather than top to bottom for each character. I'm trying to use a chip with 20Kb of SRAM, so not enough for a framebuffer.

etrombly, 2019-05-02:
Realized I was off by 8 on my buffer size (bits vs bytes). So this is lower priority, but still would be nice to have.

jamwaffles, 2019-05-04:
Not the most efficient way of doing it, but can you call `.into_iter().map()` and transform the coordinates that way? You can use [`.size()`](https://docs.rs/embedded-graphics/0.4.7/embedded_graphics/fonts/font_builder/struct.FontBuilder.html#method.size) to get the font height.

closed 2019-09-05 by jamwaffles