embedded-graphics/embedded-graphics #100: Migrate docs to intra-docs links

jamwaffles, 2019-06-11:
Blocked on rust-lang/rust#43466. When released to stable, the docs should be updated to use the less error-prone link syntax described in that PR/RFC.

bugadani, 2020-08-01:
Just a note: intra-doc links are targeted for 1.47 [stabilization issue](https://github.com/rust-lang/rust/pull/74430). What's the plan for MSRV, raise it to 1.47 then, or add a note that docs can only be build with >= 1.47?

jamwaffles, 2020-08-03:
I think that might be a good idea, although I think that's quite a leap ahead from the rest of the ecosystem. IIRC embedded-hal or a similar crate defines 1.35(?) as it's MSRV.

There are more `const` improvements coming in 1.46 as well, so a bump to 1.47 would open up quite a few doors and would be worth it IMO.

jamwaffles, 2020-11-16:
This is set to be stabilised in [1.48.0](https://github.com/rust-lang/rust/blob/stable/RELEASES.md#version-1480-2020-11-19)! At last!

rfuest, 2020-11-19:
Should we take this as a reason to update our MSRV now or wait with the migration until we have another reason to raise the MSRV?

bugadani, 2020-11-19:
> Should we take this as a reason to update our MSRV now or wait with the migration until we have another reason to raise the MSRV?

I think the docs warrant a note about the version but nothing stricter.

jamwaffles, 2020-11-19:
There was a discussion on Matrix the other day about MSRVs. People don't appear to like them very much :grin: but also suggested something like stable - 2 (e.g., stable 1.50, target 1.48). We could go that route, or keep to only upgrading the MSRV when there's a strong desire to.

rfuest, 2020-11-19:
> ... ,or keep to only upgrading the MSRV when there's a strong desire to.

That was the intent of my question. Is there a strong desire because of the intra-doc links?

How would "stable - 2" work for minor releases? A MSRV change is considered to be a breaking change. Would the same "stable - 2" be used for all releases with the same major version?

jamwaffles, 2020-11-19:
> A MSRV change is considered to be a breaking change.

An excellent point I hadn't considered. 

> Is there a strong desire because of the intra-doc links?

No, but it adds another item to the growing list of nice-to-haves to upgrade. 

If we continue the MSRV discussion, I suggest we do it in another ticket, but I'll finish here by saying: let's keep the current 1.40 for 0.7, and think about changing the MSRV for 0.8. The issues we have at the moment for being on an older version of Rust are mostly paper cuts, so there's no very strong reason to upgrade for the e-g 0.7 cycle.

jamwaffles, 2020-11-26:
This may be handy during migration: https://github.com/poliorcetics/cargo-intraconv

jamwaffles, 2022-03-27:
You're too kind 😊 

2019, wow. Feels like a decade ago.

closed 2022-03-27 by jamwaffles