embedded-graphics/embedded-graphics #121: Color types and images

rfuest, 2019-06-16:
I just reviewed the changes made to the `PixelColor` types in e-g 0.5.0 and would like to suggest some changes.

In response to issue #89 you removed the `PixelColorUx` types and now use the vanilla `ux` types instead. I would suggest to also remove the `PixelColor` trait implementations for these types, because they don't have an unambiguous interpretation. An `u16` might for example be a `Rgb565` or `Bgr565` (see issue #70). The `From<...>` implementations in `Rgb565` have similar problems, like assuming big endian or that an `u8` is representing luminance.

Getting rid of these generic `From<...>` implementations would break the current implementation of fonts and images. But I think this could be fixed by changing the way images work. Currently the image types take a `C` type parameter which specifies the `PixelType` expected by the used display and internally convert the pixel data to the appropriate type. In my opinion it would make more sense to specify the type of stored data. The `Image1BPP` for example wouldn't need a type parameter any more and always use a `BinaryColor` color type. E-G could provide a default conversion from this type to black and white in the other color formats to keep the current behavior intact. But the user would also be able to explicitly use `map` to convert the two values to different colors or make one of the colors transparent by using `filter`.

My suggestions would:
- Remove `PixelColor` for `ux` types
- Add more `PixelColor` implementations for common formats like `Rgb565` , `Bgr565`, `Rgb888`, `Bgr888`, `Luma8`, `BinaryColor`
- Provide conversion functions between different formats
- Change images and fonts (see explanation above)

If you are interested in these changes I could give it a try and prepare a PR.

jamwaffles, 2019-06-16:
Thanks for this, its useful to see e-g from other peoples' viewpoints :slightly_smiling_face:

>  I would suggest to also remove the PixelColor trait implementations for these types, because they don't have an unambiguous interpretation.

Hm, that sounds alright to me. I'd like to see what the ergonomics are like before committing to this though. 

> An u16 might for example be a Rgb565 or Bgr565 (see issue #70).

Right, but I think in that case one would use the `RGB565` or (currently non-existent) `BGR565` pixel types to make it explicit. I think that might need a bound on `Into<C>` on all the iterators in this crate so they can internally call `.into()`, but that's doable. Wouldn't that also solve the problem? 

>  In my opinion it would make more sense to specify the type of stored data.

I think this becomes an issue with 1BPP images but with 1 byte stored per pixel (only storing `0x00` or `0x01`). Not sure how widespread this is in the real world, but I think I prefer leaving the input format up to the developer and let the iter impl decide how to convert that into a pixel.

To comment on your suggestions respectively:

* I'd certainly be open to seeing what that looks like!
* More is always better :slightly_smiling_face: I think it would be prudent to start with only a couple of implementations for now to see if they work out, otherwise it'll be _N_ times the refactoring effort.
* Would the conversion functions just be `From<>` impls or methods on structs? Do you have a preference?
* I'm not _quite_ following, so if you're up for a PR could we see how it looks in code?

I'd be happy to look at a PR for all this. Thank you for all the feedback!

jamwaffles, 2019-09-08:
I think this was closed a long time ago by #114 and/or #127.

closed 2019-09-08 by jamwaffles