embedded-graphics/embedded-graphics #123: SizedDrawing trait

rfuest, 2019-06-16:
I'm not sure I understand the way `SizedDrawing` is supposed to work. Doesn't this require special `Drawable` implementations, which make sure the data is continuous and in the correct iteration order?

The example in the documentation for `SizedDrawing` does only output 56 pixels to the display, but the bounding box contains 400 pixels. This will leave the display in an illegal state after the circle is drawn.

jamwaffles, 2019-06-16:
> Doesn't this require special Drawable implementations, which make sure the data is continuous and in the correct iteration order?

Gosh I should've thought of that. Now that you mention it, yes it does but that wasn't the original intent. It should allow you to leverage the automatic row wrapping in display drivers that support sub-updates, but the example in the docs doesn't show that at all.

I think driver implementations have two options:

1. Dynamically allocate a correctly sized pixel buffer for the partial update and fill that (bad, because dynamic alloc)
2. Build and send chunks of _N_ contiguous bytes to the display. Any pixels not set by the passed-in iterator would be filled with a default value.

I like the second better, but it's a lot more complex. Maybe it could be helped with an iterator adapter that fills in transparent pixels? Do you have any thoughts on the direction you'd like this to go?

rfuest, 2019-06-17:
1. This would be much easier but, as you said, doesn't really work for memory constrained embedded devices. And if you have enough memory for this a full framebuffer might be a better solution.
2. How would you determine which N bytes to send next (e.g. for a slanted line)?

Maybe it would be a good first step to solve this for iterators over a rectangular areas like images, fonts with background color and filled rectangles. But this would require that the display is able to influence the iteration order, because it might fill its memory row or column first.

jamwaffles, 2019-06-18:
> How would you determine which N bytes to send next (e.g. for a slanted line)?

I think in this case a code example would speak a thousand words :grin:. I have an idea in mind but it's difficult to type out in `eng-lang` :joy: so I'll prep a proof of concept PR.

> But this would require that the display is able to influence the iteration order

I think this should happen in the driver impl of `SizedDrawing`. It likely requires calls to `.map()` on the iterator which adds a little bit of overhead, however it keeps e-g agnostic with a "typical" left-to-right top-to-bottom iterator which is easier to reason about IMO.

rfuest, 2019-06-18:
> I think this should happen in the driver impl of `SizedDrawing`. It likely requires calls to `.map()` on the iterator which adds a little bit of overhead, however it keeps e-g agnostic with a "typical" left-to-right top-to-bottom iterator which is easier to reason about IMO.

Ok, most modern controllers should support changing the data input direction anyway.

rfuest, 2019-09-10:
Is there any progress on this? Otherwise I would suggest to remove the broken `SizedDrawing` for 0.6 and add a working solution for this problem later. Maybe this could also be implemented as another method in the `DrawTarget` trait suggested in #175.

jamwaffles, 2019-09-10:
Hurgh yeah, that was a half baked idea. AFAIK no driver implements it, so yes let's remove it and come up with a better solution.

I think it should probably be a different trait - some displays might not be able to implement it due to e.g. inability to set a sub-area to update.

PR #180 (2019-09-14): Remove SizedDrawing trait
This closes #123 by removing the `SizedDrawing` trait.

@jamwaffles: Could you open a new issue to discuss how this might be implemented in a better way?
closed 2019-09-14 by rfuest