embedded-graphics/embedded-graphics #129: Macro-using this crate breaks panic

althonos, 2019-06-23:
Hi !

The `line` macro in this crate conflicts with the [`std::line`](https://doc.rust-lang.org/1.2.0/std/macro.line!.html) macro, which is invoked by `panic` for instance. This means that adding a 
```rust
#[macro_use]
extern crate embedded_graphics;
```
in a program will prevent usage of `panic`.

jamwaffles, 2019-06-23:
Thanks for the bug report! That's an interesting conflict. Are you using the 2018 edition? The better (IMO) way of importing macros is to `use embedded_graphics::line`, however I will get a proper fix out for this soon. Unless you can think of a better way to do it, I think prefixing all the shape macros with a `g` (like `gcircle`, `gline`) would be the best way of doing it.

althonos, 2019-06-23:
Yes I am, so it's okay and I can cherry-pick the imports the way you describe (on Rust 2015 you could do `#[macro_use(circle)]`), but I had a hard time finding the cause when all of a sudden all `panic!` broke.

I agree the right solution is most likely to use a prefix !

jamwaffles, 2019-06-25:
I've had a moment to think about this some more and I've got a couple of items:

1. Can you provide a failing test case or a link to some source code so I can build a test case to run changes against?
2. How do you feel about using `eg` as the prefix like `egcircle`? It fits with the Embedded Graphics name but keeps the macros short.

jamwaffles, 2019-06-25:
Scratch point 2.; it looks and sounds uncomfortably close to the word "erect" when used with the `rect` macro for my liking. I think `grect`, `gline`, `gcircle`, etc are a good idea.

jamwaffles, 2019-06-25:
After a bit more experimentation I've come up with #130. Please give it a quick look and see if there's anything you do or don't agree with :grin:

althonos, 2019-06-25:
Shortest test case that fails at compile time would be:
```rust
#[macro_use]
extern crate embedded_graphics;

fn main() {
    panic!()
}
```
which errors:
```rust
error: unexpected end of macro invocation
 --> script.rs:5:5
  |
5 |     panic!()
  |     ^^^^^^^^ missing tokens in macro arguments
  |
  = note: this error originates in a macro outside of the current crate (in Nightly builds, run with -Z external-macro-backtrace for more info)
```

jamwaffles, 2019-06-26:
This is now released in 0.5.1, please give it a test and re-open if you find any issues!

PR #130 (2019-06-26): Rename primitives macros
This primarily fixes #129 so that the `line!()` macro provided by std doesn't conflict with the `line!()` macro from Embedded Graphics. The macros are now renamed as such:

* `line` -> `egline`
* `triangle` -> `egtriangle`
* `rect` -> `egrectangle`
* `circle` -> `egcircle`

Note that `rect` is also now called `rectangle` for consistency with the other macros that use the shape's full name.
closed 2019-06-26 by jamwaffles