embedded-graphics/embedded-graphics #134: emebedded-graphics logo

rfuest, 2019-07-21:
I saw that you added a temporary logo to the repository, so I decided to open this issue to collect logo ideas. Having a nice logo would be really useful for a create which deals with graphics.

This is my contribution, which I created a while ago to test some changes. It is only programmer art but maybe a little less temporary. The logo is just an stylized e and g and is designed to also be usable at small pixel sizes.

Full color with text:
[image]
B/W on OLED display:
[image]

jamwaffles, 2019-07-21:
😱 so much better than my attempt! I'll post this on Twitter and see if I can get some more input. Can you think of any other places to post this for more embedded Rustaceans to see?

jamwaffles, 2019-09-03:
I just had a play around with the OLED logo example and tried it out on docs.rs. Unfortunately the image is too large to fit in the space and scale properly (notice the fuzz):

[image]

For docs.rs I quite like the transparent background/black pixels idea. Could you create a version like that with a max height of 100px to use on docs.rs? The "full colour with text" version could go at the top of the readme.

rfuest, 2019-09-06:
The 2 color version was just a very quick test and could still be tweaked a bit, but here are some versions of that logo, for testing:

Unscaled:
[image]
Scaled 3x:
[image]
Scaled 3x with pixel grid (the 2:1 ratio between gap and pixel size make the pixels look too tiny IMO):
[image]
Scaled 3x with pixel grid and light gray background around pixels (looks better on the actual gray background than on white):
[image]

>  The "full colour with text" version could go at the top of the readme.

If we should decide to use that logo I would like some feedback on the color scheme and the font.
The orange and dark gray was primarily picked because most of my recent projects used a blue color scheme and I wanted to use something different.
The font was chosen because it's relatively basic and matches the stroke width in the logo. But there are lots of other options and also different variants of the text itself: "EMBEDDED GRAPHICS", "EMBEDDED-GRAPHICS", "embedded graphics", "embedded-graphics".

jamwaffles, 2019-11-21:
Shall we just go with the icon version for now and leave the colouring/casing question for later? I think using this image:

[image]

In the README and docs logo would look really good, even without the `Embedded Graphics` text. If you're ok with that, I'll change the icon and push straight to master - required so I can permalink to the image by commit hash in the docs.

jamwaffles, 2019-11-27:
Testing that version locally doesn't look great in the dark theme:

[image]

It would also be good to expand the logo to the full 100px height if we can. I think it's a little too small and gets lost in the page as is.

rfuest, 2019-11-28:
I like the clean look of just having the logo on the solid background color but finding one color that works equally well on both background colors is hard (impossible?).

Adding a background to the logo would solve this problem. For example with an OLED like color scheme:
[image]

jamwaffles, 2019-11-28:
Yeah! Background looks great! Can you post that image on its own? I'll put that in the docs asap. If you could export it at 2x so it looks good on HiDPI displays as well that would be awesome.

As for the "horizontal" logo with icon and text for the top of the readme, what about the logo to the left and "EMBEDDED GRAPHICS" to the right as per your first post, but in the OLED Blue theme colour? Could also use the OLED background and have a lighter-text version for dark backgrounds. 

What software are you using for this? Might be worth preparing a PR with the GIMP XCF or whatever in it. Would be good to have the original asset tracked in the repo :slightly_smiling_face:

rfuest, 2019-11-28:
> Yeah! Background looks great! Can you post that image on its own? I'll put that in the docs asap. If you could export it at 2x so it looks good on HiDPI displays as well that would be awesome.

Sure, the docs logo currently only exists as this mockup, but I'll prepare a finished PNG.
 
> As for the "horizontal" logo with icon and text for the top of the readme, what about the logo to the left and "EMBEDDED GRAPHICS" to the right as per your first post, but in the OLED Blue theme colour? Could also use the OLED background and have a lighter-text version for dark backgrounds.

I can try that.

> What software are you using for this? Might be worth preparing a PR with the GIMP XCF or whatever in it. Would be good to have the original asset tracked in the repo slightly_smiling_face

Some of the original asses are in a propitiatory format at the moment, but I'll convert them to SVG and PNG files so that they can easily be used in GIMP and Inkscape.

rfuest, 2019-11-29:
I've played with the logo some more and added a hexagonal background and some subtle shading:
[image]

The logo would also work well in the docs:
[image]

I would prefer to use the same logo in the docs and use the pixel version in the "pretty picture" in the docs and README instead. But I can also prepare an pixel version for the docs if you like.

jamwaffles, 2019-11-29:
Hey that looks a bunch better than the pixel version! Yeah I like the consistency of keeping the same logo icon everywhere. Can you open a PR with these assets? We can get the docs/readme pointing to them then. There's already an `assets/logo.png` waiting to be overwritten 🙂

rfuest, 2019-11-29:
The PR for the logo itself is opened. I'm not quite sure about the look of the text and font yet, so this will be added in a another PR.

jamwaffles, 2020-03-18:
In the interests of reducing the open issue count a bit, could this be closed? I don't know if you want to make any more changes relevant to this issue, or whether we should open new ones down the road.

rfuest, 2020-03-18:
The issue was left open because I wanted to look into the version with text. But IMO we currently don't need that version and we can open another issue if this will be necessary in the future.

closed 2020-03-18 by rfuest