embedded-graphics/embedded-graphics #135: Implement Gray16 for SSD1351 and other 16bit monochrome displays

jamwaffles, 2019-07-23:
`Gray8` already exists, but so should `Gray16`. Conversion  from `Gray8` to `Gray16` should be possible. Perhaps the opposite should be too, with a warning about loss of fidelity?

rfuest, 2019-07-23:
Do you have an example of a monochrome display with SSD1351 controller? I haven't used this IC myself but I thought that it was only used for color OLEDs.

jamwaffles, 2019-07-23:
Erf, too many SSD13* chips! I meant the [SSD1322](https://github.com/edarc/ssd1322), but that doesn't use e-g. Looks like it isn't even 16bit colour so I don't know what I was on about. Do you think it's still worth having though? Maybe we see if there's any need for it and add it as necessary.

rfuest, 2019-07-24:
Ah, OK, so the SSD1322 uses 16 shades of gray (4bpp) and not 16bpp.

I don't think that we need a `Gray16` type because color channels with > 8 bits aren't often used in embedded applications.

But having a `Gray4` type might be useful if there are multiple controllers which use this.

jamwaffles, 2019-07-24:
> But having a Gray4 type might be useful if there are multiple controllers which use this.

True, although `Gray8` will provide a migration path from e-g 0.4/0.5 to 0.6 for 4bpp displays so we can add `Gray4` etc later if required. I'll close this issue for now.

closed 2019-07-24 by jamwaffles