embedded-graphics/embedded-graphics #138: Tracking issue: new colour system

jamwaffles, 2019-07-23:
This is a tracking issue for the colour features merged in #127. Comment here with issues found during implementation in drivers/projects.

jamwaffles, 2019-07-23:
I can't remember if this was discussed by myself and @rfuest, but should `u8` be convertible to `BinaryColor`? Where `0` -> `BinaryColor::Off` and `> 0` -> `BinaryColor::On`? I have quite lot of example code in the SSD1306 crate that uses `1u8.into()` which I can't decide whether to keep or not.

rfuest, 2019-08-04:
I think that `From<u8> for BinaryColor` shouldn't be implemented, because it is ambiguous. The conversion `u8` to `bool` would be a similar conversion, which is not implemented in rust and rust users are used to explicitly convert an integer to `bool` by using `i != 0`.

But `From<bool> for BinaryColor` might be a useful addition. What do you think?

jamwaffles, 2019-08-05:
Good points!

> But From<bool> for BinaryColor might be a useful addition. What do you think?

That makes loads more sense, yeah let's do that

jamwaffles, 2019-09-03:
Re #165, I think the docs could use a bit of clarifying around conversion behaviours and some common use cases.

jamwaffles, 2019-09-09:
I'm going to close this issue in preference of just creating individual new issues against the 0.6.0 project to discuss findings from using the new colours.

closed 2019-09-09 by jamwaffles