embedded-graphics/embedded-graphics #139: Option in WithStyle

rfuest, 2019-07-23:
`WithStyle::stroke` and `WithStyle::fill` currently take an `Option<C>` as an argument. For this builder style interface the `Some(X)` variant is used in almost all instances. I only found one use of `stoke(None)` in the e-g source and it wasn't even necessary in this case.

I would suggest to change `stroke` and `fill` to use `C` as an argument directly. For the rare case that someone wants to remove the stroke or fill additional methods could be added.

If you don't like this change maybe this can at least be implemented for the primitive macros, where the `Some`s look a little out of place IMHO.

jamwaffles, 2019-07-24:
Yes, losing the `Option`s would clean up both the macros and method call chains. That said, what would unsetting a style look like? `no_*()` methods would end up being pretty cumbersome I think, particularly with more style options in the future. Do you have any suggestions in mind for how the API might look?

rfuest, 2019-07-24:
I probably would use `no_*()` methods.

Implementing the functions for `T: Into<Option<C>>` (to accept `C` and `Option<C>`) would also work but make the function declarations in `WithStyle` more complex. And I don't think that this is a common pattern in Rust APIs, or have you seen this used before in any commonly used crate?

> `no_*()` methods would end up being pretty cumbersome I think, particularly with more style options in the future.

Cumbersome to use or cumbersome to implement?

jamwaffles, 2019-07-25:
> ... but make the function declarations in WithStyle more complex. 

Not that much more complex, right? The function signatures would change a bit and there would be a `.into()` call in the body, but there wouldn't be any other changes.

> And I don't think that this is a common pattern in Rust APIs, or have you seen this used before in any commonly used crate?

I haven't seen any other (ab)uses of the builder pattern like setting styles in e-g, so I don't have much to go on. 

> Cumbersome to use or cumbersome to implement?

Both; it's twice the number of methods in the trait to have to implement, but it's a pain to use as well. If someone is animating styles between `Some(...)` or `None`, it's more ergonomic to be able to just pass a variable in instead of swapping between e.g. `stroke()` and `no_stroke()` I think. Here's a contrived example:

```rust
let mut border = Some(C1);
let mut circle = Circle::new(icoord!(64, 64), 64).stroke(border);

loop {
    display.draw(circle);

    border = match border {
        Some(_) => None,
        None => Some(C1),
    };

    circle = circle.stroke(border);

    thread::sleep(Duration::from_millis(200));
}
```

This also raises another question about a mutating version of the style API, but I'll leave that line of thought alone for now.

I'd be happy to explore a solution using `T: Into<Option<C>>` as suggested, as I think being able to pass `None` for a style property keeps the API cleaner and smaller than having `no_*()` methods.

rfuest, 2019-07-25:
One way to reduce the boilerplate code would be to add default implementations to the `WithStroke` functions and add an additional getter for the `Style` struct:
```rust
pub trait WithStyle<C>
where
    C: PixelColor,
    Self: Sized,
{
    fn get_style_mut<'a> (self) -> &'a mut Style<C>;

    fn stroke(self, stroke: Option<C>) -> Self {
        self.get_style_mut().stroke_color = stroke;

        self
    }
...

```
This way the primitives would only need to implement `get_style_mut`.

rfuest, 2019-07-25:
But maybe it would be even better to completely remove the `style` parameters from the primitives. This could help to keep all the styling code in one location. I didn't think this through completely but something like this might work:
```rust
struct Styled<T, C> {
   primitive: T,
   style: Style<C>,
}
```
(Or maybe even remove the `Style` struct and move the fields directly into `Styled`)

`IntoIterator` could then be implemented for e.g. `Styled<Rectangle>` instead of `Rectangle`.

jamwaffles, 2019-08-25:
I feel like that could work. It would fit the use case for #156 nicely as well. What would an iterator impl look like? Instead of 

```rust
impl<'a, C> IntoIterator for &'a Circle<C>
where
    C: PixelColor,
{
    type Item = Pixel<C>;
    type IntoIter = CircleIterator<C>;

    fn into_iter(self) -> Self::IntoIter {
        CircleIterator {
            // ...
        }
    }
}
```

Would we then get

```rust
impl<'a, C> IntoIterator for &'a Styled<Circle, C>
where
    C: PixelColor,
{
    type Item = Pixel<C>;
    type IntoIter = CircleIterator<C>;

    fn into_iter(self) -> Self::IntoIter {
        CircleIterator {
            // ...
        }
    }
}
```

?

How would you get the style properties in the `next()` iterator impl?

rfuest, 2019-08-26:
> What would an iterator impl look like?

I think something like your example could work.

> How would you get the style properties in the `next()` iterator impl?

If you implement `IntoIterator` for `Styled<Circle, C>` you have access to the style in the `into_iter` method and can store a reference to `Style` or `Styled` in the `CircleIterator` struct.

jamwaffles, 2019-08-28:
Sounds good. What would styling look like when constructing primitives? Something like this:

```rust
let objects = Circle::new(icoord!(64, 64), 64)
    .into_iter()
    .chain(Line::new(icoord!(64, 64), icoord!(0, 64))))
    .chain(Line::new(icoord!(64, 64), icoord!(80, 80))))
    .chain(
        Font6x8::render_str("Hello World!")
            .translate(icoord!(5, 50)),
    );

let styled = Styled::new(objects).stroke(Some(C1)).fill(Some(C0));
```

? This example passes an iterator into `Styled` (or whatever it might be called), but I'd expect to be able to pass a single primitive in as well:

```rust
let styled = Styled::new(Circle::new(icoord!(64, 64), 64))
    .stroke(Some(C1))
    .fill(Some(C0));
```

Would that work with trait bounds? My tired brain is telling me similar usage patterns work with `Drawing` so should work with `Styled`, right?

rfuest, 2019-08-29:
The first example wouldn't work, because there is no way to create an iterator from a primitive without the styling information. The second example is similar to what I suggested above.

Without investigating this further I can't immediately  think of a good way to allow `Styled` to work with a single object and a chain of objects. For multiple objects of the same type it might be possible to implement `Styled` for `&[T]` where `Styled<T>` is a drawable.

jamwaffles, 2019-08-29:
> there is no way to create an iterator from a primitive without the styling information

Could e-g elements return `Styled<Self>` perhaps?

> I can't immediately think of a good way to allow Styled to work with a single object and a chain of objects.

Yes, I think this is going to be a tricky one to solve but it's a core feature of e-g so it needs to work 😕

rfuest, 2019-08-30:
> Could e-g elements return `Styled<Self>` perhaps?

Where exactly would this be returned?

> Yes, I think this is going to be a tricky one to solve but it's a core feature of e-g so it needs to work confused

This is different from the currently implemented core feature that iterators with `Pixel` items (aka `Drawable`s) can be chained, which would still work. Chaining of unstyled primitives would be a different feature which could be used to simplify applying the same styling to multiple objects (if I understood your example correctly). 

Perhaps it would be easier to try this in code. Moving the styling outside the primitives would be a first step. If this is done we can try different methods of applying styling to one or multiple primitives for drawing (like chaining or a context which holds a style).

jamwaffles, 2019-08-30:
> Where exactly would this be returned?

Like,

```rust
impl Circle {
    pub fn new(center: Point, radius: u8) -> Styled<Self> {
        // ...
    }
}
```

Which would keep the current behaviour that e-g primitives always have a default style.

> Chaining of unstyled primitives would be a different feature

Right, that's an important (new) distinction to make. Primitives at the moment _can't_ be unstyled so this would be a change in that behaviour. I'm not sure if it's for the better or not.

> Perhaps it would be easier to try this in code. Moving the styling outside the primitives would be a first step.

Certainly :), although chained _and_, styled primitives will become quite verbose if each one requires the same style applied to it (`Circle::new().stroke().fill().chain(Rectangle::new().stroke().fill())`, etc). A drawing context as you're experimenting with would start to make a lot more sense in this case, whilst still supporting iterators.

rfuest, 2019-08-30:
> Which would keep the current behaviour that e-g primitives always have a default style.

I think `new` shouldn't add any styling because of #156.

> Certainly :), although chained _and_, styled primitives will become quite verbose if each one requires the same style applied to it (`Circle::new().stroke().fill().chain(Rectangle::new().stroke().fill())`, etc)

Yes, this is very verbose and the reason I started the drawing context experiment. To reduce the code a bit you could prepare a style first and then apply that to each primitive in the chain. Something like:
```rust
let s = Style::new().stroke(...).fill(...);
Circle::new(...).style(s).chain(Rectangle::new(...).style(s))
```
But that not much better either.

jamwaffles, 2019-08-30:
> I think `new` shouldn't add any styling because of #156.

Oh for sure, that was a spitball idea that doesn't make any sense. Rust convention strongly implies that `Thing::new()` gives you a new `Thing` and not something else.

> But that not much better either.

This is where the macros might come in handy, or your context idea.

This discussion has deviated from the original issue at hand I think. Shall we open two new issues - one to discuss styling, the other for #149 to work against? Let me know if you'd like to make those issues, otherwise I'll put a couple up.

rfuest, 2019-09-01:
> This discussion has deviated from the original issue at hand I think. Shall we open two new issues - one to discuss styling, the other for #149 to work against? Let me know if you'd like to make those issues, otherwise I'll put a couple up.

Sounds good, you can open the issues.

jamwaffles, 2019-09-01:
Opened #162 and #163. As for the original issue, what are your thoughts on that? Shall I close this issue now that the other two are open?

rfuest, 2019-09-01:
Yes, this can be closed. We already decided not to implement my original suggestion (in #157).

closed 2019-09-01 by rfuest