embedded-graphics/embedded-graphics #141: Coordinate types

rfuest, 2019-07-24:
e-g currently uses two coordinate types `Coord` and `UnsignedCoord`. Their name is purely based on the data type they use internally and not on their intended use. This makes it harder to know which one a function expects.

I think it would be better to use an approach similar to servo's euclid crate or flutter's Offset and Size classes. This could be accomplished by renaming `Coord` to `Point` or `Position` and `UnsignedCoord` to `Size`.

The one place where this semantic meaning isn't already followed is the `Pixel` type which uses `UnsignedCoord` to make sure that only positive coordinates can be drawn. But changing this to use signed coordinates could be beneficial. Currently every drawing iterator has to manually check if all coordinates are >= 0, which can be quite error prone. This check could be easily implemented in the display drivers, which need to check for too large coordinates anyway. Not clipping the drawing inside the iterators would also allow to translate the pixels using `map()` without loosing the normally hidden parts. And using one type for all coordinates will get rid of many `to_unsigned()` and `to_signed()` calls.

Both examples mentioned above also implement a `Rect` type which can be constructed by combining a `Point` and a `Size`, which could be useful in many places.

jamwaffles, 2019-07-25:
Naming stuff is hard :sweat_smile:. `Position` and `Size` do make more sense than `UnsignedCoord` and `Coord`.

> Not clipping the drawing inside the iterators would also allow to translate the pixels using map() without loosing the normally hidden parts. 

I hadn't thought of that! That's definitely a bug which would be fixed by moving the bounds checks into the drivers.

> Both examples mentioned above also implement a Rect type which can be constructed by combining a Point and a Size, which could be useful in many places.

What do you mean by this? There's already `Rectangle` which uses two `Position`s, so are you suggesting another type, or a way to make a `Rectangle` with a `Position` and a `Size`?

rfuest, 2019-07-25:
> What do you mean by this? There's already `Rectangle` which uses two `Position`s, so are you suggesting another type, or a way to make a `Rectangle` with a `Position` and a `Size`?

I didn't explain that clearly. The current `Rectangle` type is a primitive that can be drawn, but the `Rect` type would simply describe a rectangular area without any attached styling. This could be uses for all instances where a rectangular area are used, like the `Dimensions` trait. A `Rect` type could also provide additional functions like testing if a `Point` is inside a given `Rect`.

jamwaffles, 2019-08-02:
Ah I see, thanks for the explanation. Can you give some examples of how `Rect` would be used in real world apps? I want to understand why `Rect` would be better than the methods provided by the `Dimensions` trait.

> A Rect type could also provide additional functions like testing if a Point is inside a given Rect.

IMO that's outside the scope of e-g. Would the purpose of `Rect` in this case be to provide a more ergonomic way of getting an area?

Also, `Rect` and `Rectangle` will get confusing pretty quick! Is there a better name you can think of for `Rect`. I'd prefer to keep `Rectangle` as is for less public API churn if possible. Maybe `Rect` could be `Quad`? `Area`? `Box`? `BoundingBox`? I've also thought about renaming `Rectangle` to `Quad` in the past, but let's avoid that as much as possible.

rfuest, 2019-08-03:
> Can you give some examples of how `Rect` would be used in real world apps? I want to understand why `Rect` would be better than the methods provided by the `Dimensions` trait.

The `Dimensions` trait is used to get the bounding box of a graphics object, but a `Rect` might be any other size. One use case could be clipping, like this:
```rust
let clip = Rect::new(0, 0, 10, 10);
let img = Image::new(...);
display.draw(img.into_iter().filter(|Pixel(p, _)| p.is_inside(clip));
```
> IMO that's outside the scope of e-g.

Maybe, but I think it's important to provide such features that are consistent with the way e-g handles coordinates. This might be internal or an external crate that uses the same semantics as e-g. The first step would be to document how coordinates and sizes should be handled exactly (see #142).

> Also, `Rect` and `Rectangle` will get confusing pretty quick!

`Rect` seems tho be the de facto standard for this struct/class. But I think we should decide if we want to add this at all before deciding the final names :wink:.

Also if we should decide to implement this: https://github.com/jamwaffles/embedded-graphics/issues/139#issuecomment-515121637 , the primitives would only describe geometric shapes and `Rectangle` would effectively be equal to the proposed `Rect`.

rfuest, 2019-08-19:
Another thing to consider is the optional nalgebra support. The internal coord type and the type alias aren't 100% compatible, which can lead to bugs  if a crate doesn't test both variants.

One such problem I encountered is that the internal types can be destructured, but the type alias can't. This brakes, for example, building the `ssd1675` crate with `nalgebra_support`:
https://github.com/wezm/ssd1675/blob/38789231efa744e3f68ba52ef07854c6f5c2deee/src/graphics.rs#L158-L170

I would prefer to choose whether to use nalgebra or not and don't make this optional, because it's hard to make sure the types are usable in the same way.

jamwaffles, 2019-08-19:
I've been thinking about removing integral Nalgebra support completely; it feels like a mistake adding integral support for it, and I don't think anyone uses it in the wild.

A better option would be to do Nalgebra calculations outside of any graphics stuff, then call `.into()` to get the right `Coord`, etc type that e-g understands.

This way, the `nalgebra` feature would enable `From<[some nalgebra type]>` for this crate's coord types and remove the requirement for tight integration. Does that make sense to you?

rfuest, 2019-08-19:
Yes, that's a good idea.

If we decide to change the names of the coord types, what do you think about using a struct with x and y fileds instead of the tuple struct? IMHO `point.x` and `point.y` are easier to read than `point[0]` or `point.0`.

jamwaffles, 2019-08-19:
> what do you think about using a struct with x and y fileds instead of the tuple struct?

Another breaking change, but we've already got enough of those to push library maintainers past the sanity line I think 😅😬. I'm not sure why I went with tuples in the first place, probably just a bit less boilerplate. Structs can still `impl Index` so that still allows `point[0]`, etc, so I'd be happy with that change.

rfuest, 2019-08-20:
> Another breaking change, but we've already got enough of those to push library maintainers past the sanity line I think :sweat_smile::grimacing:

We might have, sorry for that :wink:. But, judging by https://crates.io/crates/embedded-graphics/reverse_dependencies, most driver crates are still using e-g `0.4`, and will already need changes anyway. And hopefully the versions after `0.6` won't break the display drive API every time :grimacing:.

What is you opinion on the original topic of this thread? Should we go ahead and change the names of `Coord` and `UnsignedCoord` or leave them as they are?

jamwaffles, 2019-08-21:
Yes, thankfully most drivers are still a couple of versions behind.

> And hopefully the versions after 0.6 won't break the display drive API every time

That'd certainly be nice :joy: 

> What is you opinion on the original topic of this thread?

Let's see what `Point` and `Size` look like - they're much better names. Structs also make more sense with `.x` and `.y` fields. I assume `Point` is signed and `Size` is unsigned?

I'm also eager to fix this bug:

> Not clipping the drawing inside the iterators would also allow to translate the pixels using map() without loosing the normally hidden parts.

Are you happy to prepare another PR?

jamwaffles, 2019-08-21:
I've opened #155 for other visitors' benefit and to track this breaking change against 0.6.0.

rfuest, 2019-08-21:
> I assume `Point` is signed and `Size` is unsigned?

Having both types could also have some advantages because their range of values is equal, but having to deal with negative sizes everywhere seems to be the worse problem. And in normal use case sizes > 2^31 aren't really common.

> I'm also eager to fix this bug:
> 
> > Not clipping the drawing inside the iterators would also allow to translate the pixels using map() without loosing the normally hidden parts.

Yes, this bug became apparent again in my work on the image module. Having `Pixel` use the new signed `Point` type should clean up the iterator code for images and also the primitives.

> Are you happy to prepare another PR?

I can do that, after #154 is merged.

Two more questions about the new types:
* How should the fields in the `Size` struct be named, `x` & `y` or `width` & `height`? I've seen both variants in other libraries, but I'm not sure which one to pick.
* Some libraries use an additional third type to be more mathematically, a vector type used, for example, for translations. Do you think this has any practical advantages?

jamwaffles, 2019-08-21:
> Having both types could also have some advantages because their range of values is equal

Not sure what you mean by this. Can you elaborate?

> I can do that, after #154 is merged.

I've addressed your comments - thanks for reviewing - so that's ready to go if there are no other changes you think should be made.

> How should the fields in the Size struct be named, x & y or width & height? I've seen both variants in other libraries, but I'm not sure which one to pick.

Oh no, naming again! Maybe `x` and `y` as they're a bit more generic, and shorter to type. Could also use `w` and `h`? If `Size` is never used for anything other than, er, sizes, then either `width` and `height` or `w` and `h` would be my preference.

> Some libraries use an additional third type to be more mathematically, a vector type used, for example, for translations. Do you think this has any practical advantages?

I opened #3 ages ago to address more mathematical operations, however I think it might better left to things like Nalgebra to perform complex ops outside of e-g, then map over a pixel iterator to apply that computation. I feel like we'd just end up reimplementing half of Nalgebra otherwise.

---

I'd like to leave `Rect` for a different PR. I'm envisaging a clipping/masking API that can accept various different clip shapes, not just rectangles. e-g primitives could be reused for this purpose. For example, to get a circular mask:

```rust
// Naming/import structure to be decided!

use embedded_graphics::{coord, mask::Mask, primitives::Circle};

let mask = Mask::new(Circle::new(coord!(10, 20), 10));

let point = Point::new(15, 25);

mask.contains(&point);
```

Let's open a new issue if you'd like to discuss this further. 

---

Another naming thought: should it be `Position` instead of `Point`? I also saw `Offset` mentioned above, but I think that introduces ambiguity around what the offset is relative to. 

If I write the following:

```rust
text_6x8!("Hello World! - default style 6x8", point!(15, 15));
// or
text_6x8!("Hello World! - default style 6x8", Point::new(15, 15));
// or! lol
text_6x8!("Hello World! - default style 6x8", Point { x: 15, y: 15 });
```

The word `Point` is quite ambiguous here. With `Position`, it's a lot clearer, however `Position` is a lot more specific - `Point` is much more general purpose. Do we go with `Point` so all of e-g can use it (externally and internally) and reword the docs to state that it's the origin _point_ for an object? I'm erring on the side of "yes", but I'm interested to hear your thoughts.

rfuest, 2019-08-21:
> Not sure what you mean by this. Can you elaborate?

I just mean that you don't have to care about casting between `i32` to `u32` and the side effects this might cause for large or negative values.

> I've addressed your comments - thanks for reviewing - so that's ready to go if there are no other changes you think should be made.

Go ahead and merge it, any additional changes can be made and discussed in the new PR.

> Oh no, naming again! Maybe `x` and `y` as they're a bit more generic, and shorter to type. Could also use `w` and `h`? If `Size` is never used for anything other than, er, sizes, then either `width` and `height` or `w` and `h` would be my preference.

Naming seems to be 90% of the work. `w` and `h` aren't as common as the other variants. I would suggest to use `width` and `height` for now and see how that looks in code.

> I opened #3 ages ago to address more mathematical operations, however I think it might better left to things like Nalgebra to perform complex ops outside of e-g, then map over a pixel iterator to apply that computation. I feel like we'd just end up reimplementing half of Nalgebra otherwise.

The vector type wasn't supposed to have more functions but only to distinguish their use case. Like nalgebra's `Vector2` and `Point2` types. Excerpt from their documentation:
> In a geometric context: a point is an element of an euclidean space, while a vector is an element of the underlying vector space. In other words, a point is a location in space while a vector is a translation, i.e., the displacement required to move from one point to another:

>I also saw `Offset` mentioned above, but I think that introduces ambiguity around what the offset is relative to.

I think that flutter uses `Offset` because they don't have a separate `Vector` and `Point` type and needed a generic name for both functions.

> The word `Point` is quite ambiguous here. With `Position`, it's a lot clearer, however `Position` is a lot more specific - `Point` is much more general purpose. Do we go with `Point` so all of e-g [...]

Yes, calling the type `Position` seems too specific (and also long). I would suggest to try `Point` and then identify places where additional docs or code changes might be necessary.

This would be easier to read:
```rust
text_6x8!("Hello World! - default style 6x8", origin = point!(15, 15))
// or
text_6x8!("Hello World! - default style 6x8").translate(point!(15, 15))
```
But would break even more code...

jamwaffles, 2019-08-21:
> I just mean that you don't have to care about casting between i32 to u32 and the side effects this might cause for large or negative values.

Ah I see, yeah that can be pretty painful currently.

> Go ahead and merge it, any additional changes can be made and discussed in the new PR.

Merged!

> Naming seems to be 90% of the work. w and h aren't as common as the other variants. I would suggest to use width and height for now and see how that looks in code.

Sounds good

> I would suggest to try Point

Sounds good too!

> But would break even more code...

ᴺᵒᵒᵒᵒᵒᵒᵒᵒᵒ

PR #158 (2019-09-02): Rename (Unsigned)Coord to Point and Size
This PR implements the name changes discussed in #141. The comments aren't updated yet and some tests are ignored at the moment (mostly in the image module, which will be changed anyway in #152).

Some notes:
* I removed the `Neg` implementation for `UnsignedCoord`, which didn't seem to make sense for `Size`
* I added support for `point + size` and `point - size` which return a `Point`. `point += size` and `point -= size` are also implement.
* I removed the `To(Un)Signed` traits.
  `UnsignedCoord` to `Coord` can be replaced by `Point::orign() + size`, w [... cut]
closed 2019-09-02 by jamwaffles