embedded-graphics/embedded-graphics #142: Dimension::size() off by one error

rfuest, 2019-07-24:
The size returned by `Dimension::size` for `Rectangle` and `Circle` seems to be one pixel to small. There might be other `Drawable`s which also return the wrong size.

## Code
```rust
lett rectangle = egrectangle!((5, 5), (7, 7), stroke = Some(BinaryColor::On));
display.draw(&rectangle);
println!("rectangle: {:?}", rectangle.size());

let circle = egcircle!((15, 6), 1, stroke = Some(BinaryColor::On));
display.draw(&circle);
println!("circle: {:?}", circle.size());
```

## Stdout
```
rectangle: UnsignedCoord(2, 2)
circle: UnsignedCoord(2, 2)
```

## Display
[image]

caemor, 2019-07-25:
It seems like all of the primitives have this "error"/undocumented behaviour. (see added simulation example in my branch but with already "fixed" behaviour - you can change this in the cargo.toml of the simulator )
But I found multiple confusing parts:
- Lines: A line of size 1 is not drawn at all and there is even a test for it:
 https://github.com/jamwaffles/embedded-graphics/blob/2ddbaad976a67dcb55d5023ed81384128e795726/embedded-graphics/src/primitives/line.rs#L309-L315
 I would have expected that a line of size 1 is the same as a dot, since for a variable sized line of lengths from zero to let's say ten you would need a special case just for this line of size 1 to make it a dot instead of a line
- Circles: You can't make circles with an even size (you always get size 3,5,7...), the opposite is for filled circles without a stroke color where you only have even sized fillings
- Triangles: Triangles of Size One are not shown - is it same no dot policy as for lines?
- Rectangles: Here you can make dots with 1 sized rectangles :-D

My branch with the tests and maybe fixes can be found here: https://github.com/caemor/embedded-graphics/tree/fix-dimension-size
I also added a new simulation example called size to show the different sizes.

jamwaffles, 2019-07-25:
Thanks for adding some extra detail. Can you open a PR with your fixes so it's easier to review?

The dot/no dot thing has cropped up before with I think a PR contributed to change the behaviour to _not_ draw a dot. Is there a common convention/expectation for elements with no size that you know of? Things start to get a bit weird with dimensions as well - what size should a line from `(10,10)` to `(10,10)` have? `0`? `1`?

I would certainly like to make things more consistent with a dot/no dot, however I can see arguments against going in either direction which makes coming up with a solution tricky. 

> Circles: You can't make circles with an even size (you always get size 3,5,7...), the opposite is for filled circles without a stroke color where you only have even sized fillings

Is this related to #143?

jamwaffles, 2019-08-28:
Quick spitball here. For consistent behaviour, but also retaining the ability to draw points (think sand sims, etc), how about this:

1. Primitives with "no size" (to be defined per primitive - too tired rn) don't draw  a dot. The line currently does this, as do triangles as pointed out by @caemor. Other primitives would be updated to have similar expected behaviour at zero sizes.
2. Add a `Dot` (or other named) primitive that _only_ draws a single dot.

I could be way off the mark here, but if this is behaviour worth persuing I'll open another issue to discuss there.

rfuest, 2019-08-29:
I think we first need some clear documentation of how coordinates and sizes should be interpreted in general and then make sure the corner cases are handled correctly.

There are several different ways to handle integer pixel coordinates, which all have advantages and disadvantages. The old Mac QuickDraw for example defines points to lie on the grid lines between pixels and Windows GDI defines point to lie in the center of each pixel. These design decisions also influence other aspects, like if the bottom right corner is included or excluded in a rectangle.

Links to the mentioned APIs:
https://docs.microsoft.com/de-de/windows/win32/gdi/windows-gdi
https://developer.apple.com/library/archive/documentation/mac/pdf/ImagingWithQuickDraw.pdf

jamwaffles, 2019-08-29:
> The old Mac QuickDraw for example defines points to lie on the grid lines between pixels and Windows GDI defines point to lie in the center of each pixel.

Ahh! Thank you so much for concisely explaining the two approaches I had in my head. 

It seems to me e-g is going to have to pick one and go with it. OpenGL [uses pixel _centers_](https://stackoverflow.com/a/25468051/383609), so it might be worth going for that by convention. 

> make sure the corner cases are handled correctly.

One edge case is zero-sized elements as mentioned above. If pixel _centers_ are used, it makes sense to me for and e-g primitive with zero sized to be drawn as a dot I think. Following OpenGL further, what does it do if I try and shade a triangle made of the points `(0,0)`,`(0,0)`,`(0,0)`? If that's a dot, it makes sense for e-g to follow that behaviour. What are other peoples' thoughts?

rfuest, 2019-08-30:
> It seems to me e-g is going to have to pick one and go with it.

Yes, I think consistency and a documented behavior is more important than the actual choice.

> OpenGL [uses pixel _centers_](https://stackoverflow.com/a/25468051/383609), so it might be worth going for that by convention.

Interesting, but there are many examples of both approaches. Cairo and other canvas APIs use coordinates that lie between the pixels.

Using the grid line approach works really well for shapes like rectangles. A zero-sized rectangle is defined by `top == bottom` and `left == right` and other rectangles simple use `bottom = top + height` and `right = left + width`. A circle, which is defined by a center point and a radius, would also fit nicely into this and result in `width = height = radius * 2`. So for closed shapes I prefer this approach.

But this doesn't work great for lines. The other canvas APIs "cheat" by using floating point numbers to also be able to specify coordinates in the center of pixels. Because we can't use floating point coordinates we would need to introduce some additional conventions for line drawing. Possible solutions are an implicit 0.5 pixel offset or the drawing method used by QuickDraw (see page 1-12 in the PDF linked above).

jamwaffles, 2019-08-30:
> Cairo and other canvas APIs use coordinates that lie between the pixels.

Good point - those are 2D APIs whereas OpenGL is 3D so pixel centers might not make sense for e-g's 2D API. 

> Using the grid line approach ... A zero-sized rectangle is defined by top == bottom and left == right

Would this produce a single dot or not draw anything at all?

> So for closed shapes I prefer this approach.

Yes, it seems to nicely fix the off-by-one errors described in this issue.

> But this doesn't work great for lines.

Whatever hacks get us consistent behaviour... :grinning:

rfuest, 2019-08-30:
> Would this produce a single dot or not draw anything at all?

A zero sized rectangle wouldn't draw anything. But it is possible to draw a single dot by using `right = left + 1` and `bottom = top + 1`.

> Whatever hacks get us consistent behaviour... grinning

Ok, let's switch to floating point :wink:. Or not...

I like the current way of handling strokes on rectangles and circles, because it's simple and works the same for even and odd stoke width. But to be consistent with strokes on lines we might have to change this. I would expect a line from `(rect.left, rect.top)` to `(rect.right, rect.top)` to look the same as the upper edge of the stroke of `rect` (with equal stroke width and assuming that stroke width would be implemented for lines). This would require that the stroke of primitives is drawn centered on the outside edge and not from the outer edge inwards, like it's implemented now. Problems with this approach are that odd stoke width can't be centered perfectly (requires rounding) and that the stroke extends beyond the coordinates defined by e.g. `Rectangle`.

jamwaffles, 2019-09-04:
> I like the current way of handling strokes on rectangles and circles ...  But to be consistent with strokes on lines we might have to change this.

The current fill-from-outside-in for strokes on circles and rects was a deliberate decision to mimic the `box-sizing: border-box` property from CSS. It also makes size calculations for those primitives simpler (both programatically and mentally) as the width isn't dependent on border size. When filled triangles are supported, they should also be stroked from the outer edge inwards IMO. I don't have a good answer for filled/unfilled polygon behaviour so I'd appreciate some input there.

I'd like to center this issue around an earlier point which I think solves the original reported problem: grid lines vs pixel centers.

> Using the grid line approach works really well for shapes like rectangles. A zero-sized rectangle is defined by top == bottom and left == right and other rectangles simple use bottom = top + height [...]

That makes sense to me. Because triangles (and eventually polygons) are (will be) drawn with lines, I think that's the last bit of behaviour to figure out.

> Possible solutions are an implicit 0.5 pixel offset or the drawing method used by QuickDraw ... see page 1-12 in the PDF linked above

So a pixel for a line is defined by its top-left corner? That makes sense to me. Would a line with no length (from `(10, 10)` to `(10, 10)`, say) draw a dot or nothing at all? For consistency with the discussed rectangle behaviour, it should probably be nothing I think?

If you're happy with this behaviour, I'll write up the important bits of this thread into an issue to create PRs against.

rfuest, 2019-09-05:
> The current fill-from-outside-in for strokes on circles and rects was a deliberate decision to mimic the `box-sizing: border-box` property from CSS. It also makes size calculations for those primitives simpler (both programatically and mentally) as the width isn't dependent on border size.

Mimicking  `border-box` worked, I was immediately reminded of it when I looked at how rectangles work in e-g. But I'm still not sure that the drawing API should implement this and not leave this behavior to an higher level library, like a widget library. IMO a drawing API should simply handle strokes and fills on paths and not distinguish between strokes on lines and borders on shapes.

Let's assume you want to draw a rectangle and add additional lines directly above and below the rectangle. With the current way rectangles are drawn the top and bottom lines need to be handled differently:
```rust
let sw = 10; //stroke width
let rect = Rectangle::new(Point::new(100, 100), Point::new(200, 200))
        .stroke(Some(Rgb888::Red)).stroke_width(sw);
// top can just be offset by the stroke width
let top = Line::new(Point::new(100, 100 - sw), Point::new(200, 100 - sw))
        .stroke(Some(Rgb888::Green)).stroke_width(10);
// bottom can't because strokes and borders are drawn differently
let bottom = Line::new(Point::new(100, 200 + 0), Point::new(200, 200 + 0))
        .stroke(Some(Rgb888::Green)).stroke_width(10);
```
If rectangles would be treated the same way as lines the `y` coordinate of `bottom` would be `200 + sw` and more consistent with the `top` line.

> I don't have a good answer for filled/unfilled polygon behaviour so I'd appreciate some input there.

I would expect that an unfilled unclosed polygon behaves the same way as a set of lines. And I also expect that a closed polygon with 4 points, that are equal to a rectangle's corner, behaves the same way as a rectangle. But I don't think that is possible if strokes on rects and lines are handled differently.

> So a pixel for a line is defined by its top-left corner? That makes sense to me. 

In QuickDraw a _pen_ is dragged along the line with the top-left corner of the pen on the line.

> Would a line with no length (from `(10, 10)` to `(10, 10)`, say) draw a dot or nothing at all? For consistency with the discussed rectangle behaviour, it should probably be nothing I think?

I'm for drawing a dot in this case. To keep with the pen metaphor: Touching a pen on a single point leaves some paint behind. So I suggest that a zero sized fill doesn't draw anything (there is no area to fill), but a zero length line does.

jamwaffles, 2019-09-14:
I'm going to close this in favour of #182 which describes the underlying problem better. Feel free to reopen if you think this is an error.

closed 2019-09-14 by jamwaffles