embedded-graphics/embedded-graphics #143: Circles without stroke are drawn incorrectly

rfuest, 2019-07-24:
The stroke on circles is applied from the outside edge of the circle inwards. This means that a circle with only fill set should render the same way as one with stroke and fill set to the same color.

## Code
```rust
display.draw(egcircle!(
    (10, 16),
    3,
    fill = Some(Rgb888::RED)
));

display.draw(egcircle!(
    (22, 16),
    3,
    fill = Some(Rgb888::GREEN),
    stroke = Some(Rgb888::GREEN)
));
```

## Display
[image]

jamwaffles, 2019-07-25:
Good catch, and thanks for the test case.

rfuest, 2019-11-25:
There still seems to be a problem with this for some radii. The test `issue_143_stroke_and_fill` fails if the radius is changed to 4 pixel. I noticed this in the `fill-macros` example, which uses a 32 pixel radius, so maybe all even radii cause an issue.

PR #177 (2019-09-10): Fix circle stroke fill
TODO

* [x] Add failing unit test
* [x] Actually fix the bug
* [x] Remove debugging demo from simulator before merge

Closes #143
PR #209 (2019-11-30): Fix circle rendering (again :()
Hi! Thank you for helping out with Embedded Graphics development! Please:

- [x] Check that you've added passing tests and documentation
- [x] Add a simulator example(s) where applicable
- [x] Add a `CHANGELOG.md` entry in the **Unreleased** section under the appropriate heading (**Added**, **Fixed**, etc)
- [x] Run `rustfmt` on the project
- [x] Run `./build.sh` (Linux/macOS only) and make sure it passes. If you use Windows, check that CI passes once you've opened the PR.

## PR description

Closes #143 again. Not sure if this is a regression or the test never  [... cut]
closed 2019-11-30 by jamwaffles