embedded-graphics/embedded-graphics #147: Circle defined by center and diameter

jamwaffles, 2019-07-25:
`Circle::new()` creates a circle by center and _radius_. Is it useful to be able to create one by giving a diameter instead? The diameter can then be an odd number, however won't be symmetrical around the circle's center point. Does this matter?

rfuest, 2019-07-25:
I think it would be easier to use the top left point instead of the center as a parameter. This would for example make it easier to draw a circle inside a rectangle.
Or if we decide to add a `Rect` (see #141) type this could also be used instead.

aurelj, 2020-03-13:
I've just stumbled on this issue.

Current `Circle` defined by center and radius only allows building circles with an odd diameter (2 * radius + 1). It is impossible to build circles with an even diameter. We can only build half of the possible circles. This is an issue when you want to draw a circle centered on a display with an even width and height (pretty much any display in existence ?).

This is related to #182. With a coordinate system based on the center of pixels and a circle defined by its center, there is no way to build a circle with an even diameter. One solution would be to change the coordinate system to lie on the grid between pixels, but then we couldn't build circles with an odd diameter.

I think the best way to solve this is, as suggested by @rfuest, to define a circle by its top left point and its diameter. This allows building both odd and even diameter circles.

I'm willing to implement this, but I'm wondering what's the best way to do it:
- keep the current `Circle` and add a new `CircleDiameter` (or a better name) ? (I don't really like this)
- change the implementation of `Circle` breaking API compatibility (in 0.7.0 I guess) ?
- something else ?

I'm leaning toward the second solution. What do you think ?

rfuest, 2020-03-13:
I would also prefer the second solution. Having two types to represent circles would be confusing and unnecessary because all circles can be represented using the new type.

Changing the implementation is a breaking change, but the new type can provide additional constructors and getters to make porting old code easier. Some getters like `center` would only be able to output rounded values, but I guess that can be expected when not using floating point coordinates.

One focus of 0.7 will be geometry (like #182 you mentioned), so this will be a good time to implement this change.

jamwaffles, 2020-03-18:
I'm on the fence about `Ellipse` being a separate shape or not. Here's my thinking on an API that supports ellipses and circles at the same time:

```rust
let center = Point::new(20, 20);

// Circle
let circle = Ellipse::new(center, Size::new(10, 10));

// Ellipse
let ellipse = Ellipse::new(center, Size::new(25, 10));
```

Thoughts?

aurelj, 2020-03-18:
I'm not sure if they should be separate shape or not, but if we add support for `Ellipse`, it should be defined by top-left and bottom-right rather than center and two "radius", for the same reason discussed just above.

jamwaffles, 2020-03-18:
Oh doy yeah, should've read the issue a bit slower...

PR #274 (2020-03-24): circle: define by top-left + diameter instead of center + radius
Hi! Thank you for helping out with Embedded Graphics development! Please:

- [X] Check that you've added passing tests and documentation
- [X] Add a simulator example(s) where applicable
- [x] Add a `CHANGELOG.md` entry in the **Unreleased** section under the appropriate heading (**Added**, **Fixed**, etc) and appropriate crate (`embedded-graphics`, `simulator`, `tinytga`, `tinybmp`) if your changes affect the **public API**
- [X] Run `rustfmt` on the project
- [X] Run `./build.sh` (Linux/macOS only) and make sure it passes. If y [... cut]
closed 2020-03-24 by jamwaffles