embedded-graphics/embedded-graphics #150: Advanced example ideas

rfuest, 2019-08-03:
The recent discussions about how different functions of e-g could be implemented or improved made me think that e-g should have some more advanced examples. The current examples are pretty basic and only show how to use single aspect of e-g. Having more "real world" examples would make it much easier to evaluate where the API could be improved.

I currently don't have any really good ideas for such examples, expect from different styles of clocks (the hello world of graphics libraries :grin:). Please comment if you have any good ideas for examples, which might look visually interesting or could help to improve e-g for your use case.

jamwaffles, 2019-08-04:
Good point! I guess these should all go in the simulator so people can see what they look like. This is what I've got so far:

* Clocks sounds good - it would show some usage of loops or reusable functions (drawing the ticks) and some basic state management for updating the hands
* A super simple up/down menu system that shows some layout and style changes (like highlighting the current item)
* A "DVD player screensaver" type bouncing logo demo that shows animating an object around the screen
* Might be a bit too large for an example, but some kind of super simple game like Snake, controlled with the arrow keys.

The simulator doesn't currently support input or animation, but adding that shouldn't be too difficult, and would allow some much richer examples.

jamwaffles, 2019-08-30:
Another idea to add to the list:

A Raspberry Pi (for WiFi support) to show sunrise/sunset like DuckDuckGo does:

[image]

Would demonstrate drawing text and more complex lines/shapes.

jamwaffles, 2019-09-13:
Another one:

A basic compass using an MPU9250 or similar

rfuest, 2019-09-15:
I've merged all example ideas into the top post.

@jamwaffles I did hide your comments that only contained new ideas, which are now added to the top post. I hope that's OK.
Are there any examples you are already working on or would like to work on soon? If not I would like to start working on the menu example next.

jamwaffles, 2019-09-15:
Thanks, that's much cleaner. I'm not working on anything at the moment so feel free to take the menu example. Perhaps open a draft PR and edit its number into this thread so we can keep track of who's doing what?

rfuest, 2019-09-15:
I don't have any WIP code yet to create a draft PR, but I'll open one and add it to the list when I do. Until then I will add a comment to the list.

I don't want to split this issues in separate issues for each example, but for other issues we should start to assign ourself to issues that we are working on to keep track of who's doing what.

JerelJr, 2024-10-20:
Are these examples still needed? If so, I have worked on a 7 segment display example that I can add. I'd also be open to contributing to the other examples.

rfuest, 2024-10-23:
For some reason we haven't paid much attention to this issue for a _little while_, but more examples are always nice to have. More complex examples should go into the [`examples` repo](https://github.com/embedded-graphics/examples), which still needs to be updated for e-g 0.8. We might also want to make that repo more visible to users who want to get started with e-g, I think the links in the docs might not be discoverable enough.

Is your 7 segment display example based on the [`eg-seven-segment` crate](https://github.com/embedded-graphics/eg-seven-segment) or a custom implementation?

JerelJr, 2024-10-24:
I did not know about the `eg-seven-segment` crate at the time of writing my comment. My example is a custom implementation. However, I noticed [this example](https://github.com/embedded-graphics/eg-seven-segment/blob/master/examples/digital-clock.rs) already exists in the `eg-seven-segment` crate. 

If it is no longer needed, then I can work on the 'up/down menu' example.

rfuest, 2024-10-28:
> I did not know about the `eg-seven-segment` crate at the time of writing my comment. My example is a custom implementation. However, I noticed [this example](https://github.com/embedded-graphics/eg-seven-segment/blob/master/examples/digital-clock.rs) already exists in the `eg-seven-segment` crate.

There has been some previous discussion about adding examples that demonstrate the wider e-g ecosystem to the `examples` repo: https://github.com/embedded-graphics/examples/issues/18. A common examples repo would make it easier to discover useful crates that extend e-g. If you have an idea for a `eg-seven-segment` example, or even just want to copy the existing clock example, that would be appreciated.

> If it is no longer needed, then I can work on the 'up/down menu' example.

Both examples would be nice to have.

I need to mention that is a crate to implement menus (https://github.com/bugadani/embedded-menu/tree/main), but I want to use an external crate for this example. My goal was to implement a fairly simple menu which shows how something like this could be implemented only using e-g.

JerelJr, 2024-10-30:
Thank you for the clarification. 

I noticed that the `eg-next` examples do not yet compile. Since my examples use version `0.8`, should my examples still be added there or should I target an older version?

rfuest, 2024-10-31:
The examples repo needs to be updated to e-g 0.8, before we add anything new. `eg-next` should have been copied to a new `eg-0.8` directory when e-g 0.8 was released, but updating the examples repo isn't documented as part of our [release process](https://github.com/embedded-graphics/embedded-graphics/blob/master/doc/releasing.md) and it was unfortunately forgotten. The intention of creating a new copy of the examples for each breaking release was to allows us to work on API changes in the development version without breaking the examples for users of the crates.io version.

> I noticed that the `eg-next` examples do not yet compile. Since my examples use version `0.8`, should my examples still be added there or should I target an older version?

I think that `eg-next` did compile in the past and would still work with the latest releases, but not with the latest changes in the GIT version. Would you be interested in creating a PR to copy `eg-next` to `eg-0.8` and to replace all GIT dependencies in the `eg-0.8` directory with regular dependencies? That should fix the examples for `eg-0.8` and give you a stable target to add your examples.

JerelJr, 2024-10-31:
>  Would you be interested in creating a PR to copy eg-next to eg-0.8 and to replace all GIT dependencies in the eg-0.8 directory with regular dependencies?

Can do!