embedded-graphics/embedded-graphics #156: Additional uses for primitive shapes

rfuest, 2019-08-21:
From #141:

> I'd like to leave `Rect` for a different PR. I'm envisaging a clipping/masking API that can accept various different clip shapes, not just rectangles. e-g primitives could be reused for this purpose. For example, to get a circular mask:
> 
> ```rust
> // Naming/import structure to be decided!
> 
> use embedded_graphics::{coord, mask::Mask, primitives::Circle};
> 
> let mask = Mask::new(Circle::new(coord!(10, 20), 10));
> 
> let point = Point::new(15, 25);
> 
> mask.contains(&point);
> ```
> 
> Let's open a new issue if you'd like to discuss this further.

Being able to use the primitive shapes for more uses would be a great feature. As mentioned in #139  and https://github.com/jamwaffles/embedded-graphics/issues/141#issuecomment-517929501 I would suggest to remove the styling from the primitives and have structs that only represent geometric shapes.

jamwaffles, 2019-11-19:
Cc #202 - there seems to be some real-world desire for things like hit testing which is good validation of this issue.

rfuest, 2019-11-30:
Related comment from #162:

Having a way to iterate over the coordinates of the points inside a shape would be useful. This would make it pretty easy to implement more advanced iterators:

```rust
// a filled red rectangle
let filled_rect = rect.points().map(|p| Pixel(p, Rgb565:::Red));
filled_rect.draw(&mut display);

// inset is a method that shrinks a Rectangle that by n pixels in each direction
let inside_area = rect.inset(1);

// a rectangle with a 1px green stroke
let stroked_rect = rect.points().filter(|p| !inside_area.contains(p)).map(|p| Pixel(p, Rgb565:::Green));
stroked_rect.draw(&mut display);
```

jacobrosenthal, 2019-12-21:
Oh II think related to #202 

I think Id like to be able to optionally iterate over the primitives, not the points of chained primitives.

Im thinking like a snake game or something. I want a fast way to see if a new point or shape is contained in the existing chain, and doing so on a point by point basis would be too onerous, but I think checking only bounding boxes would be possible.

rfuest, 2019-12-23:
> I think Id like to be able to optionally iterate over the primitives, not the points of chained primitives.

I'm not quite sure I understand your comment. Chaining is only used for drawing in e-g and there is no way to chain primitives before they are converted into an pixel iterator. Could you provide a small code example of what you would expect to be able to do?

jacobrosenthal, 2019-12-23:
psuedocode. Im thinking like a game event loop or something
```
....
    let mut lava = Circle::new(Point::new(64, 64), 64)
        .into_styled(PrimitiveStyle::with_stroke(BinaryColor::On, 1))
        .into_iter()
        .chain(
            &Line::new(Point::new(64, 64), Point::new(0, 64))
                .into_styled(PrimitiveStyle::with_stroke(BinaryColor::On, 1)),
        )
        .chain(
            &Line::new(Point::new(64, 64), Point::new(80, 80))
                .into_styled(PrimitiveStyle::with_stroke(BinaryColor::On, 1)),
        );

    lava.draw(&mut display);

    let guy_fella = Rectangle::new(Point::new(0, 0), Point::new(64, 64))
        .translate(Point::new(96 + 16, 16))
        .into_styled(stroke_off_fill_on);

    loop{

      let next = handle_fella();

       if (last.intersect(guy_fella){
           reset();
       }
 
        move_fella(&guy_fella,&next,&mut display)

        guy_fella = next;
    }
}
```

rfuest, 2019-12-23:
> if (last.intersect(guy_fella){

Where does `last` come from in this example?

jacobrosenthal, 2019-12-23:
Oops. supposed to be lava. fixed it. (also do you guys hang in irc or matrix or gitter or something, I hate to take over this entire issue with my question)

jamwaffles, 2019-12-25:
> (also do you guys hang in irc or matrix or gitter or something, I hate to take over this entire issue with my question)

Never used it, but I just opened [this Discord channel](https://discord.gg/JK7Dpbs). Not sure it's the right choice, but Rust itself appears to use Discord so let's start there and see how it works out.

rfuest, 2020-08-05:
I will close this issue because everything I intended to add is implemented. Any additional features for the primitive shapes can be discussed in other, more focused, issues.

closed 2020-08-05 by rfuest