embedded-graphics/embedded-graphics #157: Improved style property setting

jamwaffles, 2019-08-21:
From https://github.com/jamwaffles/embedded-graphics/issues/139#issuecomment-515026696:

> I'd be happy to explore a solution using `T: Into<Option<C>>` as suggested, as I think being able to pass `None` for a style property keeps the API cleaner and smaller than having `no_*()` methods.

To set a style:

```rust
thing.stroke(Rgb565::RED);
// or
thing.stroke(Some(Rgb565::RED));
```

To clear a style:

```rust
thing.stroke(None)
```

Using `Into<OptionC>>` keeps the API clean with one method per property, allows ergonomic use without `Some()` peppered everywhere but also supports un-setting the style by passing `None` in.

rfuest, 2019-08-21:
I came across the gtk-rs release notes yesterday, which contained some interesting points:
https://gtk-rs.org/blog/2019/06/22/new-release.html#stepping-back
The essentially just reverted back to using `Some` and `None`.

jamwaffles, 2019-08-25:
Hmm. Their API surface is a lot larger and more complex than e-g's. Perhaps we see how ergonomic it is for e-g and change it if there are lots of confusing errors from usage in the wild?

rfuest, 2019-08-26:
OK. The link was primarily a reminder that we should check that there are no side effects we didn't consider.

Should we implement a similar change for `Point`s, by using `impl Into<Point>`? Example from #158:
```rust
display.draw(
    Rectangle::new(point(0, 0), point(64, 64))
        .translate(point(96, 0))
        .stroke(Some(BinaryColor::On)),
);
```
```rust
display.draw(
    Rectangle::new((0, 0), (64, 64))
        .translate((96, 0))
        .stroke(BinaryColor::On),
);
```

jamwaffles, 2019-08-28:
Having thought about it some more, the gtk-rs post makes a very good point:

> The second one is to go back to some Rust fundamentals: being explicit.

I agree. It's non-obvious to the programmer that a style can be un-set by passing `None` (unless they read the docs). The ergonomics of not having to type `Some()` or `Point::new()` don't really outweigh this fact IMO. I'd suggest we leave the API as is for now. There might be a better solution come up with later on. I'll close this issue for now but I'm happy to reopen if there's a strong desire for a more concise API.

jamwaffles, 2019-08-28:
I don't think I saw anything, but can you confirm that #158 and #152 don't use `Into<Point>` anywhere?

closed 2019-08-28 by jamwaffles

rfuest, 2019-08-29:
No, they don't.

I suggested this because the code example above looked strange with the `point` function. But using `Point::new` instead does make it look better.

jamwaffles, 2019-08-29:
Yeah it does look odd I agree. Slightly better with macros, but now that the names are shorter I don't think these helpers are required at all :slightly_smiling_face:.