embedded-graphics/embedded-graphics #159: Examples in README.md are outdated

rfuest, 2019-08-25:
The examples need to be updated before the 0.6 release.

Is there a way to test examples in README.md using CI?

jamwaffles, 2019-08-25:
>  Is there a way to test examples in README.md using CI?

The readme could be generated from the `lib.rs` doc comment? Or there could be a bash script/`build.rs` that greps README.md for code blocks and compiles them into tests. There are a few solutions but they all have their own considerations and tradeoffs.

> The description of the nalgebra_support also needs to be updated.

I've assigned this to the 0.6.0 project as a reminder

jamwaffles, 2019-08-26:
That looks really good. It would be great to dedupe README.md and the docs in `lib.rs` into just README.md. Doesn't docs.rs already use nightly to build, so this could be a viable thing at release time also?

jamwaffles, 2019-08-26:
> But I wouldn't mind if stable doc builds were missing the front page and, for example, only contain a warning that this page is only available in nightly builds.

If the library works in stable, so should the docs IMO.

So one problem is duplication of docs between README and lib.rs, the other is testing examples in README to make sure they're up to date. Both problems can be solved temporarily by copying the built doc output of lib.rs into README.md. To me, that's a small price to pay until stuff like `external_doc` is stabilised. Thoughts?

rfuest, 2019-08-26:
There is a `external_doc` feature in nightly, which can be used to include the README.md into the docs and run the doc tests. Maybe we could use this feature on nightly builds only at the moment? This wouldn't be a perfect, because it doesn't test with the stable toolchain, but better than no tests at all.

rfuest, 2019-08-26:
I haven't released anything to docs.rs yet, so I don't know. 

If we use this, we should make sure that the docs can still be build locally using the stable toolchain. But I wouldn't mind if stable doc builds were missing the front page and, for example, only contain a warning that this page is only available in nightly builds.

rfuest, 2019-08-26:
That's probably the easiest solution until `external_doc` is available in stable.

jonas-schievink, 2019-08-28:
Just testing README.md isn't too difficult on stable: https://github.com/jonas-schievink/crate-template/blob/134c43fff451b6e30ddccae43f20de4dc6256893/template/src/readme.rs

I don't think this can be used to deduplicate the lib.rs documentation though.

jamwaffles, 2019-08-28:
> Just testing README.md isn't too difficult on stable

That's a neat trick, thanks for pointing it out.

> I don't think this can be used to deduplicate the lib.rs documentation though.

That's ok for now :slightly_smiling_face: - I'd rather have working copypasta examples than a perfectly up to date README. I think running README as a test would surface outdated bits as test failures anyway.

closed 2019-08-29 by jamwaffles

rfuest, 2019-08-29:
I just noticed this: https://github.com/jamwaffles/embedded-graphics/tree/master/embedded-graphics/src shows the content of readme.rs.

jamwaffles, 2019-08-29:
Very odd! Maybe github just picks anything up that matches `/readme/i`? At any rate, not much cause for concern I think

rfuest, 2019-08-30:
I think it's a bit confusing for people browsing the repo. Couldn't we just rename the file to something other than `readme.rs`, like `include_readme.rs`?
The file should also contain some comments to describe what it is used for.

jamwaffles, 2019-08-30:
Fixed in 353f8ad3f350c63237eadad3c753436dae76a198 🙂

PR #160 (2019-08-29): Test and fix README examples
This PR includes the contents of `README.md` as a doc comment, meaning the examples in the readme are tested in CI. It also fixes the woefully outdated and broken examples. Nalgebra description also updated.

Closes #159 

May thanks to @jonas-schievink for pointing out [this trick](https://github.com/jonas-schievink/crate-template/blob/134c43fff451b6e30ddccae43f20de4dc6256893/template/src/readme.rs).