embedded-graphics/embedded-graphics #162: Factor styling out of e-g objects

jamwaffles, 2019-09-01:
Carrying on with #139, this issue is to discuss refactoring how styling works to decouple styles from primitives. This ties in with #156 in allowing e-g objects to be useful for more than just drawing pixels.

So far, there's been a suggestion of something like the following (from https://github.com/jamwaffles/embedded-graphics/issues/139#issuecomment-515121637):

```rust
struct Styled<T, C> {
   primitive: T,
   style: Style<C>,
}
```

`IntoIterator` could then be implemented for e.g. `Styled<Rectangle>` instead of `Rectangle`. `Rectangle` itself would be unstyled by default.

The current issue with this approach as mentioned in https://github.com/jamwaffles/embedded-graphics/issues/139#issuecomment-525946121 is that it breaks chaining. I.e., this is no longer possible:

```rust
let objects = Circle::new(icoord!(64, 64), 64)
    .into_iter()
    .chain(Line::new(icoord!(64, 64), icoord!(0, 64))))
    .chain(Line::new(icoord!(64, 64), icoord!(80, 80))))
    .chain(
        Font6x8::render_str("Hello World!")
            .translate(icoord!(5, 50)),
    );

let styled = Styled::new(objects).stroke(Some(C1)).fill(Some(C0));
```

One possible solution to this is a "context" API as explored in #149.

jamwaffles, 2019-09-01:
How would style chaining work now? One option is `Styled::new(...)` but being able to chain a style right onto a primitive would be a great feature to retain.

I've [written up](https://play.rust-lang.org/?version=stable&mode=debug&edition=2018&gist=c278173b591e5486f262fe3bf5c9633c) a quick idea of how I think it could work. The crux of it is this:

Have a default impl for `WithStyle`:

```rust
trait WithStyle<C: PixelColor>: Sized {
    fn fill(self, fill: Option<C>) -> Styled<Self, C> {
        Styled {
            item: self,
            style: Style {
                fill,
                ..Style::default()
            }
        }
    }

    /* ... etc ... */
}
```

Then, implement `Styled`:

```rust
struct Styled<I, C> {
    item: I,
    style: Style<C>
}

impl<I: Drawable, C: PixelColor> Styled<I, C>  {
    fn fill(self, fill: Option<C>) -> Self {
        Self {
            item: self.item,
            style: Style {
                fill,
                ..self.style
            }
        }
    }

    /* ... etc ... */
}
```

Then implement your primitive of choice:

```rust
struct Circle {
    /* ... etc ... */
}

impl Drawable for Circle {}

impl<C: PixelColor> WithStyle<C> for Circle {}
```

So, assuming loads of things spelled out in [the above playground gist](https://play.rust-lang.org/?version=stable&mode=debug&edition=2018&gist=c278173b591e5486f262fe3bf5c9633c), the following remains possible:

```rust
let styled = circle.fill(Some(1u8)).stroke(Some(2u8));
```

What problems would this approach cause? It doesn't fix the chained iterator styling problem, but it does dedupe all the style application code quite nicely I think.

rfuest, 2019-09-01:
At first glance this looks like it might work and replicate the current behavior nicely.

I would probably change the `Styled` implementation to use `&mut self` and change the current style instead of creating a new style every time.

rfuest, 2019-09-02:
I did think about this some more and I'm not sure anymore if the `WithStyle` methods are a good idea. The name of the methods like `fill` doesn't reflect that the struct is converted to another type and this might be unexpected to the user. This is especially true if the primitves also have other methods which don't convert the object.

A more explicit approach might be better:
```rust
let styled = circle.into_styled().fill(Some(1u8)).stroke(Some(2u8));
```
This would also reduce the duplicated code in `WithStyle` and `Styled`.

jamwaffles, 2019-09-03:
That looks good. It's expandable too - I can see other `.into_*()` methods added later without much hassle. I think this might also fix the iteration part of #163 - the following would Just Work I think?

```rust
let objects = Circle::new(Point::new(64, 64), 64)
    .into_styled()
    .fill(Some(BinaryColor::On))
    .into_iter()
    .chain(
        Line::new(Point::new(64, 64), Point::new(0, 64))
            .into_styled()
            .stroke(Some(BinaryColor::Off)),
    )
    .chain(
        Line::new(Point::new(64, 64), Point::new(80, 80))
            .into_styled()
            .stroke(Some(BinaryColor::On)),
    )
    .chain(
        Font6x8::render_str("Hello World!")
            .translate(Point::new(5, 50))
            .into_styled()
            .stroke(Some(BinaryColor::On)),
    );
```

This raises 3 points though:

1. It's long af! The code above was passed through rustfmt so it's not me going mental with newlines. It's reasonably well grouped but there are a lot of chained method calls. Ah well.
2. I'd still like to be able to do both `.translate(...).into_styled()` _and_ `.into_styled().translate()`. I think this should be pretty simple to implement but it's worth raising.
3. Should we support `.into_iter()` on "raw" e-g objects, or should e-g require the programmer to convert the object into a richer type with `.into_styled()` (or `.into_mask()` or whatever in the future)?

rfuest, 2019-09-03:
> I think this might also fix the iteration part of #163 - the following would Just Work I think?

Yes, the code in the example would work. Chaining `Styled`s would work the same way as the old primitive drawables. But I'm not quite sure how this would help with #163.

> 1. It's long af! The code above was passed through rustfmt so it's not me going mental with newlines. It's reasonably well grouped but there are a lot of chained method calls. Ah well.

It is, but at least not much worse than before and decoupling styling from shapes should make it easier to improve this in the future.

Do you think that using chained iterators would be common in real applications? It's neat that you can write a function and return an `impl Iterator<Item=Pixel>`, but passing around a display or some kind of drawing context has the advantage that you can use the same drawing commands everywhere and don't have to deal with chaining.

> 2. I'd still like to be able to do both `.translate(...).into_styled()` _and_ `.into_styled().translate()`. I think this should be pretty simple to implement but it's worth raising.

Should `primitve.translate(...)` mutate the primitve or create a new one? At first glance I think having only a mutating version is fine because the other behavior can be achieved by cloning first: `primitive.clone().translate(...)`.

I don't like the duplication of the `translate` method for `Styled`. If a primitive also has other methods it might be not be obvious which methods are only available on the primitve itself and which one are passed through to the `Styled`. `Styled` could allow access to the wrapped primitive instead: `styled.primitive.translate(...)`, but this wouldn't be usable in code like your example.

Adding `translate` for `Iterator<Item = Pixel>` would also be useful. This would wrap the iterator in a `Translate` struct and add an offset in the `next` method. This would make  `styled.into_iter().translate(...)` possible.

> 3. Should we support `.into_iter()` on "raw" e-g objects, or should e-g require the programmer to convert the object into a richer type with `.into_styled()` (or `.into_mask()` or whatever in the future)?

Having a way to iterate over the coordinates of the points inside a shape would be useful. I wouldn't use `into_iter` for this but something more explicit like `points()`. This should make it pretty easy to implement more advanced iterators:
```rust
// a filled red rectangle
let filled_rect = rect.points().map(|p| Pixel(p, Rgb565:::Red));

// inset would be a method on Rectangle that shrinks it by n pixels in each direction
let inside_area = rect.inset(1);

// a rectangle with a 1px green stroke
let stroked_rect = rect.points().filter(|p| !inside_area.contains(p)).map(|p| Pixel(p, Rgb565:::Green));
```

jamwaffles, 2019-09-19:
> Do you think that using chained iterators would be common in real applications?

Honestly, not sure, but it's a feature I'd like to keep. I think it's useful for creating building block functions that create a complex component, a little like how React works. I think the code example in https://github.com/jamwaffles/embedded-graphics/issues/162#issuecomment-527360982 works well with the suggested `Styled` struct in this issue.

> Should `primitve.translate(...)` mutate the primitve or create a new one?

I suppose one could argue either way, but the current behaviour creates a new one (with `translate_mut()` being the mutating option). What are the drawbacks of leaving it as is? From what I've seen, having `<thing>()` and `<thing>_mut()` is a common Rust idiom.

> I don't like the duplication of the translate method for Styled. If a primitive also has other methods it might be not be obvious which methods are only available on the primitve itself and which one are passed through to the Styled

Ok, that's a fair point.

> This would make `styled.into_iter().translate(...)` possible.

That seems alright to me, but I think the best way to try this is to open a PR and see what it looks like.

> Having a way to iterate over the coordinates of the points inside a shape would be useful. I wouldn't use into_iter for this but something more explicit like points().

Makes sense to me. What would `points()` return? `impl Iterator<Item = Point>`?

rfuest, 2019-09-20:
> I suppose one could argue either way, but the current behaviour creates a new one (with `translate_mut()` being the mutating option). What are the drawbacks of leaving it as is? From what I've seen, having `<thing>()` and `<thing>_mut()` is a common Rust idiom.

It is a common Rust idiom and this also seems to fit the guidelines in https://doc.rust-lang.org/1.0.0/style/style/naming/ownership.html. But I'm not sure that I've seen this naming convention used exactly the way e-g uses it, that means one variant that implicitly copies and returns `Self` and the `_mut` variant that returns a `&mut Self` to allow chaining.

One drawback is that every drawable that implements `Transform` needs to implement both variants. To be consistent this would also apply to any future transformation traits like `Rotate`.
An advantage of choosing to only use the `_mut` variant is that it makes the `clone` explicit, which might be an expensive operation on an embedded platform for a complex drawable.

> That seems alright to me, but I think the best way to try this is to open a PR and see what it looks like.

Do you want to open a PR for this or do you want me to give this a try? I'm not sure when I would have time to work on this.

> What would `points()` return? `impl Iterator<Item = Point>`?

Yes.

jamwaffles, 2019-09-22:
Yeah implementing both `translate` and `translate_mut` is already a bit of a pain. I'd be in favour of a single mutating method called `translate`. I'm not sure it needs the `_mut` suffix, just something in the docs that mentions it mutates.

> Do you want to open a PR for this or do you want me to give this a try? I'm not sure when I would have time to work on this.

Same haha, I'll add a comment to this thread and assign it to myself if I do start working on it though.

rfuest, 2019-09-22:
> I'm not sure it needs the `_mut` suffix, just something in the docs that mentions it mutates.

If there is only one method the `_mut` suffix isn't necessary any more.

rfuest, 2019-11-30:
Closed, because #198 is merged and the remaining issues are added to #156 and #211.

closed 2019-11-30 by rfuest