embedded-graphics/embedded-graphics #163: Context API/retain chained object styling

jamwaffles, 2019-09-01:
Related to #162, #156 and styling in general, if a separate `Styled` struct is created as in #162, given the following chained iterator, how could it be styled?

```rust
let objects = Circle::new(icoord!(64, 64), 64)
    .into_iter()
    .chain(Line::new(icoord!(64, 64), icoord!(0, 64))))
    .chain(Line::new(icoord!(64, 64), icoord!(80, 80))))
    .chain(
        Font6x8::render_str("Hello World!")
            .translate(icoord!(5, 50)),
    );
```

Could each object's `IntoIterator` impl return a `Styled<Self, C>`? If so, then how would that work if no style has been explicitly set? `Styled::default()`?

jamwaffles, 2020-02-17:
Can this be closed now? I think we've got a pretty good styling solution now.

rfuest, 2020-02-17:
Yes we can close this.

Should I also close #149? I still like the way translation is handled by the context but it's outdated otherwise.

jamwaffles, 2020-02-17:
Yeah I think both can be closed. There's the (poorly named) #211 which can probably take the discussion now.

closed 2020-03-18 by jamwaffles