embedded-graphics/embedded-graphics #165: Usage with 1BPP, byte-packed image data

jamwaffles, 2019-09-02:
@rfuest:

In fixing the examples for jamwaffles/ssd1306#90, I couldn't find a way to use the current `rust.raw` image data into an e-g pixelcolor type. It's a 1BPP image where 8 pixel values are packed into a byte. We have `Gray2` but no `Gray1`. Should `Gray1` be added or is there a better/current but unknown to me way to support 1BPP packed images?

Cc #138

rfuest, 2019-09-02:
Did you try `BinaryColor`?

jamwaffles, 2019-09-02:
I haven't (don't have a device with me at the moment), but how would that work with 8 pixels packed into a byte?

rfuest, 2019-09-02:
`BinaryColor` has a storage type of `RawU1` which gets handled as 8 pixels per byte by `Image`.

jamwaffles, 2019-09-02:
Ah I see now. I looked through the code and the conversion happens in `RawDataIter`. I'll test with an actual display when I get home and see if it works ok.

jamwaffles, 2019-09-03:
Seems to work just fine, thanks for the help!

closed 2019-09-03 by jamwaffles