embedded-graphics/embedded-graphics #174: Allow `Point * Scalar` and `Point / Scalar`

jamwaffles, 2019-09-08:
A question from #171

As the title. Questions:

1. Should this be supported/allowed (`Mul`, `MulAssign`, `Div`, `DivAssign`):

```rust
let point = Point::new(10, 20);
let larj_point = point * 10;

let size = Size::new(5, 5);
let larj_point = point * size;

let larj_point = point * Point::new(7, 7);
```

And naturally with `/` also

2. Should the above only support integers, or floats as well?

rfuest, 2019-09-08:
1. Supporting `Point * i32`, `Point / i32`, `Size * u32` and `Size / u32` (and the `*=` and `/=` variants) is a good idea IMO.

   I assume `Point * Size` and `Point * Point` in your example would multiply componentwise? This would, for example, multiply a `x` coordinate width a `width` which is a little weird mathematically. Did you use this function in the clock example or do you have a special use case in mind?

2. Probably integers only.

jamwaffles, 2019-09-09:
So I think the original case for this was solved nicely by introducing a `polar()` function in the clock example in #171. You're right - the mul/div behaviours are a bit weird. I'll close this as the original issue I had is no longer relevant.

closed 2019-09-09 by jamwaffles