embedded-graphics/embedded-graphics #175: Extended display API

rfuest, 2019-09-08:
PR #171 adds clear functions to the simulator. It would be useful to have this function available for all display driver crates with e-g support, without having to use lower level APIs. 

This could be implemented by changing the `Drawing` trait to a `Display` trait that can provide additional functions:
```rust
pub trait Display<C>
where
    C: PixelColor,
{
    /// Draw an object from an iterator over its pixels
    fn draw<T>(&mut self, item: T)
    where
        T: IntoIterator<Item = drawable::Pixel<C>>;

    /// Returns the display size.
    fn size(&self) -> Size;

    /// Fills the display with a solid color.
    fn clear(&mut self, color: C);
}
```
The `size` function can be used to query the size of the display and layout elements independent of display size. This could also be used to provide a default implementation for `clear`:
```rust
fn clear(&mut self, color: C) {
   let r = Rectangle::new(Point::zero(), self.size()).fill(color);
   self.draw(r);
}
```
Which could be replaced by the driver crate if a more efficient way of clearing is available.

jamwaffles, 2019-09-09:
Sounds good to me, but a couple of things:

1. I think it should have a different name to `std::fmt::Display` (even though it's in a `no-std` context most of the time) to reduce conflicts. Maybe `DrawTarget` or `DrawingTarget`?
2. Is it worth either:
    * splitting `clear()` into `fn clear(&mut self)` and `fn fill_color(&mut self, color: C)`, or 
    * just having `fn clear(&mut self)`? 

    `clear`ing a display with a fill colour seems counterintuitive to me.

rfuest, 2019-09-10:
1. Good point. I think `DrawTarget` is OK and more generic than `Display`, for example for drawing to a image file.
2. IMO it's useful to have a way to fill the display with any color. If, for example, a UI uses a gray background a `clear` method that only supports black would be useless.

jamwaffles, 2019-09-10:
1. Cool and good
2. Absolutely, my comment was about whether to split clearing the display (filling with transparency) and filling with a colour into to separate methods, or leaving it as a single `clear()`. Calling `clear()` and having the display _fill_ with colour instead of _clearing_ seems counterintuitive.

rfuest, 2019-09-14:
I don't think there are many benefits of having a separate `clear` and `fill` method. Explicitly requiring the user to specify the color has the advantage that it works the same way as other fills and doesn't require an additional convention which color is used if none is specified.

The name `clear` is also used in other APIs. OpenGL uses `glClear` to fill the color buffer with the color specified by `glClearColor` and GDI+ has a `Graphics::Clear(color)` method. But we could also use something different if you prefer. IMO `fill_color()` is also confusing because we just introduced a setter with the same name.

jamwaffles, 2019-09-14:
That all makes sense :slightly_smiling_face:. Let's go with `.clear()` and `DrawTarget` :+1:.

jamwaffles, 2019-09-16:
I just found something in the [SSD1331 datasheet](https://cdn-shop.adafruit.com/datasheets/SSD1331_1.2.pdf) (section 9.2.5): it's possible to clear only part of the display. How could we support this? Could there be `.clear_area(/* ... */)` which has a default impl that just calls `.clear()`?

rfuest, 2019-09-16:
>  Could there be `.clear_area(/* ... */)` which has a default impl that just calls `.clear()`?

But this would clear the whole display if `clear_area` isn't supported. IMO the default impl shouldn't have a different behavior.

> How could we support this?

Should we support this at all or leave this to the display driver?
If we would implement hardware accelerated drawing (as suggested in #185) the driver could detect when a black rectangle is drawn and use the `Clear Window` to accelerate this.

jamwaffles, 2019-09-16:
Ah yeah I got that backwards didn't I, never mind.

> Should we support this at all or leave this to the display driver?

I'm erring on the side of the display driver for this one. There's a lot of variance in which primitives are supported (see my SSD1331 commetns on #185) in the ecosystem as well as behavioural changes when stroke widths, etc are added. The SSD1331 also doesn't support stroke widths for lines (although it could be kludged for rectangles).

On top of that, I think it would be useful to have some real-world implementations to refer to if moving this stuff into e-g becomes feasible.

jamwaffles, 2019-09-19:
> I'm erring on the side of the display driver for this one.

I take that back. See below:

I just typed out https://github.com/jamwaffles/embedded-graphics/issues/185#issuecomment-533054346. @ReeceStevens makes some good points in #185 overall, so what do you think about extending the `DrawTarget` trait with some provided methods to draw each e-g primitive? All methods would have a default impl that just calls `.draw()` as normal, but can be overridden by display drivers to match any hardware acceleration that they support. For example:

```rust
trait DrawTarget {
  fn draw(&self, thing: IntoIterator<Item = Pixel<C>>);

  fn draw_rectangle(&self, rectangle: Rectangle) {
    self.draw(rectangle.into_iter());
  }

  // etc ...
}
```

Then, for a hypothetical display that can draw hardware accelerated rectangles, but only if they have no stroke:

```rust
impl DrawTarget for MyDisplay {
  fn draw(&self, thing: IntoIterator<Item = Pixel<C>>) {
    // ...
  }

  fn draw_rectangle(&self, rectangle: Rectangle) {
    if rectangle.style.stroke.is_some() {
      self.draw(rectangle);
    } else if let Some(fill) = rectangle.style.fill {
      self.draw_hardware_accelerated_rect(
        rectangle.top_left(), 
        rectangle.bottom_right(), 
        fill
      );
    }
  }

  // Default impls mean that no other methods are required here
}
```

There are a bunch of assumptions in the above pseudocode, but hopefully it makes my point clear. This is also relevant to #162 I think.

This approach means that displays can implement partial support or not support at all for hardware acceleration, whilst keeping the API and drawing behaviour consistent across the ecosystem. Adding methods for new primitives (like polylines, etc) in the future doesn't break drivers because a default impl is provided that calls to `self.draw()`.

rfuest, 2019-09-20:
This would require the user to call the right method to get hardware acceleration:
```rust
display.draw_rectangle(rect); // is accelerated
display.draw(rect); // is not accelerated
```
That mistake is easy to make IMO, I would prefer there to be only one way to draw each primitive.

jamwaffles, 2019-09-22:
That's a good point, but I'm not sure there's a solid way around it. For a start, maybe `draw()` should be renamed to `draw_iter()` to make it clearer? The docs could explain some of the underlying differences between `draw_iter()` (never accelerated due to type erasure) and others. `draw_rectangle()`, etc aren't guaranteed to be accelerated by the driver either, and even then may only accelerate styles that the hardware supports. This could be clarified by the e-g docs, as well as individual drivers listing what they can and cannot hardware accelerate.

If you were going to suggest removing `draw()`/`draw_iter()`, I would very much like to keep the iterator-based API as it's a USP of this crate which I think is useful. Just trying to preempt that line of discussion :wink:

rfuest, 2019-09-22:
> If you were going to suggest removing `draw()`/`draw_iter()`, I would very much like to keep the iterator-based API as it's a USP of this crate which I think is useful. Just trying to preempt that line of discussion wink

I did get that from previous discussions and was never suggesting to remove that feature.

> For a start, maybe draw() should be renamed to draw_iter() to make it clearer?

I agree with that.

> That's a good point, but I'm not sure there's a solid way around it.

One way to solve this would be to reverse the way the `draw` calls work:
```rust
rect.draw(draw_target);
circle.draw(draw_target);
```
The `draw` methods in the primitives could then call the appropriate `draw_...` method.

ReeceStevens, 2019-09-29:
Hey @jamwaffles and @rfuest, sorry for the delay in following up on this line of discussion-- I haven't had an opportunity to respond till now. I was thinking about this problem this past week; it seems like the ideal use case would be that a user just calls a single `draw()` method, regardless of whether `any_drawable_object` is a primitive or some custom drawable object, and accelerated methods would be called behind the scenes if supplied. 

As @rfuest suggested, the calling convention could be reversed so that `.draw()` is a method of the drawable object rather than the `DrawTarget`. I suspect this could be a great way to manage this situation; primitives will have a `draw()` method defined that calls their specific `draw_*()` methods on the `DrawTarget` under-the-hood.

It might be beneficial to take this one step further: make `draw()` a required method for the `Drawable` trait. This seems like the best of both worlds: for primitives, hardware acceleration methods can be used. For higher-order drawable objects, a user can put together hardware accelerated methods to draw more complex graphics objects. For example:

```rust
struct Button {
    text: &'static str;
    rect: Rectangle;
}

impl Drawable for Button {
    fn draw<T: DrawTarget>(&self, target: T) {
        self.rect.draw(target); // under the hood, calls `draw_rectangle()`
        self.text.draw(target);
    }
}
```

I'm not sure it would be possible to take advantage of HW accelerated drawing functions in higher-order graphical objects if the draw method were called from the display, but making it a trait method for `Drawable` seems like a natural way to let users take advantage of these optimizations transparently.

jamwaffles, 2019-10-02:
My initial reaction was "wow, that turns the API completely on its head" but it does make a lot of sense having thought about it. Thank you for your ideas!

> It might be beneficial to take this one step further: make draw() a required method for the Drawable trait.

I like that idea and I like your example (although I think your `text` would have to be drawn like `Text6x8::render_str(self.text).draw(target)`, no?). Should it also be `target: &mut T` so as not to not take ownership, and allow the display to be mutated?

Could `Drawable` be implemented for `Iterator` and/or `IntoIterator` as well? I'm struggling to think of the bound right now, but that would still give one method that can draw everything, iterator or not. As a last resort, we could fall back to `.draw()` and `.draw_iter()` but it would be good to keep to a single method to reduce confusion. I'll have a play with this when I've got a minute.

jamwaffles, 2019-10-02:
Here's a first pass at just the `.draw()` method (renamed to `.draw_exp()` to prevent conflicts in testing:

```rust
pub trait ExperimentalDrawing<D, C>
where
    C: PixelColor,
    D: Drawing<C>,
{
    fn draw_exp(self, display: &mut D);
}

impl<C, D> ExperimentalDrawing<D, C> for crate::primitives::Rectangle<C>
where
    C: PixelColor,
    D: Drawing<C>,
{
    fn draw_exp(self, display: &mut D) {
        display.draw(self.into_iter())
    }
}

impl<T, C, D> ExperimentalDrawing<D, C> for T
where
    T: Iterator<Item = crate::drawable::Pixel<C>>,
    C: PixelColor,
    D: Drawing<C>,
{
    fn draw_exp(self, display: &mut D) {
        display.draw(self)
    }
}
```

`draw()` has access to `self` so can choose to use hardware acceleration when available.

I would much prefer to use `&self`, however I get this error:

```
$ cargo build
   Compiling embedded-graphics v0.6.0-alpha.2 (/home/james/Repositories/embedded-graphics/embedded-graphics)
error[E0277]: `&T` is not an iterator
   --> embedded-graphics/src/lib.rs:278:17
    |
278 |         display.draw(self)
    |                 ^^^^ `&T` is not an iterator
    |
    = help: the trait `core::iter::Iterator` is not implemented for `&T`
    = note: required because of the requirements on the impl of `core::iter::IntoIterator` for `&T`

error: aborting due to previous error

For more information about this error, try `rustc --explain E0277`.
error: Could not compile `embedded-graphics`.

To learn more, run the command again with --verbose.
```

Any ideas? If this is a decent approach I'll open a PR instead of spamming this issue.

ReeceStevens, 2019-10-02:
That seems like a great idea. I have actually got an implementation of this on my side as well, I was working on it Monday but didn’t quite get it tested in time for a PR that evening. I could put it up as a WIP PR for discussion? I won’t be heartbroken if we tear it up :)

RE the iterator error, I ran into that as well. You have to specify the following trait bound:

`for<‘b> &’b Self: IntoIterator<Item=Pixel<C>>`

Where ‘b is an arbitrary lifetime. I hadn’t heard of the `for<‘_>` syntax before, took a while to find.

jamwaffles, 2019-10-02:
> I could put it up as a WIP PR for discussion?

Yes please! Much better to talk about concrete code than lil snippets for sure.

> RE the iterator error, I ran into that as well. You have to specify the following trait bound:

Ah, well done for figuring that out. I was getting sad we couldn't use iterators in the same API.

rfuest, 2019-10-04:
Thanks @ReeceStevens for the WIP PR. Af first I wasn't sure if I would like the `draw` API change that I suggested. But seeing this in real code, like the `fill.rs` example, reminds me of the builder pattern and that seems to work fine.

@ReeceStevens I did some quick experiments with the `Drawable` type and came up with this:

In `drawable.rs`:
```rust
pub trait Drawable<C>
where
    C: PixelColor,
{
    fn draw<T: DrawTarget<C>>(self, display: &mut T);
}

impl<C, I> Drawable<C> for &mut I
where
    C: PixelColor,
    I: Iterator<Item=Pixel<C>>,
{
    fn draw<T: DrawTarget<C>>(self, display: &mut T) {
        display.draw_iter(self);
    }
}
```

In the primitives:
```rust
impl<C> Drawable<C> for &Rectangle<C> where C: PixelColor {
    fn draw<T: DrawTarget<C>>(self, display: &mut T) {
        display.draw_rectangle(self);
    }
}
```

This way I was able to remove the `IntoIterator` implementations for `&mut Primitive`. But this was only a quick experiment and I have no idea if this breaks anywhere else without spending some more time with the code.

jamwaffles, 2019-10-20:
Closed by #187

closed 2019-10-20 by jamwaffles