embedded-graphics/embedded-graphics #181: Better alternative to SizedDrawing

jamwaffles, 2019-09-14:
#180 removes the ill-designed `SizedDrawing` trait. The intent of this trait was to allow partial updates for displays that supported it. This issue is to discuss a better alternative.

As I see it, a partial update requires two things:

1. A way of specifying the rectangle of pixels to update (probably a top left/bottom right coord pair)
2. Contiguous pixel data to send to this area (fills the entire area instead of only returning filled pixels from the iterator).

My questions off the top of my head are:

1. Is top left/bottom right the correct way to specify the rect to update for most/all/all known supported displays? Is there a better/more generally applicable way to do it?
2. The pixel data mentioned in (2.) above will return coordinates. Should they be relative to the defined area, or remain relative to the display origin?
3. Does the pixel data really need to be contiguous? I think it does but I might be missing an important detail.

birkenfeld, 2020-01-11:
Some way of doing this is definitely needed.

The new methods on DrawTarget for supporting acceleration are nice, but in many cases (e.g. ST7735) writing a contiguous stream of pixels is much faster than writing pixel by pixel, which requires sending the pixel address command before each color word.

This interface should be used for e.g. text rendering.

jamwaffles, 2020-02-04:
There might be some missing pieces in this idea that e-g needs to fill in, but could you use [`set_address_window()`](https://github.com/sajattack/st7735-lcd-rs/blob/master/src/lib.rs#L165) and send a stream of bytes out for e-g objects? Obviously this relies on being able to get the size of said object, but assuming you can would that work?

closed 2020-06-04 by jamwaffles