embedded-graphics/embedded-graphics #182: Coordinate system

rfuest, 2019-09-14:
We started discussing e-g's coordinate system in issue #142, but the one by off error is only one aspect of this, so I decided to open another issue for a more general discussion.

Quick recap: The issue is to determine how e-g should handle coordinates. Being limited to integer values allows two interpretations: Points can either lie in the center of pixels or on the grid line between them.

In the rest of this post I'll try describe a relatively consistent way of handling coordinates using points in the center of pixels. I'm pretty sure that this might be somewhat confusing (at least I was confused while writing this), but I hope the images make it clearer.

Drawing lines with odd widths works without any special considerations. Lines with even width are drawn with one pixel more on the left side (looking from the start to the end point). By definition lines with zero length will still draw a dot.
[image]

Rectangles are defined by their top left corner and size. The actual shape, that is described by this, lies on the edge of the top left pixel (has a half pixel offset).
To replicate the current behavior, non zero sized rectangles can also be constructed by an additional constructor that takes a top left and bottom right corner. Passing the same point as top left and bottom right corner to this constructor would result in 1px * 1px rectangle.
[image]

Strokes on rectangles are, by default, drawn by using the path shown in the following image. This makes them work consistently with lines:
[image]

Strokes on circles work the same way. The path to draw the stroke is not shown in this image, but it is defined to be half a pixel inside the outer edge and going clockwise.

[image]

To replicate the way strokes work currently on rectangles and circles a `stroke_alignment` option can be added to the style. The default centered alignment is consistent with the way strokes work on lines. The stroke alignment only affects closed shapes.

[image]
[image]

Problems:
- Because zero sized rectangles don't have a valid bottom right corner anymore the `bottom_right` getter won't work for all rectangles. One possible solution would be to return an `Option<Point>` instead.
- Polygons might need some additional considerations.
- The stroke can extend beyond the outside edge of a shape, which needs to be considered when e.g. implementing `SizedDrawing`.

jamwaffles, 2019-09-14:
I'll type up a longer reply when I've had a think, but what's the reasoning behind choosing pixel centers over their top-left corner? I'd also like to keep the current behaviour of filling inwards from the outer edge, but the offset could be a configurable option, no?

rfuest, 2019-09-14:
> I'll type up a longer reply when I've had a think, but what's the reasoning behind choosing pixel centers over their top-left corner?

Both variants have their own set of weird edge cases which need to deal with the half pixel offset and I was changing my mind a lot while I thought about this. Seeing the animated second hand in your clock example made me wonder how to correctly handle drawing 1px lines if they are defined by the grid lines. The line wouldn't rotate around the center of the starting pixel but the intersections of the grid lines, which would cause the starting pixel to shift. This could be fixed by introducing some kind of offset, but I think the center variant is easier to understand overall.

> I'd also like to keep the current behaviour of filling inwards from the outer edge, but the offset could be a configurable option, no?

I would prefer that the drawing operations work consistently in the default configuration.

almindor, 2020-02-25:
I just ran into this problem when optimizing the new [ST7789](https://github.com/almindor/st7789) driver using latest beta.

When using `Rectangle` the dimensions are inclusive, e.g. `bottom_right()` points to pixels that should still be rendered, this means that on a rect with 0, 0 -> 10,10 bounds you end up with 10x10 pixels being filled.

When using `Image` however the `Dimensions` use the "+ width and height" approach and `bottom_right()` points to one pixel to the bottom and right of the last pixel to be drawn.

This means that with `Image` of width/height 10/10 you end up with 11 * 11 pixels being considered for drawing if using `Dimensions` (e.g. in a driver to optimize frame based rendering).

It can be worked around in the driver if known but it makes it very odd to need to do `bottom_right() - Size::one()` in one case for the render frame for an image and direct `botton_right()` for render frame boundaries when doing a rect fill.

I think `bottom_right()` needs to be inclusive, as it should point to the "last pixel in the boundary" so to speak.

jamwaffles, 2020-03-18:
> The default centered alignment is consistent with the way strokes work on lines. The stroke alignment only affects closed shapes.

In implementing #271, the algorithm is conducive to changing which side of the line the stroke is drawn on. It might be useful to support this for open lines, although naming becomes more difficult (inside/outside/center vs left/right/center). Thoughts?

As for @almindor's points, yeah that sucks :( pixel and dimension behaviour should've been designed better from the start, so my bad. I think this issue will solve or at least begin to solve the inconsistency around behaviours. This should happen in 0.7.

rfuest, 2020-03-18:
If I remember correctly most (all?) APIs I've checked while researching for the initial post only apply the line positioning to closed shapes and not open lines. I can't think of a good name that can be used for both cases.

jamwaffles, 2020-03-18:
Ok, that's fair. Maybe worth adding if someone asks for line offsets in the future.

jamwaffles, 2020-03-23:
Hit testing has come up in a few places, which made me think: should primitives with a thick stroke include that stroke width in their hit testable area? I solution might be to implement whatever hit testing method we settle on for both styled and unstyled items?

rfuest, 2020-03-23:
I was thinking about adding functions to get the border and fill areas for styled primitives. These methods return a primitive that is offset by the proper amount to enclose all border pixels or fill pixels.

By using one of these areas for hit testing it's possible to decide if the stroke should be considered for hit testing or not. Another use for these function is implementing hardware accelerated drawing routines.

jamwaffles, 2020-08-10:
Can this be closed now? Your comment about stroke/border dimensions I think are covered in a checklist item in #314 ("Dimension for styled primitive should consider stroke width and alignment") so I think this issue is done.

rfuest, 2020-08-10:
OK, with #314 and #354 open we don't need to keep this open any longer.

closed 2020-08-10 by rfuest