embedded-graphics/embedded-graphics #185: Allow optional usage of hardware accelerated drawing methods

ReeceStevens, 2019-09-15:
This seems like it might fall into the related category of #175, but that issue seemed to be focused on `.clear()` in particular.

Much in the way that traits provide default implementations of trait methods that can be overridden, it would be very helpful to provide a way to supply hardware-specific drawing methods for certain primitives (in particular: rectangles, circles, and lines) instead of using the default pixel-by-pixel iterator approach. The speed increase from using these methods can be very significant (see calculations below for a concrete example of that).

I have been looking at ways to hook in the HW-accelerated methods from the implementing library's side, but so far have not found an effective way to do so. I thought I could possibly distinguish certain primitives in the impl for `draw()` and invoke the HW method in those cases, but I can't find a way to determine the type of `item` at runtime (I suspect it is erased anyways).

## Possible implementation strategy

I want to preference this by saying I am not tied to this implementation as the solution to the issue; I just want to put forth a first suggestion as a starting point for a conversation about the feature.

It seems like one way this could be done is by changing the core `Drawing` impl function from `draw`, which describes how to draw an entire drawable object, to just `draw_pixel`. It seems like most implementations of draw will be copy-pasting the iterator behavior, and the unique portion of each driver would be how it implements the actual painting of the pixel on the screen. This could provide the opportunity to add hooks into the `Drawing` trait to add accelerated drawing functions if they are supplied by the platform; something like

```rust
pub trait Drawing<C>
where
    C: PixelColor,
{

    /// Draw a pixel
    fn draw_pixel(&mut self, pixel: drawable::Pixel<C>);

    /// Draw an object from an iterator over its pixels
    /// and use a specific drawing method if available.
    fn draw<T>(&mut self, item: T)
    where
        T: IntoIterator<Item = drawable::Pixel<C>>
    {        
        // TODO: pseudo-code below, adjust these trait bounds and determine
        // if there is a more-specific draw method we can use for the given item
        match item {
            Circle => self.draw_circle(),
            Rectangle => self.draw_rectangle(),
            _ => self.draw_generic()
        }
    };

    /// Draw an object from an iterator over its pixels
    fn draw_generic<T>(&mut self, item: T)
    where
        T: IntoIterator<Item = drawable::Pixel<C>>
    {
        // TODO: adjust these trait bounds and determine
        // if there is a more-specific draw method we can use for the given item
        for pixel in item {
            self.draw_pixel(pixel);
        }
    };

    /// Draw a circle. Override with possible HW-accelerated method
    fn draw_circle(&mut self, item: Circle) {
        self.draw();
    };

    /// Draw a rectangle. Override with possible HW-accelerated method
    fn draw_rect(&mut self, item: Rectangle) {
        self.draw();
    };
}
```

I am not familiar with all of the design decisions that went in to this library, so I apologize if this is an uninformed suggestion! And of course, if this is something that does not fit into the roadmap, I understand. However, if it does seem like a feature you would like added, I would be happy to work on it. It's pretty critical for my use case so I am motivated :)

## Concrete use case motivation

Particularly for larger displays, drawing pixel-by-pixel quickly becomes a speed limiting factor. My concrete use case: I am working with a display that is 800x480, and drawing a rectangle to fill that screen pixel-by-pixel would take quite a large amount of time. Specifically, with an SPI baud rate of 3.2 MHz:

```
(800 * 480 pixels) * (13 SPI transfers per pixel) * (8 bits per transfer) / (3200000 bits per second) = 12.48 seconds
```
13 transfers per pixel seems like a lot, but unfortunately that is the interface exposed by the device. However, it also provides a hardware-accelerated method of drawing a rectangle requiring only 50 transfers.

```
(50 transfers) (8 bits per transfer) / (3200000 bits per second) = 0.125 milliseconds
```

jamwaffles, 2019-09-16:
Thanks for taking the time to write all this out. I agree that a way to leverage hardware drawing functionality is important, however there are some issues that come to mind. For example, the [SSD1331](https://cdn-shop.adafruit.com/datasheets/SSD1331_1.2.pdf) display only has hardware functionality for drawing lines and rectangles (see p. 32 in the linked datasheet), with no support for line/border thickness. Because there's such a range of displays out there, I think this functionality might be better left to the display driver itself. You could add methods to the driver's interface that take a `Rectangle`, `Circle`, etc and extract the appropriate properties for each primitive to send to the display.

For your use case specifically, two questions come to mind:

1. Why 13 transfers per pixel? That seems like a lot of over head for this particular display.
2. Can you tell the driver to only update a sub-area of the display? The `SizedDrawing` trait was supposed to help with this, but there are problems with it in its current form so it's being removed in #180. We'll add it back in when the design is better thought out, at which point it would reduce the data transfer required for non-full-screen updates. There's some discussion going on in #181.

ReeceStevens, 2019-09-19:
Sorry for the delayed response on my end, it's been a busy week. Thanks for the prompt and thorough response!

> Because there's such a range of displays out there, I think this functionality might be better left to the display driver itself.

I am still relatively new to Rust, so I might be missing something regarding ways to use this API. However, my understanding was that because the trait method for `Drawing` is `draw()`, the method that is directly used to render a `Drawable`, there was no way for an implementing library to detect primitives like `Rectangle` or `Circle` and invoke the appropriate driver-specific drawing function. In my attempts to implement this primitive detection, I was unable to distinguish the concrete type of `item` beyond `IntoIterator<Item = drawable::Pixel<C>>`, so I could not pick up on when drawing primitives were passed in.

```rust
    /// Draw an object from an iterator over its pixels
    fn draw<T>(&mut self, item: T)
    where
        T: IntoIterator<Item = drawable::Pixel<C>>
    {
        // TODO: is there a way to determine the concrete type of `item`? 
    };
```
When you suggest adding primitive-specific methods to the driver's interface, do you mean creating methods directly for the specific device driver and outside of this API? While that would allow usage of these HW-accelerated functions, it seems like that also takes away a significant portion of the value of using an interface such as this. In particular, it would prevent switching out the display driver for a different `embedded-graphics` compatible driver since other drivers may not support the same accelerated methods, and it would also prevent using the simulator to test the display logic.

> For example, the SSD1331 display only has hardware functionality for drawing lines and rectangles (see p. 32 in the linked datasheet), with no support for line/border thickness.

It seems like if the trait implementation method were `draw_pixel`, and other primitives like `draw_circle` or `draw_rectangle` had a default implementation of `draw_generic`, the current API behavior would be maintained. In that case, a display such as SSD1131 would not provide any alternative implementation for `draw_circle` or `draw_rectangle`, and such primitives would be rendered using the same iterator approach as is used now. However, if a display had acceleration functions, the driver could supply those as an optional trait implementation for `draw_circle` or `draw_rectangle`.

(I'm using the function names from that original snippet I posted above; also, I'm using `draw_circle` and `draw_rectangle` as examples of graphics primitives that could have HW support, but I'm sure all of the primitives defined in `embedded_graphics::primitives` could fall in that category)

Does this make sense? If it would help the conversation to put together a proof-of-concept, I would be happy to do that. Also, I realize my responses are getting a bit lengthy, so thank you for your patience in reading through all of this :) 

## Specific use case questions

> Why 13 transfers per pixel? That seems like a lot of over head for this particular display.

That's a good question, I thought the same thing. Here's how those transfers break down:

```
writeReg: 2 transfers, one for register and one for data
writeCommand: 2 transfers, one for "COMMAND_MODE" and one for the command
SPI->transfer: 1 transfer, direct function call
```
```
// Draw pixel method for RA8875 display
  2        writeReg(RA8875_CURH0, x); 
  2        writeReg(RA8875_CURH1, x >> 8);
  2        writeReg(RA8875_CURV0, y);
  2        writeReg(RA8875_CURV1, y >> 8);  
  2        writeCommand(RA8875_MRWC);
  0        digitalWrite(_cs, LOW);
  1        SPI->transfer(RA8875_DATAWRITE);
  1        SPI->transfer(color >> 8);
  1        SPI->transfer(color);
  0        digitalWrite(_cs, HIGH);
------------------------------
  13
```

> Can you tell the driver to only update a sub-area of the display?

I _think_ so? I'm actually not sure on this one and would need to do some more digging to figure that one out. I think that by default, the RA8875 driver chip does partial display updates.

jamwaffles, 2019-09-19:
>  there was no way for an implementing library to detect primitives like Rectangle or Circle and invoke the appropriate driver-specific drawing function. 

Correct, as far as I know. This is of course solved by the `draw_*()` methods taking a concrete type.

> When you suggest adding primitive-specific methods to the driver's interface, do you mean creating methods directly for the specific device driver and outside of this API? ... In particular, it would prevent switching out the display driver for a different embedded-graphics compatible driver

That's a good point. I was originally against including these methods in e-g itself as it's more API surface to maintain, but tbh it's not much extra work. I'm definitely warming to the idea of adding e-g primitive drawing methods to `Drawing` (or `DrawTarget` in #175). Adding default impls certainly eases the boilerplate for driver implementers too, particularly when new primitive types are added (e.g. #77).

> It seems like if the trait implementation method were draw_pixel, and other primitives like draw_circle or draw_rectangle had a default implementation of draw_generic, the current API behavior would be maintained.

Yes, default trait impls that defer to drawing bitmaps would probably work. It would also have to check the style of the object passed in if the hardware doesn't support e.g. circles with thick borders, but that shouldn't be hard to do if we factor this use case in with #162.

> If it would help the conversation to put together a proof-of-concept, I would be happy to do that.

That would be great! Could you add these methods to your display driver to see what it looks like? There might be some edge cases we haven't covered. I'll see if I can find some time to experiment with the SSD1331 in the same way.

> I realize my responses are getting a bit lengthy, so thank you for your patience in reading through all of this :)

No problem :). I'd rather have more info than "it dusnt werk" lol

---

Thanks for explaining the transfers. I've been looking through the [datasheet](http://www.buydisplay.com/download/ic/RA8875.pdf) and the [Adafruit driver](https://github.com/adafruit/Adafruit_RA8875/blob/master/Adafruit_RA8875.cpp#L678) and it looks like you might be able to do chunked transfers of some description? I couldn't find any clarification in the datasheet, but that would mean the 8 transfers to set the pixel coordinates would have to be called far less. YMMV - I don't have a physical display to test on, but that might help your overhead troubles.

jamwaffles, 2019-09-19:
I'm going to soft-deprecate this issue in favour of #175. Let's continue the discussion there around adding more methods to `DrawTarget`, as I think hardware accelerated commands is a sub-feature of that. Let me know if you don't think that's a good idea.

ReeceStevens, 2019-10-01:
@jamwaffles I think it makes sense to move the conversation to #175 . Feel free to deprecate this issue. Thanks for the response!

jamwaffles, 2019-10-20:
Closed by #187

closed 2019-10-20 by jamwaffles