embedded-graphics/embedded-graphics #188: Difficulties implementing custom fonts

lukesutton, 2019-10-03:
- Version of embedded-graphics in use (if applicable): 0.6.0-alpha.2
- Target: Simulator
- If using real hardware: N/A

I'm attempting to implement my own custom font using the `FontBuilderConf` trait. Firstly, this is much simpler than earlier versions, which is great to see! 

However, I'm having some difficulty understanding how it actually works. I know i need to implement the `char_offset` function, but it's not obvious what it's doing. What is this index that it returns? How is that translated into byte array in the conf?

Essentially I'm having difficulty translating the layout of fonts in the data files into an implementation of `char_offset`. Each of the existing fonts seems to have a different layout and `char_offset` implementation, so unfortunately I couldn't glean much from them.

I think the documentation here needs to be improved. I'd be happy to contribute there, but I would need some assistance in understanding it.

rfuest, 2019-10-03:
The `char_offset` function converts a Unicode `char` to the position where that glyph is stored in the font image. For example: If the first glyph in the upper left corner of the font image is an `'A'` calling `char_offset('A')` would return `0`. And if the next glyph to the right is `'B'` it would return `1`.

The translation to indices into the `FONT_IMAGE` array is done in `FontBuilderIterator::next`. The glyphs in the font image are stored starting width glyph `0` in the top left corner and ascending indices to the right and bottom. The rows of glyphs are wrapping after `FONT_IMAGE_WIDTH` pixels.

I'm not sure this description was clear enough, but comparing the font images in `embedded_graphics/data/font*.png` with the return values of the `char_offset` methods should make it clearer how this works is practice. 

> I think the documentation here needs to be improved. I'd be happy to contribute there, but I would need some assistance in understanding it.

Yes, the documentation isn't great. I'm not sure if the code will change in the future but help with better documentation is always appreciated. This could also make a great code example in the simulator, like a simple seven segment font.

jamwaffles, 2019-10-03:
> This could also make a great code example in the simulator, like a simple seven segment font.

That's a great idea. I'd like to create a short guide in the Rust docs and a longer, tutorial-style article on [my blog](https://wapl.es) at some point, maybe after the release of 0.6.0. I think implementing a 7 segment font would be a good target for the guide. Plenty of pictures to explain the index offsets too!

If nobody else wants to write that guide up I'm happy to have a go. I'd open a PR to review so I'm not missing anything from @lukesutton's point of view.

rfuest, 2019-10-03:
@jamwaffles While I was looking at the code to write my reply I thought that it might be simplified by removing the wrapping feature. Having only one row of glyphs would simplify the code and also make the relation of char index and glyph position inside the font image clearer. What is your opinion on this?

jamwaffles, 2019-10-03:
I'd prefer to keep it:

* Existing fonts like [those on the Uzebox wiki](http://www.uzebox.org/wiki/index.php?title=Font_Bitmaps) can still be used.
* It (IMO) doesn't add much complexity.
* Font glyph maps are easier to view/debug/inspect due to their squarer proportions
* I don't think performance is an issue - happy to be corrected

rfuest, 2019-10-03:
OK, fair points.

lukesutton, 2019-10-04:
@rfuest Thank you for that explanation. That makes sense to me. I'll have another go at my implementation. 

@jamwaffles Based on that I'm happy to have a go at writing a short guide for the docs.

> While I was looking at the code to write my reply I thought that it might be simplified by removing the wrapping feature. 

That would make it easier to understand, but I do agree with @jamwaffles' points.

jamwaffles, 2019-12-17:
@rfuest added an example of a custom seven segment font [here](https://github.com/jamwaffles/embedded-graphics/tree/master/simulator/examples/custom-font) a while ago. Like all the examples, it needs some better documentation in the header, but it might be enough to get you started.

I still want to do a walkthrough blog post at some point as well. Just haven't got round to it yet :sweat_smile:

jamwaffles, 2020-03-18:
I'm gonna close this issue as there's a custom font example in the simulator, and I have a custom font tutorial on my personal blog posts TODO anyway, so I'll pick that up... at... some point.

closed 2020-03-18 by jamwaffles