embedded-graphics/embedded-graphics #195: Style::stroke_width type

rfuest, 2019-11-06:
`stroke_width` is currently stored as an `u8` which limits the stroke width to 255 pixels, which is less than the size of common displays like 320x240 LCDs. Most strokes are much smaller but IMO it should still be possible to set larger stoke sizes.

I would suggest to use an `u32` instead to be consistent with the `Size` type.

jamwaffles, 2019-11-06:
That makes sense. I was wondering if it makes sense to shrink as many things as possible into `u16`s but that's probably a premature optimisation over `u32`.

rfuest, 2019-11-06:
I did think about reducing the width of the coordinate types in the past to reduce memory consumption. But there might be benefits of having the additional bits for some applications.

Maybe this could be implemented as a feature to choose between the default 32 bit values and a reduced memory footprint 16 bit version.

Could you open another issue if you think that we should discuss this further?

closed 2019-11-06 by rfuest

jamwaffles, 2019-11-06:
> Could you open another issue if you think that we should discuss this further?

I'll do this if/when Rust gets 8 bit AVR support, or if there's a lot of desire in the community for it I think. At that point, cramming things into `u16`s or `u8`s for space/perf will make a lot more sense.

PR #196 (2019-11-06): Change type of Style::stroke_width to u32
Hi! Thank you for helping out with Embedded Graphics development! Please:

- [x] Check that you've added passing tests and documentation
- [x] Add a `CHANGELOG.md` entry in the **Unreleased** section under the appropriate heading (**Added**, **Fixed**, etc)
- [x] Run `rustfmt` on the project
- [x] Run `./build.sh` (Linux/macOS only) and make sure it passes. If you use Windows, check that CI passes once you've opened the PR.

## PR description

This PR changes the type of `Style::stroke_width` to `u32`.
Stroke widths that can't be represented as an `i32 [... cut]