embedded-graphics/embedded-graphics #202: Hit test for touch UI

lupyuen, 2019-11-17:
- Version of embedded-graphics in use (if applicable): 0.5.2
- Target: [real hardware]
- If using real hardware: st7735_lcd

## Description of the problem/feature request/other

Thanks for creating `embedded-graphics` and making it so easy to render Rust screens :-) I'm using it to render the PineTime smart watch UI: https://medium.com/swlh/sneak-peek-of-pinetime-smart-watch-and-why-its-perfect-for-teaching-iot-81b74161c159?source=friends_link&sk=d9301466f5499bece3e7b638e99ec20d

To build an interactive watch UI, I need to implement Hit Testing: Given the touched screen coordinates (X,Y), I need to identify which shape (or text) was touched. How can I do this with `embedded-graphics`? Thanks!

jamwaffles, 2019-11-18:
Hi! Thanks for the issue.

Unfortunately this isn't something e-g currently supports. There's work underway to decouple styling from e-g primitives so that other properties (like hit testing) can be implemented, but it's incomplete. 

The best I can suggest right now is to use [Dimensions](https://docs.rs/embedded-graphics/0.6.0-alpha.2/embedded_graphics/geometry/trait.Dimensions.html) for bounding-box checking. If the touch coordinates land within a given bounding box, you could then iterate over that primitive's pixels to see if the touched coordinate is a transparent or filled pixel. Hope that helps!

lupyuen, 2019-11-18:
Cool I'll explore that. Keep up the great work! :-)

closed 2019-11-18 by lupyuen