embedded-graphics/embedded-graphics #204: Update README.md

rfuest, 2019-11-21:
Some changes made to the doc comments in `lib.rs` weren't also applied to `README.md`.

I would suggest to leave this issue open for now and check this before the `0.6` release.

jamwaffles, 2019-11-21:
Is this related to #203 or is there something else in there that didn't get picked up?

rfuest, 2019-11-21:
Not directly related to #203. But while looking at the new pretty picture I noticed some out of date information in the readme.
For example: The description of the `nalgebra` feature still uses `Coord` and `UnsignedCoord`.

jamwaffles, 2019-11-21:
Ah I see. It would be great to either generate `README.md` from `lib.rs` (or vice versa) so these issues are less of a problem.

closed 2020-03-21 by jamwaffles