embedded-graphics/embedded-graphics #205: Use const fn for constructors

rfuest, 2019-11-22:
Using `const fn` for constructors would allow them to be used to initialize constants. This is especially beneficial in an embedded environment because these object can be stored in flash and don't consume RAM.

Many constructors in e-g are implemented in `impl` blocks with trait bounds, which can't be used to implement `const fn`s in stable Rust. But this should be reevaluated when more `const_fn` features are stabilized.

jamwaffles, 2019-11-22:
Tracking issue for const_fn: rust-lang/rust#57563

RFC for generic const fn parameters: rust-lang/rfcs#2632 (I think this is relevant)

jamwaffles, 2020-05-19:
Something specific that would allow quite a more fns to be const would be the const-stabilisation of `saturating_add` and `saturating_sub`. The feature was added in rust-lang/rust#58246, but is still behind the `const_saturating_int_methods` feature as shown in rust-lang/rust#53718.

peterstuart, 2021-12-29:
const `saturating_add` and `saturating_sub` have been stabilized in [1.47](https://blog.rust-lang.org/2020/10/08/Rust-1.47.html). I'd be happy to work on this, but I guess it'd be a breaking change for the minimum supported Rust version? Is that ok?

jamwaffles, 2021-12-29:
Yeah, 1.47 would be a breaking change. I'd like to not bump the MSRV too often, so perhaps it would be a good idea to leap ahead to 1.56.1 (edition 2021) instead?

That would give us even more `const` items, particularly https://github.com/rust-lang/rust/pull/80962 (along with `.pow()`) as well as min_const_generics.

rfuest, 2021-12-29:
> ... so perhaps it would be a good idea to leap ahead to 1.56.1 (edition 2021) instead?

IMO this would be a good idea, I'm getting tired of having to work around limitations of our ancient MSRV.

jamwaffles, 2021-12-29:
Me too. I'll start working on a PR.

jamwaffles, 2021-12-29:
If you were curious, 1.40.0 came out almost exactly [two years ago](https://github.com/rust-lang/rust/blob/master/RELEASES.md#version-1400-2019-12-19). Not that ancient I suppose ;)

@peterstuart looks like we're settled on 1.56.1 MSRV, so if you want to take a stab at constifying a bunch of methods, please base it on that version (and #638 if it's not merged yet).