embedded-graphics/embedded-graphics #212: Builder for styles

rfuest, 2019-11-30:
Instances of styles are currently often created like this:
```rust
let decoration_style = PrimitiveStyle {
    fill_color: Some(BinaryColor::Off),
    stroke_color: Some(BinaryColor::On),
    stroke_width: 1,
};
```

One problem with this approach is that it isn't possible to add additional fields to the `PrimitityStyle` struct without breaking the user code. Using a builder pattern additional fields could be added without causing a breaking change.

jamwaffles, 2019-11-30:
> One problem with this approach is that it isn't possible to add additional fields to the `PrimitityStyle` struct without breaking the user code.

Yeah I've been thinking that throughout the refactorings.

Do you have any thoughts on what the API should look like?

rfuest, 2019-11-30:
> Do you have any thoughts on what the API should look like?

Not really, without taking a closer look at this first.

But I think it would be a good idea to prevent users from using the pattern in the first post by adding a private dummy field or making all fields private.

I would also like to keep `PrimitiveStyle::with_fill` and `PrimitiveStyle::with_stroke` and only require the builder for more complex styles.

jamwaffles, 2019-11-30:
> But I think it would be a good idea to prevent users from using the pattern in the first post by adding a private dummy field or making all fields private.

A dummy field seems like a better bet (if a little more messy) as having the fields public is pretty handy for getting style properties IMO. 

> I would also like to keep PrimitiveStyle::with_fill and PrimitiveStyle::with_stroke and only require the builder for more complex styles.

That definitely makes sense, particularly if additional properties are added in the future.

rfuest, 2019-12-19:
> A dummy field seems like a better bet (if a little more messy) as having the fields public is pretty handy for getting style properties IMO.

`#[non_exhaustive]` was just stabilized in `1.40.0`, which is much nicer that a dummy field: https://blog.rust-lang.org/2019/12/19/Rust-1.40.0.html#[non_exhaustive]-structs,-enums,-and-variants

rfuest, 2019-12-23:
> Thoughts on requiring 1.40.0 as the MSRV?

I don't mind using the latest stable version as the MSRV in this early stage of e-g and the embedded ecosystem in general.

jamwaffles, 2019-12-23:
Sweet, that's what I was thinking too. I only asked because I'm seeing a few more (non embedded) crates start specifying minimum versions.

PR #215 (2019-12-26): Add PrimitiveStyleBuilder
Hi! Thank you for helping out with Embedded Graphics development! Please:

- [x] Check that you've added passing tests and documentation
- [x] Add a simulator example(s) where applicable
- [x] Add a `CHANGELOG.md` entry in the **Unreleased** section under the appropriate heading (**Added**, **Fixed**, etc)
- [x] Run `rustfmt` on the project
- [x] Run `./build.sh` (Linux/macOS only) and make sure it passes. If you use Windows, check that CI passes once you've opened the PR.

## PR description

This PR adds a builder for `PrimitiveStyle`, which was discussed in #212.

I [... cut]
closed 2019-12-26 by jamwaffles