embedded-graphics/embedded-graphics #216: tinytga crate panics with (some) rle compressed images

david-sawatzke, 2019-12-15:
- Version of embedded-graphics in use (if applicable): git master (4548c) and 0.6.0-alpha2
 (tinytga crate version 0.1.0)
- Target: both

## Description of the problem/feature request/other

When the xcf in the attached tar is exported with RLE compression to tga and loaded in the tga example, it panics (the tga file is also attached). Without compression, it works fine. 

## Test case (if applicable)
Replace `rust-pride.tga` with the attached tga and then run it:
```
$ cargo run --example tga
    Finished dev [unoptimized + debuginfo] target(s) in 0.03s
     Running `[..]/embedded-graphics/target/debug/examples/tga`
thread 'main' panicked at 'attempt to multiply with overflow', /[..]/.cargo/registry/src/github.com-1ecc6299db9ec823/tinytga-0.1.0/src/packet/raw_pack
et.rs:29:27                                                                                                                                                 
note: run with `RUST_BACKTRACE=1` environment variable to display a backtrace.

$ cargo run --example tga --release
    Finished release [optimized] target(s) in 0.04s
     Running `[..]/embedded-graphics/target/release/examples/tga`
thread 'main' panicked at 'index out of bounds: the len is 0 but the index is 0', [..]/.cargo/registry/src/github.com-1ecc6299db9ec823/tinytga-0.1.0/src/lib.rs:226:42
note: run with `RUST_BACKTRACE=1` environment variable to display a backtrace.
```
[reproduce.tar.gz](https://github.com/jamwaffles/embedded-graphics/files/3965497/reproduce.tar.gz)

jamwaffles, 2019-12-16:
Thanks for the bug report! I have to admit the RLE code hasn't been tested particularly well. I'll get round to investigating a fix when I have some spare time.

rfuest, 2019-12-16:
While working on the `image` module I also did some work on `tinytga`, like a nom 5 port and adding additional test images. Because this was outside the scope of that PR I didn't commit the changes back then and forgot about theme afterwards. Maybe these changes would also influence this bug.

jamwaffles, 2019-12-16:
Both `tinytga` and `tinybmp` could use some polish and updates so even if your changes don't resolve this issue it would be good to get them committed.

closed 2019-12-17 by rfuest

jamwaffles, 2019-12-17:
This is fixed by the awesome @rfuest in version [0.2.0](https://crates.io/crates/tinytga/0.2.0) of `tinytga`. Please reopen this issue if you continue to have problems. Thanks!

david-sawatzke, 2019-12-19:
Thanks @rfuest, the image works fine now

PR #217 (2019-12-17): tinytga improvements
Hi! Thank you for helping out with Embedded Graphics development! Please:

- [x] Check that you've added passing tests and documentation
- [x] Add a simulator example(s) where applicable
- [x] Add a `CHANGELOG.md` entry in the **Unreleased** section under the appropriate heading (**Added**, **Fixed**, etc)
- [x] Run `rustfmt` on the project
- [x] Run `./build.sh` (Linux/macOS only) and make sure it passes. If you use Windows, check that CI passes once you've opened the PR.

## PR description

This PR is a collection of improvements to tinytga:
* Port to nom 5
* Color map s [... cut]
commit 50585915 (2019-12-17): tinytga improvements (#217)

* Port tinytga to nom 5

* Add mock display color mappings for Rgb888 and Gray8

* Make footer optional and add color map to header

* Add color map support to TgaIterator and e-g tests

* Fix integer overflow in rle_packet

* Fix link to Font6x8

* Add changelog entries

Fixes #216