embedded-graphics/embedded-graphics #221: Simple Pixel Drawing

jacobrosenthal, 2019-12-19:
In working with basic pixel examples for teaching I noticed some issues
I started on an embedded device and went with something like this which was agreeable
```
    display.draw((0..100).map(|i| Pixel(Point::new(i, i * 2), RgbColor::RED)));
```

When I wanted to draw pixels in a loop and I had to wrap in an iter, which is fine but I think we could impl for pixel?
```
    use embedded_graphics::pixelcolor::Rgb565;
    use core::iter;
    for x in 0..100 {
        let pixel = Pixel::<Rgb565>(Point::new(x, x * 2), RgbColor::RED);
        display.draw(iter::once(pixel));
    }
```

But then wanted to move to simulator and neither of those work. Simulator has drawtarget for draw_pixel so this does work but device doesnt have it so thats not ideal.
```
    for x in 0..100 {
        display.draw_pixel(Pixel(Point::new(x, x * 2), BinaryColor::On));
    }
```
I think we need to implement into iterator for pixels on simulator maybe? 

Also I wanted to be able to a row column iterator to draw all the pixels in a region, basically like this
```
    display.draw((0..DISP_SIZE_X).rows(0..DISP_SIZE_Y).map(|(x, y)| {
        ///something clever
        Pixel::<Rgb565>(Point::new(xprime as i32, yprime as i32), color)
    }
```
but couldnt find a name for it. I PRed it to itertools to see if some variation can live there, otherwise maybe in here? Any thoughts/comments?
https://github.com/rust-itertools/itertools/pull/390

jacobrosenthal, 2019-12-19:
Aha, I guess that iterator IS in itertools as cartesian_product
```
    display.draw((0..DISP_SIZE_X).cartesian_product(0..DISP_SIZE_Y).map(|(x, y)| {
        ///something clever
        Pixel::<Rgb565>(Point::new(xprime as i32, yprime as i32), color)
    }
```
Id like to get a little simple pixel example with all this info for newbies

rfuest, 2019-12-19:
> In working with basic pixel examples for teaching I noticed some issues

Great!

Which e-g version did you use? `DrawTarget::draw_pixel` will be available for all device drivers which support e-g `0.6`, once it is released. And your code uses `display.draw(drawable)` which was changed to `drawable.draw(&mut display)` in `master` to make hardware acceleration possible.

Implementing `Drawable` for `Pixel` would make single pixels drawable without having to use iterators. IMO that would be a good addition to the API.

>     display.draw((0..DISP_SIZE_X).cartesian_product(0..DISP_SIZE_Y).map(|(x, y)| {
>         ///something clever
>         Pixel::<Rgb565>(Point::new(xprime as i32, yprime as i32), color)
>     }

Code like this will soon be supported by iterators for unstyled shapes. https://github.com/jamwaffles/embedded-graphics/issues/156#issuecomment-560014569 provides some example of these changes.

jacobrosenthal, 2019-12-19:
Looks like I was 0.6.0-alpha.2 in the toml but master in the simulator and that was my main problem. With 0.6.0-alpha.2 everywhere I can indeed do all three methods cross platform
```
    display.draw((0..100).map(|i| Pixel(Point::new(i, i * 2), BinaryColor::On)));
```
```
    use core::iter;
    for x in 0..100 {
        let pixel = Pixel::<BinaryColor>(Point::new(x, x * 4), BinaryColor::On);
        display.draw(iter::once(pixel));
    }
```
```
display.draw((0..DISP_SIZE_X).cartesian_product(0..DISP_SIZE_Y).map(|(x, y)| {
    ///something clever
    Pixel::<BinaryColor>(Point::new(xprime as i32, yprime as i32), color)
}
```

But I think the actionable item here would be as you said to Implementing Drawable for Pixel to remove the iter::once when people cant/wont use a functional style

rfuest, 2019-12-19:
I've implemented this in PR #222:

These examples all work with these changes applied:
```rust
(0..100).map(|i| Pixel(Point::new(i, i * 2), BinaryColor::On)).draw(&mut display);
 ```
```rust
for x in 0..100 {
    let pixel = Pixel::<BinaryColor>(Point::new(x, x * 4), BinaryColor::On);
    pixel.draw(&mut display);
}
```
```rust
use itertools::iproduct;

iproduct!(0..DISP_SIZE_X, 0..DISP_SIZE_Y)
    .map(|(x, y)| {
        let color = something_clever(x, y);
        Pixel::<BinaryColor>(Point::new(x as i32, y as i32), color)
    })
    .draw(&mut display);
```

jacobrosenthal, 2019-12-20:
I also have the 0.6.0 branch with  drawtarget implemented for st7735 https://github.com/jacobrosenthal/st7735-lcd-rs and that seems to be working.

How about cutting a new alpha release when this is merged?

jamwaffles, 2019-12-20:
I've been thinking it's about time for another alpha actually.

Looking at the issues list, I think I'd like to get #208 into master before doing another release, but that's the only thing I can see.

PR #222 (2019-12-21): Implement Drawable for Pixel
This PR implements `Drawable` for `Pixel` to allow drawing of single pixels in the same way as styled primitives and texts.

Closes #221
closed 2019-12-21 by rfuest