embedded-graphics/embedded-graphics #223: Add table of drawing operations to docs

rfuest, 2019-12-20:
From https://github.com/jamwaffles/embedded-graphics/pull/222#pullrequestreview-335126128:
>Forgot we had DrawTarget::draw_pixel! Looks good to me, but please add an entry to the changelog under Added - it's a public API change that someone out there might want to know about.

The ability to be able to draw single pixels by using `Pixel(...).draw(&mut display)` isn't very discoverable IMO. I've added some examples to the docs, but they are hard to find inside the `drawable` module.

I suggest to add a table to the front page of the docs, which provides a visual reference of the supported drawing operations:

| Image | Example |
|--------|-----------|
| (Image of a single pixel) | <p>Link to Pixel</p> **Example:** <br> `Pixel(...).draw(&mut display)` |
| (Image of a rectangle) | <p>Link to Rectangle</p> **Example:** <br> `Rectangle(...).draw(&mut display)` |
| ... | ... |

jamwaffles, 2019-12-20:
A table is a really good idea, particularly with screenshots. Visual docs for a visual library 👍.

If they get added to the main page docs, I think we need to do something like the hack in `check_readme.md` and include that huge doc comment from an external markdown file. `lib.rs` is getting pretty long already, and editing lots of text in `///` comments gets painful fast.

jamwaffles, 2019-12-20:
To add: the screenshots should be of the simulator with window title. In the example above, `(Image of a single pixel)` is going to look very strange without any context.

jamwaffles, 2020-03-19:
This is the last meaty issue before we can put 0.6.0 out I think. Do we still want to get it done before release? If so I'm happy to take a stab at it. Do you have any suggestions on how to automatically generate the screenshots?

rfuest, 2020-03-19:
I wouldn't mind if this was postponed. And this might also be a good candidate for the appendix of an eventual future e-g book.

But if you want to take a stab at it go ahead. I would create a program that generates the screenshots using multiple `display.to_image_buffer(&output_settings).save('some.png')` calls (and not the environment variable). This program could be called manually, because the output doesn't change very often and automating this further won't be worth the effort IMO.

jamwaffles, 2020-03-20:
I made a start in #279 but it's becoming a bit of a can of worms to do it in a maintainable way I think. I'm going to postpone this as well as #279 for after 0.6.0 release, which I'll do later today (:tada:)

PR #279 (2020-04-01): Add example generation harness
Hi! Thank you for helping out with Embedded Graphics development! Please:

- [x] Check that you've added passing tests and documentation
- [x] Add a simulator example(s) where applicable
- [x] Add a `CHANGELOG.md` entry in the **Unreleased** section under the appropriate heading (**Added**, **Fixed**, etc) and appropriate crate (`embedded-graphics`, `simulator`, `tinytga`, `tinybmp`) if your changes affect the **public API**
- [x] Run `rustfmt` on the project
- [x] Run `./build.sh` (Linux/macOS only) and make sure it passes. If you use Windows, check that CI pass [... cut]
closed 2020-04-01 by jamwaffles