embedded-graphics/embedded-graphics #225: Text positioning improvements

jamwaffles, 2019-12-22:
From https://github.com/jamwaffles/embedded-graphics/pull/224#pullrequestreview-335608934

> IMO top_left is OK for the current way e-g handles text positioning. But maybe text positioning in general should be discussed further, because most drawing libraries use the baseline to specify text positions to make it easier to mix different font sizes on a single line.

I'd like to be able to position text relative to a given point, not just it's `top_left` coordinate. For example, I might want to position text relative to the bottom-center point of the bounding box. Nine positions (denoted by blue dots) would be supported:

[image]

This could be specified with a new `origin` property added to `TextStyle` defined by the following enum:

```rust
enum TextOrigin {
    TopLeft,
    TopCenter,
    TopRight,
    CenterLeft,
    Center,
    CenterRight,
    BottomLeft,
    BottomCenter,
    BottomRight,
}
```

The `top_left` property should probably be renamed to something like `position` as it will now define where the text origin is put, not the top left corner.

To retain current behaviour:

```rust
egtext!(
    text = "Hello!", 
    position = Point::zero(10, 10), 
    font = Font6x8, 
    origin = TextOrigin::TopLeft
);
```

I'm open to bikeshedding the `origin` and `position` names - I'm sure there's a better pair of words out there.

rfuest, 2019-12-23:
Could we postpone these changes to `0.7`?

I would like to take a closer look at font rendering in general and add support for external font rendering libraries. But this will take a while and shouldn't block the `0.6` release IMO.

jamwaffles, 2019-12-23:
Sounds good to me. I've switched the project to 0.7.0.

bugadani, 2020-08-01:
For some reason I've missed this issue. Horizontal alignment is already implemented by embedded-text, with vertical alignment also planned for a future release.

jamwaffles, 2020-08-06:
I think it might be best to leave the behaviour originally described to an external crate, particularly now that the ball is rolling on that. I'll close this issue.

closed 2020-08-06 by jamwaffles