embedded-graphics/embedded-graphics #229: tinytga and tinybmp dependencies

rfuest, 2019-12-26:
TGA and BMP support in e-g consists of code in the `tinytga` and `tinybmp` crates and additional code inside e-g itself. This way e-g needs to depend on every supported image format which isn't very scalable if other formats get added in the future.

I would suggest to move all format  depended code out of e-g into the `tinytga` and `tinybmp` crates and add an `embedded-graphics` feature to them. E-g would need to provide some interface that these external crates can implement. This would also serve as an template for other image format crates.

jamwaffles, 2019-12-26:
> This way e-g needs to depend on every supported image format which isn't very scalable if other formats get added in the future.

I... actually think this is ok. What scaling problems might there be? There are quite a few formats out there, granted, but I think the number e-g will support will pretty much always stay in the single digits. It also gives e-g the freedom to (for example) choose between multiple JPEG libraries that don't have to know what e-g is. 

I did it this way round so that only one dep and list of features is required, instead of having to provide `embedded-graphics` and then `tinybmp` with `features = ["embedded-graphics"]`. IMO `tinybmp` and `tinytga` should remain e-g agnostic, with e-g pulling them in as required.

That said, the `tiny*` crates could probably do more to support e-g's use case, so there's not so much duplicated in e-g.

rfuest, 2019-12-26:
> What scaling problems might there be? There are quite a few formats out

Some of these format might only be useful for a limited use case, like an specialized internal file format for a single project. Such a format shouldn't be added to e-g which means that the user has to implement their own version of the image drawable, which could otherwise be provided by e-g.

>  It also gives e-g the freedom to (for example) choose between multiple JPEG libraries that don't have to know what e-g is.

That's a good point. But for such cases an adapter crate could be implemented, which contains the e-g specific code.

IMO both ways have their benefits and it depends on how you look at e-g in general. I like to think of e-g as a core component which can be extended by adding external crates. Display drivers already work this way and external font rendering engines would also benefit from this approach. That's the reason I suggested a similar implementation for images.

> That said, the `tiny*` crates could probably do more to support e-g's use case, so there's not so much duplicated in e-g.

That's something I wanted to address, independent of this. `tinybmp` should at least provide a pixel iterator like `tinytga` IMO.

jamwaffles, 2019-12-29:
> IMO both ways have their benefits and it depends on how you look at e-g in general. I like to think of e-g as a core component which can be extended by adding external crates.

We're definitely aligned on that. This has been a vision of mine from day one. As long as something can be turned into an `Iterator<Item = drawable::Pixel>` then any e-g implementing display driver can draw it.

> tinybmp should at least provide a pixel iterator like tinytga IMO.

Definitely. #235 opened.

jamwaffles, 2020-01-20:
I'm beginning to implement this issue, starting with moving the image pixel iterators into their respective crates. This raises a question about the iterators and optional features though. I've got two approaches:

1. Keep the tinybmp/tga iterator exposed by those crates as is, just returning pixel values without coordinates. The `embedded-graphics` feature (need a better name - `e-g`? `graphics`? `e-g-support`?) implements `Drawable` for the `Bmp` or `Tga` struct and handles pixel coordinates within that.

    * Pros
        * **Non-breaking**
    * Cons
        * Hides pixel coordinates away from user in `Drawable::draw()` impl - no way to get an iterator of value _and_ coordinate.

2. Change the crate iterators to instead return `(u32, (u16, u16))` (or `(u32, (u32, u32))`) and do less work in the e-g code. This gives users pixel coordinates without having to pull in e-g.

    * Pros
        * Exposes pixel coordinates to users without requiring e-g
        * Much less work to do in `Drawable::draw()`
    * Cons
        * **Breaking change**
        * Potential performance impact of having to throw away pixel coords if just the pixel values are desired (e.g. loading the image into an OpenGL texture). This could be remedied by adding a `.color_data() -> Iterator<Item = u32>` method to `Bmp` and `Tga`.

Is there an obvious third solution I could use instead?

I'm leaning towards option 2. It's a breaking change, but AFAIK the only consumer of `tinytga`/`tinybmp` is e-g itself, so that's not a huge issue. It also exposes more useful functionality to the user without requiring e-g.

rfuest, 2020-01-21:
From an e-g integration perspective I like the first approach, because `ImageIterator` already implements the 1D to 2D pixel iterator transformation. This way it should be easy to handle Raw, BMP and TGA images the same way.

But the second approach has the advantage that it would also support other iteration orders. A memory efficient JPEG decoder, for example, could decompress and output individual 8x8 blocks without needing a huge buffer.

> The `embedded-graphics` feature (need a better name - `e-g`? `graphics`? `e-g-support`?)

IMO it should be the same as used in the majority of display driver crates. I think you have used `graphics` in your drivers, but I haven't checked if other drivers use the same.

> Potential performance impact of having to throw away pixel coords if just the pixel values are desired (e.g. loading the image into an OpenGL texture).

I don't thinks that's a huge issue, because the coords are used internally anyway. I especially if you have enough resources to run OpenGL.

jamwaffles, 2020-01-24:
> But the second approach has the advantage that it would also support other iteration orders.

Didn't think of that, yeah that's pretty nice. I'm going to at least prototype the second approach to get a feel for what the code/integration looks like and we'll take it from there.

> IMO it should be the same as used in the majority of display driver crates.

`graphics` it is 👍. I've used that in most of the other places I recall adding the feature flag.

> I don't think that's a huge issue, because the coords are used internally anyway. And if you have enough resources to run OpenGL this probably won't matter at all.

More scope creep from me. I'll leave this alone until someone asks for it (if ever).

PR #247 (2020-02-17): "Reverse" integration of tinytga and tinybmp crates
Hi! Thank you for helping out with Embedded Graphics development! Please:

- [x] Check that you've added passing tests and documentation
- [x] Add a simulator example(s) where applicable
- [x] Add a `CHANGELOG.md` entry in the **Unreleased** section under the appropriate heading (**Added**, **Fixed**, etc)
- [x] Run `rustfmt` on the project
- [x] Run `./build.sh` (Linux/macOS only) and make sure it passes. If you use Windows, check that CI passes once you've opened the PR.

## PR description

Closes #229 

TODO

* [x] Split changelogs - close [... cut]
closed 2020-02-17 by jamwaffles